arendst/Tasmota #3560: How to send raw commands via SerialBridge to Nextion displays?

renne, 2018-08-23:
- [x] Searched the problem in issues
- [x] Hardware used :  [Electrodragon ESP8266 SMD Adapter Breakout Boards R3.0](http://www.electrodragon.com/product/esp8266-smd-adapter-breakout-board-r3/) + [Nextion NX4024T032_011 display](https://nextion.itead.cc/resources/datasheets/nx4024t032_011/)

Nextion displays use a sequence of 3x `0xFF` as delimiter. I want to use SerialBridge via MQTT with Node-Red:
```
backlog Baudrate 9600; SerialDelimiter 255; SerialSend3 h0.val=75\0xFF\0xFF\0xFF;
```
does not work. The display does not react while the command works in linux with minicom and a USB-UART adapter.

```
08:36:21 CMD: backlog Baudrate 9600; SerialDelimiter 255; SerialSend3 h0.val=75\0xFF\0xFF\0xFF;
08:36:21 SRC: WebConsole from 192.168.100.30
08:36:21 RSL: Received Topic /backlog, Data Size 73, Data Baudrate 9600; SerialDelimiter 255; SerialSend3 h0.val=75\0xFF\0xFF\0xFF;
08:36:21 RSL: Group 0, Index 1, Command BACKLOG, Data Baudrate 9600; SerialDelimiter 255; SerialSend3 h0.val=75\0xFF\0xFF\0xFF;
08:36:21 SRC: Backlog
08:36:21 RSL: Received Topic /Baudrate, Data Size 4, Data 9600
08:36:21 RSL: Group 0, Index 1, Command BAUDRATE, Data 9600
08:36:21 APP: Set Baudrate to 9600
08:36:21 MQT: stat/sonoff/RESULT = {"Baudrate":9600}
08:36:21 SRC: Backlog
08:36:21 RSL: Received Topic /SerialDelimiter, Data Size 3, Data 255
08:36:21 RSL: Group 0, Index 1, Command SERIALDELIMITER, Data 255
08:36:21 MQT: stat/sonoff/RESULT = {"SerialDelimiter":255}
08:36:21 SRC: Backlog
08:36:21 RSL: Received Topic /SerialSend3, Data Size 24, Data h0.val=75\0xFF\0xFF\0xFF
08:36:21 RSL: Group 0, Index 3, Command SERIALSEND, Data h0.val=75\0xFF\0xFF\0xFF
08:36:21 MQT: stat/sonoff/RESULT = {"SerialSend":"Done"}
08:36:22 CFG: Saved to flash at F7, Count 10, Bytes 3584
```

Does anyone have an idea what's wrong with the command?

Jason2866, 2018-08-23:
Hallo Renne,

i think as SerialDelimiter is only 1 Character possible. So SerialSend3 wont work here.
May you try SerialSend4 (Raw Serial Send)

ascillato2, 2018-08-25:
Hi,

@renne 

SerialSend4 had worked for you?

ascillato2, 2018-08-27:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks.

renne, 2018-08-28:
@ascillato2 
No, SerialSend4 does not work because of the triple `0xff` as delimiter.
I've started working on a Nextion-specific MQTT-softserial bridge.

gitolicious, 2019-01-12:
I also tried this but failed with `SerialSend3`... I switched to calculating the hex values in node-red and sending them via `SerialSend5`. A _function_ node is doing the following converstion:
```
// str to hex
msg.payload = Buffer.from(msg.payload).toString('hex');

// append delimiter
msg.payload += 'ffffff';

return msg;
```
Works, but is kind of unfortunate as the documentation of `SerialSend3` looks like it should actually support `\0xFF\0xFF\0xFF`

@renne @martinmatias Any more progress on your side?

GianCann, 2019-01-13:
Hi to all,

I'm trying to figure out how to send commands to the Nextion, via Serial Send.
@gitolicious can you confirm that command like txtStatus.txt="Hello word" it's possible to send with this command?:

SerialSend5 7478745374617475732e7478743d2248656c6c6f20576f726c6422FFFFFF

Where _74 78 74 53 74 61 74 75 73 2e 74 78 74 3d 22 48 65 6c 6c 6f 20 57 6f 72 6c 64 22_ is hex _conversion of txtStatus.txt="Hello word"_

However, if it were possible to send the Nextion command in Ascii format, adding the FF FF FF delimiter, would be optimal...

I want to use Tasmota to 'talk' with Nextion Display (via Serial Bridge) because i've a ESP12 in a place that execute only a single task (relay activation) ad i want to use also to display status & value from another device via MQTT.

I hope to see good news on this topic ;)

GianCann, 2019-01-13:
@ascillato can you reopen this issue?

gitolicious, 2019-01-13:
Yes, for me it is working without any problems with the above str2hex conversion.
I am using the regular RX/TX pins of my NodeMCU though. Haven't tried Software Serial Bridge. As I am only doing OTA updates, I don't need RX/TX for anything else and have the generic GPIOs free for other purposes.

GianCann, 2019-01-13:
Ok, thank you @gitolicious 
But i don't want to use Node-Red... 
Software Serial Bridge seems to not support SerialSend4 command but, for me, is not a problem because i don't need hardware serial for other purpose.

However, if the Tasmota guys explain how to send 0xFF 0xFF 0xFF without hexadecimal conversion, it's much better ;)

gitolicious, 2019-01-13:
You can do the conversion also anywhere else, doesn't need to be node-red. Where does your _txtStatus.txt="Hello word"_ text originate from? Maybe there you can hex-encode it.

GianCann, 2019-01-13:
I use Home Assistant and i don't find the way to convert string in hex data, in a template/automation.

GianCann, 2019-01-13:
I try to put an eye to Node Red, but only for a sigle string-to-hex formatting seems to shot a mosquito with a bazooka :)

I hope in a smart (ad simple) solution from Tasmota Team!

gitolicious, 2019-01-13:
Sure, you are right on that...
I had a look into the code but my C skills are not that great. It is somewhere along the lines of

https://github.com/arendst/Sonoff-Tasmota/blob/15e4918238e49e6e8d6ce504b2e0786e49387447/sonoff/sonoff.ino#L1060-L1069

What confuses me is that the [documentation](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#hardware-serial-bridge-commands) states that number values 128-255 would disable the delimiter. Looking at the implementation I don't think that's true...

Now for our problem, probably a new `SerialDelimiter2` command would need to be introduced accepting multi-character hex delimiters...? Or make the current implementation "hex-aware" by looking for `.startsWith("0x").`

ascillato2, 2019-01-14:
Reopened as requested by @GianCann

GianCann, 2019-01-14:
it seemed too simple ...

SSerialSend2 g0.dis=10ÿÿÿ
07:51:22 CMD: 
07:51:22 RSL: stat/pioggia/RESULT = **{"Command":"Unknown"}**

Sending _ÿÿÿ_ don't work :(

gitolicious, 2019-01-14:
Too bad. Did you check what has actually been sent out on a hex level? (Should be easy to do so with a serial usb adapter.)
What I saw is that when you do `SerialDelimiter ÿ` it is setting it to 193, not 255. Sounds fishy to me.

GianCann, 2019-01-14:
> (Should be easy to do so with a serial usb adapter.)

Can you explain how to do?

> What I saw is that when you do `SerialDelimiter ÿ` it is setting it to 193, not 255. Sounds fishy to me.

https://httplab.it/ASCIICode

The use of ÿÿÿ are reported also in Nextion documentation
https://nextion.itead.cc/resources/documents/instruction-set/

> General Rule or Practice
> 
> 1 | All instructions over serial: are terminated with three bytes of 0xFF 0xFF 0xFF
> ie: decimal: 255  or  hex: 0xFF or  ansichar: ÿ  or binary: 11111111
> ie  byte ndt[3] = {255,255,255}; write(ndt,3);  or print(“\xFF\xFF\xFF”);  or  print(“ÿÿÿ”)

Jason2866, 2019-01-14:
@GianCann A fault that is often made https://httplab.it/ASCIICode  is NOT ASCII Code it is ANSI!
Nextion wrote ANSI...
[image]

GianCann, 2019-01-14:
AAAAH! I have not read with due attention :(

gitolicious, 2019-01-14:
> > (Should be easy to do so with a serial usb adapter.)

> Can you explain how to do?

Connect your ESPs Rx/Tx and GND pins with your USB serial adapter, then any serial monitor of your liking should do. I saw this one being recommended as it has a "hex" switch: https://sites.google.com/site/terminalbpp/
Haven't tried it though.

gitolicious, 2019-01-14:
Alright, I installed the terminal tool and sent out `SerialSend3 ÿÿÿ`. The `ÿ`s are sent out as `0xC3 BF`. So this can't work.
```
Terminal log file
Date: 14.01.2019 - 18:51:03
-----------------------------------------------
C3 BF C3 BF C3 BF 
-----------------------------------------------
Date: 14.01.2019 - 18:51:08
End log file
```
My last resort as an improvement of the workflow is to send out the following:
```
Backlog SerialSend3 g0.dis=10; SerialSend5 FFFFFF
```
By this you save yourself translating the command to hex.

@Jason2866 I don't really get the implications of what you are saying...

Jason2866, 2019-01-14:
@gitolicious 
I just say that ANSI code is not equal ASCII code

GianCann, 2019-01-14:
The Nextion display is a very powerfull (and simple to program) HMI device with touch screen.

At the end, to integrate this device in Tasmota, without sending double Serial Send command, is needed just a simple SetOptionXX 1/0 to activate/deactivate the addition of FFFFFF to command string sended via serial bridge.

GianCann, 2019-01-15:
Ok, i've done some test...

1) (**Software serial bridge** - display connected to GPIO4/GPIO5) I've edit the xdrv_08_serial_bridge.ino adding this code in **SerialBridgeCommand** function and works well:
```
else if ((CMND_SSERIALSEND == command_code) && (XdrvMailbox.index > 0) && (XdrvMailbox.index <= 4)) {
    if (XdrvMailbox.data_len > 0) {
      if (1 == XdrvMailbox.index) {
        SerialBridgeSerial->write(XdrvMailbox.data, XdrvMailbox.data_len);
        SerialBridgeSerial->write("\n");
      }
      else if (2 == XdrvMailbox.index) {
        SerialBridgeSerial->write(XdrvMailbox.data, XdrvMailbox.data_len);
      }
      else if (3 == XdrvMailbox.index) {
        SerialBridgeSerial->write(Unescape(XdrvMailbox.data, &XdrvMailbox.data_len), XdrvMailbox.data_len);
      }
      //For Nextion LCD Display
      else if (4 == XdrvMailbox.index) {
        SerialBridgeSerial->write(XdrvMailbox.data, XdrvMailbox.data_len);
        SerialBridgeSerial->write(0xFF);
        SerialBridgeSerial->write(0xFF);
        SerialBridgeSerial->write(0xFF);
      }
```
I send a Nextiot command like this:

**SSerialSend4 t0.txt="Hello world from Tasmota"**

Note: SSerialSend4 not exist in Tasmota, but only SerialSend4

Nextion elaborate the command and show _Hello world from Tasmota_ in textbox named t0

2) (**Hardware** serial bridge - display connected to GPIO4/GPIO5) I tried with SerialSend2 (nextion-command) + SerialSend5 FFFFFF but this not work. I believe that the problem is serial debug that in boot time send wrong data (log data) to Nextion. 

3)  (Hardware serial bridge) I tried with _rule do disable serial log on system boot_, but nothing to do using _Backlog SerialSend2 t0.txt="Prova"; SerialSend5 FFFFFF_

At this point, i ask to @arendst if it's possibile to add a SetOption_XX_ 0/1 to add/remove 'FFFFFF' for use SerialSend2/SSerialSend2 in trasparent mode for who use the Nextion display.

closed 2019-01-17 by ascillato2

jkrijger67, 2023-05-04:
Hello, 

at the end was there a solution for writing serial Commands to nextion displays?