arendst/Tasmota #3564: Use 5 volt I2c LCD display with 6.1.1.8

red-kooga, 2018-08-23:
How would i connect a 5 volt 1602 I2c lcd display, I have looked for wiki but not found anything
Thanks

Jason2866, 2018-08-23:
@red-kooga 
Via level shifter or go get a 3.3 Volt one. 
Display support is brand new so wiki has to be done from someone. Feel free to do :-)

Jason2866, 2018-08-23:
The I2C modules work pretty well with 3.3 Volt. 
So you have to buy a 3.3 Volt 1602 LCD and solder a usual I2C modul on it.

red-kooga, 2018-08-23:
Thank you, I uncommented  #define USE_DISPLAY in user_config.h and connected i2c interface to 5 volts turned the contrast pot and got time and date display after power cycle.
Now to learn how to use it!

Thanks

Jason2866, 2018-08-23:
On long term your ESP device will get defect. The IO pins are technical designed for 3.3 Volt level.
If you connect the display to 5 Volt the I2C pins are working with 5 Volts.

red-kooga, 2018-08-24:
Thanks I didn't think of that I will use resistors the same as the ultrasonic sensor to lower the voltage on the  i2c pins

Thanks

closed 2018-08-24 by ascillato2

red-kooga, 2018-08-27:
i had to use a level converter, resistor divider didn't work