arendst/Tasmota #3566: When enabling DEBUG, Platformio can't compile

ascillato, 2018-08-23:
**Describe the bug**
_A clear and concise description of what the bug is._

When enabling DEBUG, Platformio can't compile giving the error message:

```
.pioenvs\sonoff-ES\src\sonoff.ino.cpp.o:(.text._Z12DebugFreeMemv+0x4): undefined reference to `g_cont'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\sonoff-ES\firmware.elf] Error 1
```

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used :_ NODEMCU
- [x] _Provide the output of command_ ``status 0`` :
```
13:58:20 MQT: stat/cocina/STATUS = {"Status":{"Module":18,"FriendlyName":["Cocina"],"Topic":"cocina","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:58:20 MQT: stat/cocina/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:01:33","StartupUTC":"2018-08-23T16:56:47","Sleep":0,"BootCount":47,"SaveCount":147,"SaveAddress":"3F9000"}}
13:58:20 MQT: stat/cocina/STATUS2 = {"StatusFWR":{"Version":"6.1.1.9","BuildDateTime":"2018-08-23T13:55:32","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
13:58:20 MQT: stat/cocina/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NetWireless","NetWireless"],"TelePeriod":300,"SetOption":["0E008009","55818000","00000000"]}}
13:58:20 MQT: stat/cocina/STATUS4 = {"StatusMEM":{"ProgramSize":487,"Free":2584,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00002C0A","1F80A984","8C000010","23021796","00000000"]}}
13:58:20 MQT: stat/cocina/STATUS5 = {"StatusNET":{"Hostname":"cocina-1757","IPAddress":"192.168.1.32","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:F7:A6:DD","Webserver":2,"WifiConfig":5}}
13:58:20 MQT: stat/cocina/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F7A6DD","MqttUser":"homeassistant","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:58:20 MQT: stat/cocina/STATUS7 = {"StatusTIM":{"UTC":"Thu Aug 23 16:58:20 2018","Local":"Thu Aug 23 13:58:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"08:07","Sunset":"19:01"}}
13:58:20 MQT: stat/cocina/STATUS10 = {"StatusSNS":{"Time":"2018-08-23T13:58:20"}}
13:58:20 MQT: stat/cocina/STATUS11 = {"StatusSTS":{"Time":"2018-08-23T13:58:20","Uptime":"0T00:01:33","Vcc":3.037,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","RSSI":62,"APMac":"18:D6:C7:80:38:2C"}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_

Enable DEBUG in user_config.h
Compile with Platformio

**Expected behavior**
_A clear and concise description of what you expected to happen._

To Compile

**Screenshots**
_If applicable, add screenshots to help explain your problem._

None

**Additional context**
_Add any other context about the problem here._

Taking out the _free stack_ information of the _xdrv_99_debug.ino_ file (line 196), it compiles and work without problems.

Original code
```
  snprintf_P(log_data, sizeof(log_data), PSTR(D_LOG_DEBUG "FreeRam %d, FreeStack %d (%s)"),
    ESP.getFreeHeap(), 4 * (sp - g_cont.stack), XdrvMailbox.data);
```
Code without free stack
```
  snprintf_P(log_data, sizeof(log_data), PSTR(D_LOG_DEBUG "FreeRam %d (%s)"),
    ESP.getFreeHeap(), XdrvMailbox.data);
```

How can be solved this issue with `g_cont` ?

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-08-24:
Seems to be a problem introduced with core 2.4.2

Will investigate.

arendst, 2018-08-24:
Due to change in handling stack space the buffer was moved to another area making the cont_t area pointer accessible.

Should be fixed now.