arendst/Tasmota #3567: More relais than 8

brainnex, 2018-08-23:
**Have you look for this feature in other issues and in the wiki?**
yes

**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
On generic device like wemos are available up to 11 GPIOs. I have maybe not a very common automation task but i would like\need to use all GPIOs as output.

**Describe alternatives you've considered**
- using EasyESP but would like to stay on tasmota.
- existing workarounds welcome

**Additional context**
nothing

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-08-23:
On a ESP 8266 (WemosMini) you can use 8 Gpios without restrictions.
The 3 Gpios ( left cannot used without restrictions. No good choice for relais.
If you need more GPIOs ESP8266 is not the solution for your needs...

ascillato, 2018-08-23:
The restrictions Jason said are explained in the wiki at:

https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs

benzino77, 2018-08-24:
Have a look here: [wemos d1 mini pins description](https://wiki.wemos.cc/products:d1:d1_mini#pin)
As you can see:

1. D3 and D4 has pullups which is not super comfortable for relays which are Active High - it will "blink" your relay every time you restart Wemos
2. D8 has pulldown which is not super comfortable for relays which are Active Low - it will "blink" your relay every time you restart Wemos
3. You can use RX as long as you do not have plans to send commands via Serial console
4. Every time you restart Wemos there are some information printed on the serial, TX pin (even if you disable serial logging by `seriallog 0`) which is not very comfortable for relays - it will "blink" your relay every time you restart Wemos

> If you need more GPIOs ESP8266 is not the solution for your needs

for couple of day it is (!!!!), thanks to @andrethomas who implements OUTPUT feature for MCP23xxxx device: [https://github.com/arendst/Sonoff-Tasmota/wiki/MCP23008-MCP23017](https://github.com/arendst/Sonoff-Tasmota/wiki/MCP23008-MCP23017)

Jason2866, 2018-08-24:
@benzino77 
I know the driver from Andre :-) With this driver relais are not configured as relais in Tasmota.
You have to configure it with Sensor29
So the restriction to 8 relais in Tasmota is no problem.

benzino77, 2018-08-24:
That's true. Relays connected via MCP230XXX are not "typical" Tasmota relays.

brainnex, 2018-08-24:
@benzino77 
1) the the problem with pull up/down can be bypassed by connecting a mosfet that will trigger the relays
2) "blinking" relays  on startup can be a problem. It depends on the use case if it is a problem or not.
3) wow! the MCP23xxxx is a really nice way to control even more than 8. Didn't see it that is supported by tasmota. This will be indeed the clean way. 

Thanks guys for your feadback!

closed 2018-08-24 by brainnex

andrethomas, 2018-08-24:
@ascillato2 How do you add labels? :)

Jason2866, 2018-08-24:
@andrethomas 
Gurus have more rights 😎😃

andrethomas, 2018-08-24:
haha - thought I was missing a button somewhere LOL

ascillato, 2018-08-24:
Hi, what do you meant? The labels for the issues?