arendst/Tasmota #3568: Can't put B1 into AP mode

asafl, 2018-08-23:
I flashed a Sonoff B1 R2 with the precompiled binary ver. 6.1.1, but since it has no "button", I can't seem to use the 4 button presses needed to put it into AP mode and configure the device. 
I tried connecting and disconnecting GPIO0 to GND 4 times, but it doesn't seem to work. Can't find any other viable solution anywhere. 
Any suggestions? 

Cheers
Asaf

ascillato, 2018-08-23:
6.1.1 has the wifi AP as default. You don't need to press anything. After power on, you should see the Tasmota Wifi to connect to configure your wifi

asafl, 2018-08-23:
Thanks! 
So if I don’t see it I probably messed something up... is there another way to know if the module is still alive?

ascillato, 2018-08-23:
By connecting the Serial and seeing the output.

asafl, 2018-08-23:
Thanks very much.

closed 2018-08-23 by asafl

ascillato, 2018-08-23:
You are welcome :+1: