arendst/Tasmota #3570: master and development branches aren't synchronized

bikemike, 2018-08-24:
It doesn't appear that the development branch is kept in sync with the master branch in the traditional way (merge) which makes it very difficult to move changes between the branches.

Is there a reason it is done the way it is or could this be rectified?

arendst, 2018-08-24:
The reason is that I do not use all features developed in the development branch in a the stable master branch.

Some master branch files are different from development branch for a good reason. To easy master branch releases I make sure the package contains just what is needed for a stable experience.

User PR should be directed to the development branch to keep the master branch free from user experiments.

bikemike, 2018-08-24:
Ok. It sounds like a very manual process. Could the process be improved at all by using feature branches or some other branching/merging strategy?

Maybe some information on the two branches could be added to the wiki. Information such as why some files differ between the two branches and when one should use the master branch or the development branch.

I could only find one reference in the wiki that stated to use the development branch but it didn't give a reason why. 

https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build

closed 2018-08-24 by bikemike