arendst/Tasmota #3571: RuleTimers - timeleft

benzino77, 2018-08-24:
**Have you look for this feature in other issues and in the wiki?**
Yes.
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
There is possibility to start ruletimers in rules. From yesterday you can even overwrite default number (8) of available ruletimers, to other value which fits your needs. In my projects, it will be nice to have an option to see what time left on each ruletimer to finish countdown. It could look like this:
do cmd ` status 12` and the response could be:
```
{"StatusRTMR":{"Time":"2018-08-24T08:16:33", RuleTimers: {"T1":0,"T2":0,"T3":120,"T4":0,"T5":99,"T6":0,"T7":0,"T8":0}}}
```
Which will indicate that the timer "T3" has 120sec. left to finish countdown and "T5" has 99sec.
It can be useful for applications/solutions/controllers which then can show the information:
"The Fan will be off in 6 minutes and 10 seconds"
or
"The garage door will close in 23 seconds"

**Describe alternatives you've considered**
I have no alternative to propose.

**Additional context**
For my use case:
I want to know (show/display on my website) how much time left to finish watering particular zone in my garden.

**(Please, remember to close the issue when the problem has been addressed)**
Ok, I will.

arendst, 2018-08-24:
Love your (status 12) pull solution to keep push messages at a minimum.

Will investigate.

benzino77, 2018-08-24:
Looks promising. I can see that it is only reported when ruletimerX command is executed inside a rule. It is OK. I can implement a "pull" mechanizm by using "unused" timer:
ie.:
```
on event#timeleft do ruletimer12 1 endon
```
the I can get information on time left for each timer by simply call cmnd `event timeleft`

***Observation/hint***
I have noticed that the stat is reported that way:
```
19:53:03 MQT: stat/watering-wm01/RESULT = {"T1":248,"T2":19,"T3":10,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0,"T9":0,"T10":0,"T11":0,"T12":0}
```

I thing there should be rather :
```
19:53:03 MQT: stat/watering-wm01/RESULT = {***RuleTimers***:{"T1":248,"T2":19,"T3":10,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0,"T9":0,"T10":0,"T11":0,"T12":0}}
```

controllers subscribed to `stat/<topic>/result` should be able to check/determine what is the context of the stat result.

arendst, 2018-08-25:
The result is the same as for command ``gpio``:
```
11:51:22 MQT: stat/ring2/GPIO = {"GPIO0":"17 (Button1)","GPIO1":"0 (None)","GPIO2":"56 (Led1i)","GPIO3":"0 (None)","GPIO4":"6 (I2C SDA)","GPIO5":"0 (None)","GPIO12":"21 (Relay1)","GPIO13":"0 (None)","GPIO14":"5 (I2C SCL)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
```
To read the ruletimers all you need is a ruletimer status message which is provided using command ``RuleTimer`` in line with all commands:
```
11:52:50 MQT: stat/ring2/RULETIMER = {"T1":0,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
```

benzino77, 2018-08-25:
Every day I learn something about Tasmota. Definitely you are right: `SetOption4 1` is my friend ;)
Thanks!!!

closed 2018-08-25 by benzino77