arendst/Tasmota #3572: SDM120

sewkol, 2018-08-24:
I uploaded the 6.1.1 version to the WeMos D1 mini, connected the Eastron SDM120 Modbus meter and Tasmota reads the data from the meter nicely.
But, how do I send the total energy consumption to the domoticz by mqtt? I can send current power, voltage, electricity but I do not see the option to send the total energy consumed.

arendst, 2018-08-25:
Make a rule that sends the value to domoticz.

See wiki - rules https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#5-use-of-variables-and-tele-

ascillato2, 2018-08-27:
Hi,

Closing as your question has been answered. Please, if you have further questions just ask to reopen this issue.

closed 2018-08-27 by ascillato2

Sergio-tix, 2019-05-03:
I don't understand how to write a rule. I want to send Active Power to MQTT (SDM120, iobroker).
Here is my rule:
`rule2 on tele-SDM120#ActivePower do ActivePower %value% do publish tele/sonoff/STATE {"ActivePower":%value%} endon`
It does not work, there is an error. What did I write wrong?