arendst/Tasmota #3575: Need help with rules on Wemos-D1 application

MAWMN, 2018-08-24:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x ] Hardware used : Wemos-D1 mini
- [ ] Provide the output of command ``status 0`` :
STATUS 0 OUTPUT HERE
13:44:51 MQT: tasmota/kamer/sonoff05/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Device_Hal_Led1"],"Topic":"sonoff05","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}

**(Please, remember to close the issue when the problem has been addressed)**
Question:
Understood from all the many documents I read that there should be a possibility of making use of rules for my Wemos-D1 device to be able to switch ON and OFF the Leds in my application, but after many tries I was not able to.
I am not using a HomeAutomation application in this moment, but only a MQTT broker.

Is there anyone who can give me any directions in this ?

What is it I am trying to achieve :
My application has 3 LEDs which I would like to Switch ON/OFF as described below, the hardware configuration is as follows :
PIN			GPIO	Function	Description
============================================
D4			2		LampOut		Relay1
D3			0		Mom.Switch	Button1
D5			14		LedOnOff	Led1 (Led RED)  [Working OK and following the Relay1)]
D2			4		LedOnline	???? Led2 (Led GREEN) ???? 
D1			5		LedData		???? Led3 (Led BLUE) ????

I would like to Switch the Led2 ON/OFF depending on the Online state with the MQTT Server, and the Led3 ON for 500 msec when Receiving any MQTT command.

thanks in advance.

ascillato, 2018-08-24:
Hi,

You did not provide the full `status 0`, so some information is missing in order to help you properly. Please, share your` status 0`.

> I would like to Switch the Led2 ON/OFF depending on the Online state with the MQTT Server

As there is no `status 0` complete, I don't know which version you are using, but in last version the led have all 3 behaviours working for the same led. Please read the wiki at command ledstate (https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#main) to choose what you want.

It's main goal is that all 3 behaviour are working at the same time in one led (so as to save other GPIOs for other uses). So, you can see the power status of the relay + wifi connection +  mqtt messages
(if there is no wifi connection or mqtt connection, the led will be blinking)

If you don't like that, you can make the status of the mqtt server be displayed on that other led (led2) configuring that gpio as a relay and using [rules ](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#trigger) (`Mqtt#Connected` when MQTT is connected and `Mqtt#Disconnected` when MQTT is disconnected)

> and the Led3 ON for 500 msec when Receiving any MQTT command.

Again, as before, you can add that behavior to the same led1.

There is no command or event to trigger when there is a mqtt reception message.

curzon01, 2018-08-24:
For Led2 use D2 (GPIO4) =Relay2i
`GPIO4 30`

Command `Power2 ON`/`Power2 OFF` now switches Led2 on/off.

Rule for Mqtt connection:
```
Rule1 on Mqtt#Disconnected do Power2 OFF endon on Mqtt#Connected do Power2 ON endon
Rule1 1
Rule1 4
```

And as @ascillato already mentioned you can't have a trigger for received MQTT command - use LedState instead, not the same you like but I think enough.

MAWMN, 2018-08-24:
Hello,

     thanks for your replies, I will give it a try tomorrow.

For completeness following the full Status 0 message :
16:47:38 CMD: status 0
16:47:39 MQT: tasmota/kamer/sonoff05/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Device_Hal_Led1"],"Topic":"sonoff05","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
16:47:39 MQT: tasmota/kamer/sonoff05/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:30:13","StartupUTC":"2018-08-24T15:17:26","Sleep":0,"BootCount":39,"SaveCount":190,"SaveAddress":"3F6000"}}
16:47:39 MQT: tasmota/kamer/sonoff05/stat/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-08-23T17:17:54","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:47:40 MQT: tasmota/kamer/sonoff05/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mna_AP",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
16:47:40 MQT: tasmota/kamer/sonoff05/stat/STATUS4 = {"StatusMEM":{"ProgramSize":537,"Free":2532,"Heap":13,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDAE794","00000000","23B6179E","00000000"]}}
16:47:40 MQT: tasmota/kamer/sonoff05/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff05-2847","IPAddress":"10.0.0.195","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"5C:CF:7F:3A:0B:1F","Webserver":2,"WifiConfig":3}}
16:47:40 MQT: tasmota/kamer/sonoff05/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.110","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3A0B1F","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:47:41 MQT: tasmota/kamer/sonoff05/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Aug 24 15:47:40 2018","Local":"Fri Aug 24 16:47:40 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:33","Sunset":"19:41"}}
16:47:41 MQT: tasmota/kamer/sonoff05/stat/STATUS10 = {"StatusSNS":{"Time":"2018-08-24T16:47:41"}}
16:47:41 MQT: tasmota/kamer/sonoff05/stat/STATUS11 = {"StatusSTS":{"Time":"2018-08-24T16:47:41","Uptime":"0T00:30:15","Vcc":2.747,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mna_AP","RSSI":74,"APMac":"E0:3F:49:09:D2:D0"}}}

ascillato, 2018-08-26:
+ For the device to not restart when losing wifi you need to change your wificonfig to 5. Type in the console wificonfig 5

MAWMN, 2018-08-26:
@ascillato
Still was NOT sattisfied with the behaviour of the module because the RED Led always switched on when de GREEN LED was switched ON, so there was NO more indication if the Relay1 (The Lamp) was Switched ON or OFF.

**What did I change :**
1   GPIO0 17	(Button1)
2   GPIO2 21	(Relays1)
3   GPIO4 30	(Relays2i)
4   GPIO14 52	(Led1)
5   The rule needed to be changed, the values **ON** and **OFF** needed to be exchanged..
		so now the rule is : Rule1 on Mqtt#Disconnected do Power2 ON endon on Mqtt#Connected do Power2 OFF endon
6   Rule1 1
7   Rule1 4
8   LedState 7
9   WifiConfig 5

**_The behaviour is now (and also very much better):_**
  **RED Led** - Blinks when receiving a message, Switches ON when the Relay1 Lamp is ON, Switches ON When MQTT Broker is down and simultaneously switches OFF the GREEN (Online) Led.
  **GREEN Led** - Switches OFF when the Mosquitto broker is down, Switches ON when the Mosquitto broker is online again. (Wifi Disconnections are NOT shown anymore in this moment)
  
**Question :** is there a possibility to show the Wifi connection or disconnection on the third BLUE Led ??

ascillato, 2018-08-26:
Hi

You are using Tasmota v6.1.1 and the wiki in rules chapter say besides wifi#connected that it is available from v6.1.1c

That is why it is not working for you. 

Please, flash the new Tasmota version from the development branch (you will need to compile by yourself)

MAWMN, 2018-08-27:
@ascillato
Changed the software to 6.1.1.12 and recompiled for my board, and working perfect now.

**_My configuration is now :_**
1   GPIO0 17	(Button1)
2   GPIO2 21	(Relays1)
3   GPIO4 30	(Relays2i)
4   GPIO5 31	(Relays3i)
5   GPIO14 52	(Led1)
6   Rule1 on Mqtt#Disconnected do Power2 ON endon on Mqtt#Connected do Power2 OFF endon
7   Rule2 on Wifi#Disconnected do Power3 ON endon on Wifi#Connected do Power3 OFF endon
8   Rule1 1
9   Rule1 4
10 Rule2 1
11 Rule2 4
12 LedState 7
13 WifiConfig 5

**_The behaviour of the Leds is now :_**
  **RED Led** - Blinks when receiving a message when Wifi is down and when Mosquitto is offline, Switches ON when the Relay Lamp is ON, Switches ON, when Green Led is ON and BLUE Led is ON and Lamp is Relay1 (Lamp Relay) is OFF RED Led is also OFF.
  **GREEN Led** - Switches OFF when the Mosquitto broker is down, Switches ON when the Mosquitto broker is online again.
  **BLUE Led** - Switches OFF when the Wifi is down, Switches ON when the Wifi is up again.
  
  **RED Led** behaves as an **ATTENTION** Led.
  **GREEN Led** behaves as an Mosquitto **SERVER ONLINE / OFFLINE**  Led.
  **BLUE Led** behaves as an **Wifi Signal UP / DOWN** Led.
  
thanks for your support.

**_My Status 0 is now :_**
13:09:22 CMD: status 0
13:09:23 MQT: tasmota/kamer/sonoff05/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Device_Hal","Sonoff2","Sonoff3"],"Topic":"sonoff05","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":7,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:09:24 MQT: tasmota/kamer/sonoff05/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:22:32","StartupUTC":"2018-08-27T11:46:51","Sleep":0,"BootCount":71,"SaveCount":468,"SaveAddress":"3F4000"}}
13:09:24 MQT: tasmota/kamer/sonoff05/stat/STATUS2 = {"StatusFWR":{"Version":"6.1.1.12","BuildDateTime":"2018-08-27T13:43:15","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:09:24 MQT: tasmota/kamer/sonoff05/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mna_AP",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
13:09:25 MQT: tasmota/kamer/sonoff05/stat/STATUS4 = {"StatusMEM":{"ProgramSize":470,"Free":2600,"Heap":14,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDAE794","000003A0","23B6179E","00000000"]}}
13:09:25 MQT: tasmota/kamer/sonoff05/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff05-2847","IPAddress":"10.0.0.195","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"5C:CF:7F:3A:0B:1F","Webserver":2,"WifiConfig":5}}
13:09:25 MQT: tasmota/kamer/sonoff05/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.110","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3A0B1F","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:09:26 MQT: tasmota/kamer/sonoff05/stat/STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 27 12:09:25 2018","Local":"Mon Aug 27 13:09:25 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:38","Sunset":"19:35"}}
13:09:26 MQT: tasmota/kamer/sonoff05/stat/STATUS10 = {"StatusSNS":{"Time":"2018-08-27T13:09:26"}}
13:09:27 MQT: tasmota/kamer/sonoff05/stat/STATUS11 = {"StatusSTS":{"Time":"2018-08-27T13:09:26","Uptime":"0T00:22:35","Vcc":2.740,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"mna_AP","RSSI":72,"APMac":"E0:3F:49:09:D2:D0"}}}

curzon01, 2018-08-27:
**(Please, remember to close the issue when the problem has been addressed)**