arendst/Tasmota #3576: Bunch of errors trying to compile

claudioita, 2018-08-24:
I receive a bunch of errors

In file included from sketch\user_config.h:1:0,

                 from F:\Users\Claudio\Downloads\Sonoff-Tasmota-6.1.1\Sonoff-Tasmota-6.1.1\sonoff\sonoff.ino:33:

F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\Adafruit_ESP8266/Adafruit_ESP8266.h:28:35: error: section attribute not allowed for 'Pchr'

 typedef const PROGMEM char        Pchr; // Ditto, kindasorta

                                   ^

F:\Users\Claudio\Downloads\Sonoff-Tasmota-6.1.1\Sonoff-Tasmota-6.1.1\sonoff\xdrv_05_irremote.ino: In function 'boolean IrSendCommand()':

xdrv_05_irremote:299:7: error: 'StaticJsonBuffer' was not declared in this scope

       StaticJsonBuffer<128> jsonBuf;

       ^

F:\Users\Claudio\Downloads\Sonoff-Tasmota-6.1.1\Sonoff-Tasmota-6.1.1\sonoff\xdrv_05_irremote.ino:299:7: note: suggested alternative:

In file included from F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson\src/ArduinoJson/StaticJsonDocument.hpp:8:0,

                 from F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:10,

                 from F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from F:\Users\Claudio\Downloads\Sonoff-Tasmota-6.1.1\Sonoff-Tasmota-6.1.1\sonoff\sonoff.ino:54:

F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'

 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {

       ^

xdrv_05_irremote:299:29: error: 'jsonBuf' was not declared in this scope

       StaticJsonBuffer<128> jsonBuf;

                             ^

xdrv_05_irremote:301:17: error: 'class ArduinoJson::JsonObject' has no member named 'success'

       if (!root.success()) {

                 ^

F:\Users\Claudio\Downloads\Sonoff-Tasmota-6.1.1\Sonoff-Tasmota-6.1.1\sonoff\xdrv_07_domoticz.ino: In function 'boolean DomoticzMqttData()':

xdrv_07_domoticz:168:5: error: 'StaticJsonBuffer' was not declared in this scope

     StaticJsonBuffer<400> jsonBuf;

     ^

F:\Users\Claudio\Downloads\Sonoff-Tasmota-6.1.1\Sonoff-Tasmota-6.1.1\sonoff\xdrv_07_domoticz.ino:168:5: note: suggested alternative:

In file included from F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson\src/ArduinoJson/StaticJsonDocument.hpp:8:0,

                 from F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:10,

                 from F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from F:\Users\Claudio\Downloads\Sonoff-Tasmota-6.1.1\Sonoff-Tasmota-6.1.1\sonoff\sonoff.ino:54:

F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'

 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {

[... 150 lines cut ...]
In file included from F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson\src/ArduinoJson/StaticJsonDocument.hpp:8:0,

                 from F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:10,

                 from F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from F:\Users\Claudio\Downloads\Sonoff-Tasmota-6.1.1\Sonoff-Tasmota-6.1.1\sonoff\sonoff.ino:54:

F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'

 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {

       ^

xplg_wemohue:652:29: error: 'jsonBuffer' was not declared in this scope

       StaticJsonBuffer<400> jsonBuffer;

                             ^

Più di una libreria trovata per "ArduinoJson.h"
Usata: F:\Users\Claudio\Desktop\Sonoff-Arduino\libraries\ArduinoJson
Non usata: F:\Users\Claudio\Downloads\arduino-1.8.6-windows\arduino-1.8.6\libraries\ArduinoJson-5.11.2
exit status 1
'StaticJsonBuffer' was not declared in this scope

ascillato, 2018-08-24:
Hi,

Please, follow the instructions of the wiki.

You are not installing the ibraries.

ascillato, 2018-08-24:
Remember you can also use the precompiled bins at https://github.com/arendst/Sonoff-Tasmota/releases

ascillato, 2018-08-24:
Have you installed the Tasmota libraries? It is just copy and paste.

ascillato2, 2018-08-24:
Hi,

Could you solve your issue?

closed 2018-08-25 by ascillato2