arendst/Tasmota #3577: KNX support for PWM

jkx, 2018-08-24:
Hi, 
I m'driving LEDs strips or simple LEDs light from Tasmota (I used Weemos D1). This works fine.
But, i'm using KNX (mainly xaal/knx bridge), so I'm searching for a way to send / receive PWM 
values over KNX.

ascillato, 2018-08-24:
Hi,

Have you tried using rules with the [KnxTx_Val1](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#knx) command and the [event#knxrx_val1](https://github.com/arendst/Sonoff-Tasmota/wiki/KNX-Features#usage-examples) ?

jkx, 2018-08-24:
I'm facing several questions : 
1) what is the event matching color change. I tried something like rule on event#color do .. but doesn't work
2) sending / receiving RGB as float is bad.

ascillato, 2018-08-24:
Could you sent / receive PWM values using KNX ?

ascillato2, 2018-08-28:
Hi,

@jkx 

Any news?

jkx, 2018-08-28:
I'm too busy the week to finish my test. I will look at this next week-end.

ascillato2, 2018-09-03:
@jkx

As this issue is on HOLD since 24 aug, I'm closing it temporarily until you have time to continue with it. Please, as soon as you have more information/questions, ask to reopen this issue and we gladly continue with it. Thanks.

closed 2018-09-03 by ascillato2

henfri, 2019-01-26:
Hello,

knx does not send RGB as Float, but as three Integers (each is one group adress).
So, Red would be:
1/0/1 = 255
1/0/2 = 0 
1/0/3 = 0 

Would this make it possible to control RGB/PWM?

Greetings,
Hendrik

ascillato, 2019-01-26:
Hi, color is not implemented yet in knx driver. It is on the to do list.