arendst/Tasmota #3580: Rename toggle Button is Webserver , Possible ?

Chipz0r, 2018-08-24:
Hi .

Is it possible to rename the "toggle" buttons in the webserver ?

I can't see anything in supported commands , wondering if this is possible at all .

Many thanks

Frogmore42, 2018-08-24:
I don't think you can do it with a command.  But, you have access to the source.

https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/language/en-GB.h

look for toggle and change the one that is in parenthesis.  Assuming you are using the English language.

Chipz0r, 2018-08-24:
Thanks for your reply .

I thought this maybe the case , i don't want to recompile as i don't seem to have much luck doing that , i like to use the pre-compiled releases .

Cheers

ascillato2, 2018-08-24:
Hi,

Closing this issue as the question has been answered. Please, ask to reopen if needed. Thanks

closed 2018-08-24 by ascillato2