arendst/Tasmota #3581: Sleep is Stretching the Time for Several Features

ascillato, 2018-08-25:
**Describe the bug**
_A clear and concise description of what the bug is._

When activated, SLEEP feature is stretching the time in Tasmota used for several other features:

- PULSETIME and BLINKTIME
- TELEPERIOD
- PULSETIME
- LWT for MQTT (Leading to MQTT disconnections and reconnections)
- UPTIME
- RULETIMER

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_

- [x] _Searched the problem in issues and in the wiki_
  - Sleep and TelePeriod #1596
  - Sleep & Pulsetime #2850
  - Wiki: https://github.com/arendst/Sonoff-Tasmota/wiki/Energy-Saving

- [x] _Hardware :_ ALL
- [x] _Provide the output of command_ ``status 0`` : N/A

**To Reproduce**
_Steps to reproduce the behavior:_

Use ``Sleep`` command

**Expected behavior**
_A clear and concise description of what you expected to happen._

Features to work close as they do when ``Sleep`` is 0.

**Screenshots**
_If applicable, add screenshots to help explain your problem._

N/A

**Additional context**
_Add any other context about the problem here._

The SLEEP feature is very useful so as to save Energy. As many of the smart automation devices are being powered all the time, any reduction in the device's energy consumption will led to a significant reduction of the power bill all year long as explained in the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Energy-Saving#example-of-power-consumption
So, it is good to make some time corrections as to expand the use of SLEEP to as many as USE-CASES possible.

From the release notes, we can see that the use of SLEEP will not fit all the use-cases as there are several limitations as: Expect overall button/key/switch misses and wrong values on Sonoff Pow. Also, PWM Values could vary resulting in a flickering of the light's dimming and color. Nevertheless, a time correction will be very useful for allowing the use of SLEEP in other cases. Also, SLEEP can be turn on and off using rules, depending on the actual desired behaviour of the device.

A limitation of the Sleep Feature of the ESP8266 is that only works using the ESP8266 Board Library v2.3.0. If Using 2.4.0, 2.4.1 or 2.4.2 the device freeze. According to Arduino issues, sleep feature still have problems because sleep is inside the espressif sdk.
So, regarding the ESP8266 Board Libraries:
* v2.4.2(with lwIP 1.4): works great and fast but without sleep function
* v2.3.0(with lwIP 1.4): works great and have the sleep function working properly.

**(Please, remember to close the issue when the problem has been addressed)**

reloxx13, 2018-08-25:
https://github.com/esp8266/Arduino/issues/4658

arendst, 2018-08-25:
Guys, try to keep this issue as clean as possible;

It's about the time stretch problem that Tasmota will need to solve.

It's NOT about the sleep command that fails on core 2.4.1 and 2.4.2. There are other issues about this "feature".

arendst, 2018-08-25:
Status update:
- RULETIMER not observed during testing. Should not be an issue as it uses a different timing approach.
- All other timings under investigation

arendst, 2018-08-26:
As far as I can tell all sleep related timing issues should now be resolved in v6.1.1.11.

Pls test.