arendst/Tasmota #3582: D1 Mini Pin configuration

tango156157, 2018-08-25:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [- ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
Hello Everybody,

I am configuring the pins on my D1 mini ESP8266 for my home automation device. I am using the latest tasmota firmware 6.1.1. Now I am stuck in a problem. I am selecting the device type to Generic for assigning the pins on D1 Mini. I had assigned the D5 and D6 as switch1 and switch2 and D1 and D2 as Relay1 and Relay2. and the circuit works fine. Now when I assign the SSID and Password on the firmware, it configures it perfectly and  my device is connected to the Wifi. But when I want to change my SSID and Password to a new one, I need to short ground and D3 Pins to bring it to the default state i.e. 192.168.4.1. But it doesn't activates the wifi manager option. I also made a separate reset switch and somehow it isn't working. 
Can anyone tell me as where I am wrong or there's something else needs to be done? Is there any settings in pin configuration module that needs to be changed. I stuck there and there's no way out. Any help will be highly appriciated.

Thank You

Frogmore42, 2018-08-25:
D3 is GPIO 0 shorting that to ground and powering up the device will put it into bootloader mode.  On Sonoff devices, normally button 1 is connected to GPIO0.  Pretty sure you need to define a Button 1 to get multi-press feature that is described in the wiki.

ascillato2, 2018-08-30:
Hi,

Please, share your results when you have time. Thanks.

closed 2018-08-30 by ascillato2