arendst/Tasmota #3583: sonoff basic with TWO toggle switches in the webfrontend

iz3man, 2018-08-25:
i need to control two switches with the sonoff basic (6.1.0. tasmota running), which is easy if you use GPIO[4,14].
but it still shows only ONE toggle button in the webfrontend. and this is the only way (for this application of the switch) to control it.
how would i solve that? thanks!

ascillato, 2018-08-25:
Hi,

As you are not using the sonoff basic as standard, you need to choose GENERIC as module in module configuration and add another relay to control.

For electrical considerations see wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs

iz3man, 2018-08-25:
thanks ascillato. some things are so easy, and still you don't see it :) works great!

ascillato, 2018-08-25:
Great!

Happy to know that now it is working for you. Please, close this issue. Thanks :+1:

closed 2018-08-25 by ascillato2