arendst/Tasmota #3585: Sonoff S31 does not reliably report power (v6.1.1)

grantalewis, 2018-08-25:
When the S31 is in use, the main menu / console output for the S31 does not reliably report correct energy usage.

- [X] Searched the problem in issues and in the wiki
- [X] Hardware used : Sonoff S31 flashed with v6.1.1 (ESPEasy) using FTDI FT232RL USB to TTL Adapter
- [X] Provide the output of command:
```
14:19:54 MQT: stat/sonoff_coffee/RESULT = {"POWER":"ON"}
14:19:54 MQT: stat/sonoff_coffee/POWER = ON
14:19:57 MQT: tele/sonoff_coffee/SENSOR = {"Time":"2018-08-25T14:19:57","ENERGY":{"Total":2.622,"Yesterday":0.325,"Today":0.271,"Power":1022,"Factor":0.99,"Voltage":120,"Current":8.564}}
14:20:26 MQT: tele/sonoff_coffee/SENSOR = {"Time":"2018-08-25T14:20:26","ENERGY":{"Total":2.630,"Yesterday":0.325,"Today":0.279,"Power":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
```

**To Reproduce**
This occurs every time I use the S31. The unit turns on, begins drawing power, and for a brief time correctly reports energy usage. Then Current, Power, and Power Factor drop to zero. Voltage is always reported correctly.

**Expected behavior**
Correct reporting of energy usage.

ascillato, 2018-08-25:
Hi,

Have you make the calibration?

https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow-and-Pow-R2#calibration

grantalewis, 2018-08-25:
No, I haven't. But surely if the energy usage readings are initially accurate, calibration isn't going to fix whatever that's causing them to drop to zero.

arendst, 2018-08-25:
You didn't do the initial issue log tasks like searching here for issues which might solve your problem.

Try this one https://github.com/arendst/Sonoff-Tasmota/issues/3425

And keep searching the issues for answers first please.

ascillato, 2018-08-25:
Ahhh, this problem with S31 is on the v6.1.1 and solved in the code.

Please try to use last version from development branch.

Closing as the issue is duplicated.

Please, do not hesistate on asking anything else you need.

closed 2018-08-25 by ascillato2

grantalewis, 2018-08-25:
Well, I did search, but evidently my searching skills are less than stellar. Thanks for pointing me to the above thread.

I'm doing OTA updates and am not sure how to apply versions from the development branch.. Are there step-by-step instructions somewhere?

ascillato, 2018-08-25:
Yes,

https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite

To use last development code, you will need to compile the code using Arduino IDE or Platformio.

All the steps are in the wiki.

grantalewis, 2018-08-25:
Thank you very much.