arendst/Tasmota #3587: Dimmed LEDs flicker occasionally

seidler2547, 2018-08-25:
**Describe the bug**
I have two applications where I dim lights with Tasmota: H801 LED controller and a custom circuit where I use D7 to PWM drive a small MOSFET to control the 10V dim control line of a dimmable LED ballast (of course with appropriate resistors).
Both worked absolutely fine (bar some minor flickering on low PWM values, which is explainable) with 5.12.0. I have flashed everything with 6.1.1 (built myself with 2.3.0) and also for testing with 5.14.0.

The problem with these newer versions is that from time to time (a couple of times per hour), the lights flicker like crazy, going though 0-100% states randomly. This lasts somewhere between 0.5 and 5 seconds. The ESP does NOT reboot, uptime continues. Log show nothing.

- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : WEMOS D1 mini, H801 from eBay
- [X] _Provide the output of command_``status 0`` :
```
19:33:37 MQT: stat/widim2/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"widim2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
19:33:37 MQT: stat/widim2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.22.170:80/api/arduino/sonoff.ino.bin","RestartReason":"External System","Uptime":"0T00:28:37","StartupUTC":"2018-08-25T18:05:00","Sleep":1,"BootCount":23,"SaveCount":139,"SaveAddress":"F9000"}}
19:33:37 MQT: stat/widim2/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-08-23T15:07:32","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:33:37 MQT: stat/widim2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Access Pointless",""],"TelePeriod":300,"SetOption":["000A8029","55818000"]}}
19:33:37 MQT: stat/widim2/STATUS4 = {"StatusMEM":{"ProgramSize":473,"Free":528,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
19:33:37 MQT: stat/widim2/STATUS5 = {"StatusNET":{"Hostname":"widim2-6427","IPAddress":"192.168.22.245","Gateway":"192.168.22.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.22.1","Mac":"68:C6:3A:D6:39:1B","Webserver":2,"WifiConfig":5}}
19:33:37 MQT: stat/widim2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.22.17","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D6391B","MqttUser":"","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:33:37 MQT: stat/widim2/STATUS7 = {"StatusTIM":{"UTC":"Sat Aug 25 18:33:37 2018","Local":"Sat Aug 25 19:33:37 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
19:33:37 MQT: stat/widim2/STATUS10 = {"StatusSNS":{"Time":"2018-08-25T19:33:37"}}
19:33:37 MQT: stat/widim2/STATUS11 = {"StatusSTS":{"Time":"2018-08-25T19:33:37","Uptime":"0T00:28:37","Vcc":2.785,"POWER":"ON","Dimmer":19,"Fade":"ON","Speed":4,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Access Pointless","RSSI":40,"APMac":"0E:EC:DA:14:29:F7"}}}
```

**Additional context**
I can try to make a video, but the brightness flashes are hard to catch on my auto-exposure phone camera.

seidler2547, 2018-08-25:
Sorry, but that's not it. I disabled Sleep now (must have been a leftover from a reset to default values) and it still flickers. And please read my description, it's not that it's a little unsteady, it goes crazy over the whole brightness range.

ascillato2, 2018-08-25:
If sleep is 0 and your light flickers, the problem is another thing. Reopening...

seidler2547, 2018-08-25:
Thanks. Also, with the H801, sleep is actually not a problem because the LED control is done externally. At least I can set sleep to 250 (for testing) and the LEDs are steady as ever, even when dimmed down very much - until the crazy flickering sets in, which is, it seems, independent of the actual brightness.

arendst, 2018-08-26:
sleep is set to 0 by the lights driver so that explains why you see no difference when sleep is configured as 250.

If you enable more logging and monitor the webconsole you might get more info.

arendst, 2018-08-26:
I had the same once using WS2812 going mad suddenly. After flash erase and restore firmware it was gone. Unexplainable.

ascillato2, 2018-08-28:
@seidler2547 

Have you tried last development version?

seidler2547, 2018-08-28:
I haven't, but I have a suspicion that it might have something to do with WiFi connectivity issues, as in: bad WiFi reception, bad MQTT connectivity, waiting for MQTT and then PWM fails. To verify that I have applied the [patch](https://github.com/arendst/Sonoff-Tasmota/pull/3558) for the Tasmota MQTT library and flashed one of the problematic devices with it. I'll see how it behaves. If it continues, I'll try the dev version, but this dimmer is hidden behind a dry wall on the ceiling and mounted close to the dry wall is the LED panel, so if anything goes wrong with the firmware it means almost an hour work for me to get it down, flashed and up again.

seidler2547, 2018-08-28:
I think I'll need to explain a bit better: I was using PubSub MQTT because of the reconnection bug with the Tasmota library. Now that the fix for this bug is available, I am trying Tasmota MQTT again because it is non-blocking (and thus maybe less prone to network issues).

ascillato, 2018-08-30:
Hi,

If your issue is solved, please, remember to close this issue. Thanks

closed 2018-09-03 by ascillato2