arendst/Tasmota #3588: Wemos RFID RC522

leschNUX, 2018-08-25:
Hi,
is it possible to use Tasmota on a wemos with connected  RC522 RFID Reader Module?

Br

closed 2018-08-25 by ascillato2

CazMan, 2019-01-28:
Just for anyone looking for this solution, I found using the SerialBridge feature of tasmota can be used to integrate sensors and other feature currently not supported byTasmota (Connect a arduino nano to the device, send the sensor data over serial and the tasmota device publish it to MQTT...
[https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#serial-bridge](url)

Gio-dot, 2020-02-07:
> Just for anyone looking for this solution, I found using the SerialBridge feature of tasmota can be used to integrate sensors and other feature currently not supported byTasmota (Connect a arduino [...]

I, i tried this solution, but i can't get communication working between an arduino nano and a Wemos D1 mini Tasmota. Do you have some sketch example and/or connection example?