arendst/Tasmota #3590: status of input pins

Meinsda, 2018-08-25:
Hello,

at the Moment i test the shelly 1 to work with tasmota. 
I can't get to run and can't find the correct configuration. Is it possible to view the status of the Input pins or set the Relais? The only Information i have is, that 
GPIO4 - Relay -> HIGH is on
GPIO5 - Switch -> Low when the SW is shorted

Peter

andrethomas, 2018-08-25:
You can use the status 11 command - in the json you can extract the states I think definately for the relay/power - I do not have a switch configured but it should also be in there - you can do a status 0 also to see the other status values available.

andrethomas, 2018-08-25:
Status 10 will give you switch status:
`stat/lounge1/STATUS10 = {"StatusSNS":{"Time":"2018-08-25T21:39:04","Switch1":"OFF"}}`
Status 11 will give you the rest:
```
stat/lounge1/STATUS11 = {"StatusSTS":{"Time":"2018-08-25T21:39:04","Uptime":"0T00:00:20","Vcc":3.219,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Wireless","RSSI":100,"APMac":"00:00:00:00:00:00"}}}
```

ascillato, 2018-08-25:
Hi,

Sorry but this is a duplicated issue.

All the information that you need to set shelly 1 is at #3559

From that issue:

> The Shelly1 and Shelly2 require pinMode(INPUT) for their switch terminals.

So you need to use last Tasmota version and set the new pin definitions (82 switch1n) in module configuration of the Tasmota webmenu.

closed 2018-08-25 by ascillato2