arendst/Tasmota #3596: Replacing Sonoff Push button switch with rocker switch

Vrnreg, 2018-08-26:
Hello,

I have successfully tazmotized my sonoff 4ch R2 over air without using FTDI adaptor. Can anyone please tell me which 2 points out of the 4 on my sonoff push button has to be connected to the rocker switch if I want to connect it to physical switch to be in sync. I understand that I'll need to change the switchmode tp 3 or 4 as well. But I'm really struggling with the connection points which I need to use with my physical switch. Any help will be highly appreciated.﻿

[image]

Frogmore42, 2018-08-26:
Do you have a DMM?
It is probably pin 4 and one of the others, my guess is that either 1 or 3 would work.  It depends on the orientation of the switches.  They normally short 2 pins to the other 2.

Or, you could use 4 and ground, which looks like it is probably the big trace on the top of the board that goes under the switches.

Be aware that on some versions, GPIO 0 is attached to one of the buttons, you better not replace that with a rocker or you will put the device into bootloader mode when the switch is closed, which probably won't be what you want.

Vrnreg, 2018-08-26:
Yeah. I'm not planning to use rocker on the button which is attached with GPIO. I do have a DMM. But I'm not sure what to do with it. Am I supposed to check the continuity? I'm not much of a electronics guy. But I i just desperately need this to work.

Frogmore42, 2018-08-26:
You can put it on the diode setting and it will beep when the switch is closed/pushed and not beep when it isn't, when you have the correct two pins connected to the meter.

pmknowles, 2018-08-26:
ng it's the right way up Pin 3 is Common and Pin 4 is the signal. It repeats down the board as the lower right one is the signal. You only need to connect one wire to Pin 3 and can ignore the rest of that column.

pmknowles, 2018-08-26:
BTW they are real pig to unsolder. I essentially destroyed the button so that I could desolder the individual pins. I found my solder sucker about an hour later...

andrethomas, 2018-08-26:
@pmknowles I find hot air rework stations to be very useful for those kinds of removal jobs. Mine has a really thin nozzle so it's very directional unlike some of the chinese ones from fleabay :)

Frogmore42, 2018-08-26:
You don't actually have to remove the push buttons.  If the rocker switch is on, pushing the push button will do nothing.  If the rocker switch is off, then the push button will work.  If the push button is continuously pressed, you would not be able to turn off the circuit with the rocker, but that is a highly unlikely occurrence, so much so that I can say don't do that 😀.

Vrnreg, 2018-08-27:
@pmknowles "Assuming it's the right way up Pin 3 is Common and Pin 4 is the signal. It repeats down the board as the lower right one is the signal. You only need to connect one wire to Pin 3 and can ignore the rest of that column."
So you are suggesting that I can connect one end of my switch to 3 and where do I need to connect the other end?

ascillato, 2018-08-29:
Hi

Have you managed to solve your issue?

Vrnreg, 2018-08-29:
So,. I tried like this.

Pressed the push button switch and connected the ends of DMM with the connectivity mode. It was showing value 1 when switched on to that mode. And when the push button is pressed, I pointed one end on 4 and for other 2 points I got some value with a sound when the push button is pressed and nothing when the push button is released. So I assume both those points are connected and I can use any of those points along with my point 4. Am I correct? 

Sorry for being noobish.

pmknowles, 2018-08-29:
0V is on 3 and the GPIO is connected to 4. Yes, you can connect the rocker switch between 3 and 4. The pushbutton connects the other two points (they aren't connected to anything on the board).
I cannot empasise enough that you should not connect aflying lead with just solder to a board with mains on it. Solder is classed as an electrical connection not a mechanical one.  Wires should pass through the board and then be soldered which will give you a safer mechanical connection. That means you should remove the pushbutton and solder the rocker between 3 and 4.
A pull on your flyng lead without that could result in the PCB track peeling off the board and the loose end coming in contact with mains on the board.
You haven't explained why you need a rocker switch by the way.

Vrnreg, 2018-08-29:
Thanks a lot. 
Actually I was planning to connect all the led lights on my new home with 4ch R2 Pros and fans to sonoff POW.
I needed them to be controlled both by physical switches and home-assistant. I found that the normal connection of physical switches without soldering will result in status not syncing with sonoff board. For example if I'm switching the light off using physical switch, then the status will still be "On" on home-assistant. So all I'm trying to do is to sync the sonoff board even if I used the physical switch. I believe soldering/connecting my physical switch with sonoff push button is the only way to accomplish that. Please correct me if I am wrong.
If I'm not soldering my switch to board, I was planning to use this circuit. 
[image]

Frogmore42, 2018-08-29:
@pmknowles while you are correct about solder, since the pushbuttons are switching very low voltage (3V3 nominal) it is fine to remove them, but not necessary.  The rockers in this scenario should not be connected to mains voltage.  If they are, bad things will happen to the esp8266.

@Vrnreg if you connect the rocker switches as shown in the picture the Sonoff won't be able to turn on the lights if the rocker switch has them turned off.  That is a feature or a bug depending on your requirements.  The best way to do it is going to depend on what wiring you have where and how certain you are of what is connected to it.  I would not use something like that hardwired to my house.  Too many chances for something to go wrong in a bad way.

Vrnreg, 2018-08-29:
@Frogmore42 I tried the above picture scenario before flashing my sonoff and it worked fine. However, I didn't try it after flashing with tasmota. I used 2 way switch as shown in the diagram and it worked properly and I was able to control my lights using the both sonoff and physical switch. The only thing is that I was not able to know the correct status of the light as I was using physical switch too.

Frogmore42, 2018-08-29:
@Vrnreg your picture shows the wiring for a set of 3-way switches (where the Sonoff replaces one of the 3-way switches).  Yes, that will allow both the Sonoff and the rocker switch to control the light, but as you discovered, the Sonoff will never really know if the light is on or not.

The old-school way of wiring would add one or more 4-way switches if you needed to control from more locations than 2.  The 4-way switch is really a DPDT switch that reverses the two blue wires.  This allows any of the switches to control the power to the light, but none of them really know if the light is on or not.  I have that setup for my hall lights, which have 3 or 4 switches for each of upstairs and downstairs.

The other way of wiring it is that the Sonoff controls the light directly.  This means you need to locate the Sonoff (does not need to be a Pro) somewhere where it has access to the black and red power wires and the blue and red wires from the rocker AND the red wire to the lamp.

The blue and the red wire for the switch will need to be attached to the push button pins (gnd and GPIO).  The blue wire from the Sonoff will go directly to the red wire from the bulb.  This way, the Pro is always controlling the bulb, so it knows what state it is in.  The rocker switch can then just tell the Sonoff to turn the light on and off (instead of controlling it directly behind the Sonoff's back).

While other people have done similar things, keeping it safe when working with both high and low voltages is challenging.  Even if you are wiring it all together in the same box, it would be hard to do this and meet all safety requirements.  If you don't care about the requirements, think about how you would feel in the event of the death of a loved one due to electrocution or fire.  I am not saying it is likely, but even if it is a little bit wrong, there is a good chance someone will get hurt.  I believe this is the same reason that @pmknowles said not to depend on solder alone.  Not because this wire is carrying mains voltage, but because other wires/components on the board are.  He is correct about that.  As the saying goes, exercise caution and be safe.

Vrnreg, 2018-08-29:
Thanks a lot guys. I guess I'll just stick without soldering. I'll compromise on the correct switch status for safety then.

Vrnreg, 2018-08-29:
@pmknowles So if I solder it using the best wire out there which we currently use for our home wiring and solder it properly so that it won't get pulled on, will it be safe?

pmknowles, 2018-08-29:
Ironically a heavy cable is more likely to cause problems. There’s no current involved so I would probably use a solid core light wire (like telecomms or CAT5 cable), drill a small hole in the enclosure and fit a jack socket (or other suitable 2 pole socket) and then use a twisted pair cable to your switch to minimise interference. I suppose you could even use coax to run to the switch. Unfortunately the whole lid lifts off so you need to leave some slack on the connector for later adjustments.

Vrnreg, 2018-08-29:
@pmknowles Great! So I guess ill go ahead with drilling a hole on sonoff enclosure and soldering a CAT5 cable properly. I'll use some of hot glue at the drilled enclosure as well.

Frogmore42, 2018-08-29:
@pmknowles I am sure you have some knowledge on how to do things safely, perhaps more than most people.  I know enough to know that I don't have enough knowledge to do it safely enough to meet all the requirements for certification.  Does that stop me from doing things, no, not at all.  But, do I encourage someone who seems to have even less knowledge of how to do it, definitely not.

Generally, there won't be an issue.  But as evidenced by the reports you cited, even trained professionals get it wrong sometimes.  The consequences of getting it wrong, though unlikely, can be quite extreme.  So, as Dirty Harry said, "do you feel lucky ..."

pmknowles, 2018-08-30:
Just realised that you could use the pin headers to achieve this without soldering. I would still use a jack plug and use FTDI connectors (for a few pennies)  to connect to the pin header. You could use the GPIO2, Tx or Rx and connect the other lead to Gnd.

ascillato, 2018-08-30:
Please, read the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs for restrictions on connections. Thanks

ascillato2, 2018-09-03:
Hi,

Have you managed to solve your issue?

Vrnreg, 2018-09-03:
Hey there,
Thanks a lot for all the responses. I managed to get it work. But it was still working as a push button, a slight touch on the rocker switch changes the switch to ON or Off.
I guess guess it has something to do with the the switchmode. I think changed it to 4 and after that it's not connecting to my WiFi. Looks like I have to flash it again.

Vrnreg, 2018-09-05:
I'm facing issue with 5he switch mode now. Can anyone suggest which switch mode I can use for rocker switch?

pmknowles, 2018-09-05:
Part of this may be down to using a switch on a 4Ch (which is configured with pushbuttons) It may be worthwhile trying it as a Generic and manually adding the relays and buttons (and switch).

Regards

Phil K

Frogmore42, 2018-09-05:
Read the wiki on switch mode?
https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

A rocker switch that stays in place is just like a toggle switch and needs to use mode 1 or 2.  A rocker switch that is momentary is just like a push button and would us mode 3 or 4.

Vrnreg, 2018-09-05:
I'll try it out. Thanks.

ascillato2, 2018-09-05:
Great.

Please, try that and report back. If you need more assistant, may be that the Tasmota Chat would be faster. Thanks

closed 2018-09-05 by ascillato2

Vrnreg, 2018-09-06:
Okay. If you don't mind, can you please link me to tasmota chat?