arendst/Tasmota #3598: sonoff POW resetting by itself, or what's happening?

iz3man, 2018-08-26:
i got a POW with 6.1.1 running for my heatpump. the heatpump got an inverter and is very easy on the current side. it ramps up very slowly, and it takes about 5-10min until it reaches full speed and then it uses around 1.500W, so just below 7A @220V.

but after some minutes the heatpump shuts off (the display goes out, so it really is completely OFF). after that it takes some time for it to recover (heatpumps don't like to be power toggled).

i opened the loggin window to see what happens, but it doesn't give a lot of info.

```
16:37:38 RSL: tele/waermepumpe/STATE = {"Time":"2018-08-26T16:37:33","Uptime":"0T00:28:07","Vcc":3.238,"POWER1":"ON","Wifi":{"AP":1,"SSId":"izeman","RSSI":88,"APMac":"78:8A:20:2A:02:73"}}
16:37:43 RSL: tele/waermepumpe/SENSOR = {"Time":"2018-08-26T16:37:38","ENERGY":{"Total":483.597,"Yesterday":12.654,"Today":5.497,"Period":6,"Power":1539,"Factor":1.00,"Voltage":212,"Current":6.853}}
16:37:56 MQT: Attempting connection...
16:37:56 MQT: Connected
16:37:56 MQT: tele/waermepumpe/LWT = Online (retained)
16:37:56 MQT: cmnd/waermepumpe/POWER = 
16:37:56 MQT: stat/waermepumpe/RESULT = {"POWER1":"OFF"}
16:37:56 MQT: stat/waermepumpe/POWER1 = OFF (retained)
16:37:56 MQT: stat/waermepumpe/RESULT = {"POWER1":"ON"}
16:37:56 MQT: stat/waermepumpe/POWER1 = ON (retained)
16:37:57 UPP: Multicast (re)joined
16:37:59 MQT: tele/waermepumpe/STATE = {"Time":"2018-08-26T16:37:59","Uptime":"0T00:28:33","Vcc":3.237,"POWER1":"ON","Wifi":{"AP":1,"SSId":"izeman","RSSI":86,"APMac":"78:8A:20:2A:02:73"}}
16:37:59 MQT: tele/waermepumpe/SENSOR = {"Time":"2018-08-26T16:37:59","ENERGY":{"Total":483.605,"Yesterday":12.654,"Today":5.505,"Period":8,"Power":52,"Factor":0.42,"Voltage":216,"Current":0.560}}
```
can anybody identify any issues with this log? i don't see any reboot message or anything.
but on the other hand this is WEBLOG, so maybe i need to attach a serial log?

Frogmore42, 2018-08-26:
You didn't provide the output from status 0.
Weblog is fine.  What you did proved shows your device reconnecting with MQTT and your broker has retained messages.

Let me guess, you are using Home Assistant.  For some reason, people seem to think retained messages are a good idea.  Maybe they are in some cases, and they can be for things that are not changed regularly and you have clients that need to know about them.  But using retained messages for switches on Tasmota is a generally not good idea.  Tasmota is perfectly capable of remembering the state.  It doesn't need something external to remember.  Having something external remember causes the issue you are seeing.

iz3man, 2018-08-26:
you're totally right. it's HOMEASSISTANT. i got the info to use retained messages. but i can turn it off it helps.
that's ALL output there is from working, to OFF, to working again. 
sorry, but i don't understand what you ask me to do :) 
to me  it looks like a RESETTING sonoff?! so better try serial log? afaik weblog can only show what there is as soon as the sonoff is connected to wifi again?!

iz3man, 2018-08-26:
if i read the log, it says that the unit was turned on at 16:09 (this seems right, as i had mainenance work and had the power removed before that), and the issues i observed happened at 16:37. 
so it seems the unit was not rebooted? what caused the relais toggle if not a sonoff's reset?!

ascillato, 2018-08-26:
Your problem seems that your mqtt broker is turning off your sonoff pow. Tasmota is not being rebooted.

You need to configure in home assistant the retain to false and reset your mqtt broker database.

Please, read in the wiki the examples for configuration and also search in the issues for more troubleshooting.

The weblog is the same to a seriallog. If you want to see more information use weblog 4. Never connect any device to the gpios of the pow. It is very DANGEROUS. So, never use the seriallog with a pow while it is connected to mains AC.

Please, for next time, be kind and use the troubleshooting template. That give us a lot of information in order to help you properly

iz3man, 2018-08-27:
i did some "logging" and found this, while doing a 

`mosquitto_sub -v -t 'cmnd/#'`

which gave me this output: 

```
cmnd/waermepumpe/POWER (null)
cmnd/poolpumpe/POWER (null)
```

and it made the sonoff shutdown, but i came up again immediately after (the thin small red bar near the end of the graph)

[image]

this btw is the sonoff next to my problem, and it has a similar problem, where the pump turns on all the time for a single instant, where it should be OFF (almost) all the time. the little green slot are what puzzles me. the longer ones were intended, and happend during testing.

[image]

this all looks to me to be an issue with the MQTT broker and not tasmota or sonoff itself. i will look into "state retaining" and try to fix it that way.

iz3man, 2018-08-27:
result after 1,5h - no problems so far. seems to work, by just setting "retain: false" and to delete mosquitto.db after is stopped it's daemon.
i still get "cmnd/waermepumpe/POWER (null)", but they don't seem to affect the sonoff's behaviour.
any idea what they mean, and where they could come from?`
i'll take some look at it for the next 24h and see how it goes.

ascillato2, 2018-08-27:
> cmnd/waermepumpe/POWER (null)

Seems tha cames from your MQTT broker. May be some extra configuration in your HA?

It is not from Tasmota.

If you feel that your issue is solved, please close it. Thanks.

iz3man, 2018-08-27:
sure. i'll keep an eye on it till tomorrow and then close it. thank you for your support!!

closed 2018-08-28 by iz3man