arendst/Tasmota #3599: Issues with WiFi

svh1985, 2018-08-26:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x]xSearched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Hardware used : WeMos D1 Mini v2.1.0
- [x] Provide the output of command ``status 0`` :
```
19:35:08 CMD: status 0
19:35:08 RSL: STATUS = {"Status":{"Module":18,"FriendlyName":["WeMos D1 Mini"],"Topic":"WeMos Sensor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:35:08 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://ota.xxx.nl/tasmota/svh/wemosd1mini.bin","RestartReason":"External System","Uptime":"0T00:00:46","StartupUTC":"2018-08-26T17:34:22","Sleep":0,"BootCount":6,"SaveCount":9,"SaveAddress":"3FB000"}}
19:35:08 RSL: STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-08-24T16:41:28","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:35:08 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Pretty Fly For A WiFi",""],"TelePeriod":300,"SetOption":["00008001","55818000"]}}
19:35:08 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":527,"Free":2544,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0F8AE794","00000001","23B6179E","00000000"]}}
19:35:08 RSL: STATUS5 = {"StatusNET":{"Hostname":"WeMos Sensor-1333","IPAddress":"172.16.0.51","Gateway":"172.16.0.1","Subnetmask":"255.255.255.0","DNSServer":"172.16.0.2","Mac":"60:01:94:0C:25:35","Webserver":2,"WifiConfig":2}}
19:35:08 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun Aug 26 17:35:08 2018","Local":"Sun Aug 26 19:35:08 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:41","Sunset":"20:42"}}
19:35:08 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-08-26T19:35:08"}}
19:35:08 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-08-26T19:35:08","Uptime":"0T00:00:46","Vcc":2.771,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Pretty Fly For A WiFi","RSSI":96,"APMac":"00:1D:AA:69:D5:3A"}}}
```

Hi,

I'm having issues with the WiFi connection; most of the time it says the password is incorrect, but connects every now and then.

I keep seeing this on the serial monitor:
00:00:00 Project WeMos Sensor WeMos D1 Mini (Topic WeMos Sensor, Fallback DVES_0C2535, GroupTopic sonoffs) Version 6.1.1-2_3_0
00:00:00 WIF: Connecting to AP1 Pretty Fly For A WiFi in mode 11N as WeMos Sensor-1333...
00:00:10 WIF: Connect failed with AP incorrect password
00:00:11 WIF: Connecting to AP1 Pretty Fly For A WiFi in mode 11N as WeMos Sensor-1333...
00:00:20 WIF: Connect failed with AP incorrect password
00:00:20 WIF: WifiManager active for 3 minutes

My WiFi password only consists of numbers and letters.

Thanks!

svh1985, 2018-08-26:
Hi,

Pre flashing I already ran the esptool.py erase_flash command, did it again but it did not help.
-I have a Draytek 2926ac
-Around 40 networks
-WPA2

Maybe the large numbers of networks are causing the issue, but it is not uncommon in the city.

Jason2866, 2018-08-26:
Your SSID contains spaces. Could make troubles. Replace the spaces with underscores or even better try a easy SSID.

Frogmore42, 2018-08-26:
Yes, it's a good idea to pick a channel that has the least interference.  In the US band, channels 1, 6, and 11 are non-overlaping.  But with 40 networks, it is highly unlikely you will find a totally clear one.  In that case, use one of the in between ones, base it on the strength of ones you can see.  There are apps for your phone that will give you a good idea of what is visible.

I live in the country, so no powerful networks anywhere near me.  Nearest is probably 300m.  But, I have some old esp8266 devices that I have not yet updated to Tasmota.  They are in both AP and client mode, so they probably cause some disruption on my network.  It hasn't been bad enough to fix it, but it is noticable in certain locations and some devices handle it better than others.

Ikkeme, 2018-08-27:
I can confirm these issues. My Sonoff Basic R2 is also unable to see my Wifi network and complains about a wrong password even when it is right next to my router. So far it has succeeded once to connect a very short time and it reported a RSSI value of 100. My password consist of lowercase alphabetic characters so nothing fancy. A regular ESP8266 module (actually several of hem) can detect my Wifi nertwork without any problem.
Finding a relatively free channel is not going to work in my case because two large providers offer free public internetaccess for their subscribers so about every modem router in my neighborhood transmits two signals on two different channels and I can detect literally dozens of AP's.

Frogmore42, 2018-08-27:
One thing you can try is completely erase the flash.  Reflash it and then set the password again with the serial terminal connection.

svh1985, 2018-08-29:
I have to agree with @Ikkeme that it is nearly impossible to find a free WiFi channel in crowded areas. Also, most routers switch channels every now and then. My solution for now is to add a password to secure the WIFImanager to prevent hijacking and wait for TASMOTA to reboot after 3 minutes, usually after the reboot it will connect.

Is there a way to change the WiFi retry count? Alternatively adding the same SSID and password for AP2 also makes it reconnect 4 times (2x AP1, 2x AP2), hopefully successfully connecting on one of those retries.

ascillato, 2018-08-29:
Please, use command `wificonfig 5` to make tasmota not use the wifi_manager. See wiki at commands for more information.

svh1985, 2018-08-29:
@ascillato there are cases where I would like the ESP to start the WiFi manager, especially at first setup when the ESP is pre-flashed without the customers SSID and password or the network changed. That's because I cannot flash the firmware including the button settings, so I first need to login via WiFi to configure the button settings to use the button press modes (https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage).

ascillato, 2018-08-29:
`wificonfig 5` is a command for the console. You will have the wifi_manager on your initial config. And after that you can use that command to change the behaviour of Tasmota and not make the AP, just reconnect

svh1985, 2018-08-30:
@ascillato didn't thought of that! But seems like a good solution! And if you use the button to reset the module, it goes back to its defaults and launches the wifi_manager?!

closed 2018-08-30 by svh1985

ascillato, 2018-08-30:
Yes, exactly