arendst/Tasmota #3600: Unable to connect to wifi if password starts with a number

GiorgioAresu, 2018-08-26:
**Describe the bug**
There seem to be a bug with command parsing, because if your WiFi password happen to start with a digit, you won't be able to configure tasmota to connect to it. Response to `Password1 1whatever` will be `{"Password1":""}` and it won't connect.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff Basic
- [ ] _Provide the output of command_``status 0``

**To Reproduce**
Flash the standard sonoff.bin firmware, then connect with termite and send `Password1 1whatever`. The response will be `{"Password1":""}` and it won't connect

**Expected behavior**
The response should be `{"Password1":"****"}` and it should connect

**Additional context**
Tried swapping the first and last characters in the password and it worked, so it's not a matter of special characters (there aren't) or lenght.

arendst, 2018-08-27:
Thnx for the info. Might have caused some problems to some users ;-)

Should be fixed now in v6.1.1.12

ascillato2, 2018-08-27:
Hi,

Closing as this issue is fixed. Please ask to reopen if needed. Thanks.

closed 2018-08-27 by ascillato2