arendst/Tasmota #3605: Serial protocol between EFM8 & ESP

Portisch, 2018-08-27:
**Describe the bug**
The firmware just look for the data 0x55 (sync end) on receiving RF data from the EFM8 chip.
The len have to be also checked when receiving data from the EFM8. Otherwise it is possible to loose data:

https://github.com/arendst/Sonoff-Tasmota/issues/3419#issuecomment-411289174
or:
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/26#issuecomment-416122675

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_

**To Reproduce**
Just receive RF data from the EFM8 with 0x55 in the data. The data will be truncated.

**Expected behavior**
The len of the expected data to be received should be checked.

arendst, 2018-08-27:
Did you check the latest development version?

Pls provide status 0 output as requested when the issues was raised.

I know I made some changes regarding this 0x55 so I need to know what version you experience this problem with.

Portisch, 2018-08-27:
No, I didn't test your firmware. I just got issues reported by sniffing unknown RF data when the data is including 0x55.

In your function: `SonoffBridgeSerialInput`

```
    if (sonoff_bridge_receive_raw_flag) {
      if (!serial_in_byte_counter) {
        serial_in_buffer[serial_in_byte_counter++] = 0xAA;
      }
      serial_in_buffer[serial_in_byte_counter++] = serial_in_byte;
      if (0x55 == serial_in_byte) {  // 0x55 - End of text
        SonoffBridgeReceivedRaw();
        sonoff_bridge_receive_flag = 0;
        return 1;
      }
    }
```
This will break the receive. You have to check for each command the expected len to be received. If the len got received the next (last) byte have to be the sync end 0x55.
For sure it is the bad thing that the original EFM8 firmware do not have such len byte in the protocol...

arendst, 2018-08-27:
So in short:
- Original firmware only uses commands A0 - A5 without len byte and can be decoded until first 0x55 end-of-text
- Portisch supports original commands A0 - A5 also without the len byte
- In addition Portisch supports commands from A6 up where the byte after A6 will contain the len byte excluding the 0x55 end-of-text

Is this correct?

Portisch, 2018-08-27:
Data from EFM8 -> ESP:
0xA0: len = 0
0xA2: len = 0
0xA3: len = 9
0xA4: len = 9

Alternative commands:
0xA6: len = data[2]
0xAA: len = 0
0xAB: len = data[2]
0xB1: do not have a len byte

The len is every time data[len position + 1] ... data[n-1]

arendst, 2018-08-27:
Ok.

So only A6 and AB contain a len byte I have to check and once received the next byte would be 0x55?

closed 2018-08-31 by ascillato2