arendst/Tasmota #3607: MagicHome with IR Receiver get stuck by using Rules to dimm or fade by receiving a IR Code

HoerMirAuf, 2018-08-27:
Hi,

Im using some MagicHome RGB LED Stripe Controller with IR Receiver.
FW is Tasmota 6.1.1.11 but already used 6.11 and other releases.

I'am using some rules to changing the colors, dimming, setting scheme by IR Remote directly in the Device:

rule1 on IrReceived#Data=FFB04F do power1 on endon on IrReceived#Data=FFF807 do power1 off endon on IrReceived#Data=FF906F do dimmer + endon on IrReceived#Data=FFB847 do dimmer - endon on IrReceived#Data=FFB24D do speed + endon on IrReceived#Data=FF00FF do scheme 2 endon on IrReceived#Data=FF58A7 do scheme 0 endon on IrReceived#Data=FF30CF do speed - endon

rule2 on IrReceived#Data=FF9867 do color2 #FF0000 endon on IrReceived#Data=FFE817 do color2 #FF4000 endon on IrReceived#Data=FF02FD do color2 #FF8000 endon on IrReceived#Data=FF50AF do color2 #FFBF00 endon on IrReceived#Data=FF38C7 do color2 #FFFF00 endon on IrReceived#Data=FFD827 do color2 #00FF00 endon on IrReceived#Data=FF48B7 do color2 #00FF40 endon on IrReceived#Data=FF32CD do color2 #00FF80 endon on IrReceived#Data=FF7887 do color2 #00FFBF endon on IrReceived#Data=FF28D7 do color2 #00FFFF endon

rule3 on IrReceived#Data=FF8877 do color2 #0000FF endon on IrReceived#Data=FF6897 do color2 #4000FF endon on IrReceived#Data=FF20DF do color2 #8000FF endon on IrReceived#Data=FF708F do color2 #BF00FF endon on IrReceived#Data=FFF00F do color2 #FF00FF endon on IrReceived#Data=FFA857 do color2 #FFFFFF endon

If i enable fading 'Fade 1' or any scheme the Device will stuck and is not longer accessable. It turns into a boot loop and the only way to unbreak is to flash it. Also the dimming dimmer +/dimmer - will not set the brightness in 10 steps.

ascillato, 2018-08-27:
Hi, please, share your status 0

HoerMirAuf, 2018-08-27:
Current i am running 6.1.1.c ...
after countless flashings it works for a time ... i tought by myself ... damn, what have done? It's running. after one more flashing a other release to reproduce the error i got it again with varoius firmwares. The error especially appear with fading and dimming.... 

the current ststus 0 is:
00:00:04 WIF: verbunden
00:00:04 DNS: initialisiert
00:00:04 HTP: Web-Server aktiv bei sonoff-5325.local mit IP-Adresse 192.168.178.98
00:00:04 RSL: INFO1 = {"Module":"MagicHome NEW","Version":"6.1.1c","FallbackTopic":"DVES_8754CD","GroupTopic":"sonoffs"}
00:00:04 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5325","IPAddress":"192.168.178.98"}
00:00:04 RSL: INFO3 = {"RestartReason":"Hardware Watchdog"}
00:00:06 RSL: RESULT = {"POWER":"ON"}
00:00:06 RSL: POWER = ON
00:00:11 CMD: status 0
00:00:11 RSL: STATUS = {"Status":{"Module":36,"FriendlyName":["WOHN_WAND_LED"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:11 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:00:10","StartupUTC":"","Sleep":0,"BootCount":24,"SaveCount":119,"SaveAddress":"F5000"}}
00:00:11 RSL: STATUS2 = {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018.08.02 10:28:23","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:00:11 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HMA",""],"TelePeriod":300,"SetOption":["00108001","55818000","00000000"]}}
00:00:11 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0FDAE794","00000000","B7B6179E","00000000"]}}
00:00:11 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-5325","IPAddress":"192.168.178.98","Gateway":"192.168.178.5","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.5","Mac":"DC:4F:22:87:54:CD","Webserver":2,"WifiConfig":5}}
00:00:11 RSL: STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:11 1970","Local":"Thu Jan 01 00:00:11 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":99,"Sunrise":"07:43","Sunset":"16:03"}}
00:00:11 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:11"}}
00:00:11 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:11","Uptime":"0T00:00:10","Vcc":3.221,"POWER":"ON","Dimmer":21,"Color":"350000","HSBColor":"0,100,21","Channel":[21,0,0],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"HMA","RSSI":98,"APMac":"34:81:C4:D2:8F:52"}}}
00:00:13 RSL: STATE = {"Time":"1970-01-01T00:00:13","Uptime":"0T00:00:12","Vcc":3.219,"POWER":"ON","Dimmer":21,"Color":"350000","HSBColor":"0,100,21","Channel":[21,0,0],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"HMA","RSSI":98,"APMac":"34:81:C4:D2:8F:52"}}

ascillato2, 2018-08-27:
The Fade and Scheme commands work if you type directly on the console?

gemu2015, 2018-08-28:
we probably should have a mechanism to disable rules automatically if they induce a crash. i had this several times during developing (not with magichome) and its frustrating having to reflash especially when the device is already encased.

HoerMirAuf, 2018-08-28:
Yip, i get the same behavior if i work with dimmer/scheme commands directly in the console.... it result in a boot loop and the only solution to unbrick is to flash the device.

But .... it seem's i have found a workaround. The hardware watchdog will not appear by using the same firmware compiled on 2.4.1 platform. Ok, there are some trouble to get a Webserver in default AP Mode :/ (haven't included my wifi settings in source code), so i flashed first the 2.3.0 binary, setup my wifi, update to the 2.4.1 binary through the Web-Interface and did all the other settings (same as befor).

So far no hardware watchdog or bootloop behavior.... but i'm still in testmode ;)

ascillato2, 2018-08-28:
Hi,

> if i work with dimmer/scheme commands directly in the console.... it result in a boot loop and the only solution to unbrick is to flash the device.

So, this means that the problem is not rules, neither IR, nor magichome.
If you think it is Ok, let me change the title of the issue 

> But .... it seem's i have found a workaround. The hardware watchdog will not appear by using the same firmware compiled on 2.4.1 platform.

So, seems that the library or the compilation was the source of the problem.

Have you tried esplib v2.4.2 ? That library works really good. _(The only downside of v2.4.2 is that sleep feature don't work but for your setup has to be to 0, so it is not a restriction for you)_

> So far no hardware watchdog or bootloop behavior.... but i'm still in testmode ;)

Great! Please, let us know how it goes and if you think that it is solved please remember to close this issue. Thanks :+1:

HoerMirAuf, 2018-08-30:
Hi ascillato2,

... oke ... after testing now round about 24 hours, no crashes or reboot with the 2.4.1 plattform binary.
All work well. Also i gave 2.4.2 palttform a chance. Can't recognize any different to 2.4.1. No accessible default web interface on ip 192.168.4.1 in AP Mode. After etablish a Wifi Connection with a 2.3.0 binary and update to 2.4.2 through Web Interface all work well... no watchdog no reboot loop. 

I think i am satisfied with this solution ;)

Thank's a lot for your efforts !

closed 2018-08-30 by HoerMirAuf