arendst/Tasmota #3609: Best practice question

jpadie, 2018-08-27:
Hi
Looking for best practice advice for the following circumstances

Two t1 2ch, one by the door another by the bed. 

One at the moment has a load attached in the form of a light.  The light will be replaced by a fan+light. Ifan2 on order but unlikely to fit in the fan cowling.  So probably will use a sonoff-base for the light and pull cord for fan. 

I want both t1s to be illuminated when off and not when on.  And I want them to follow each other (in sync), but without using the relays.  Ie use them as buttons only (once the load is removed). This avoids the clicking and reduces the power consumption.  
They currently communicate via knx. Ie there is no mqtt broker that is controlling them although they do report to a broker.  The internet connection is not stable enough to rely on an external broker to intermediate the syncing.  
So in short: looking to keep two switches in sync with led state but not have any relay action in the switches. 
One final question: can I suppress the led behind the WiFi logo via software on a t1 or do I need to desolder that (or its resistor)?

Thanks
Justin

ascillato, 2018-08-27:
Hi,

> Two t1 2ch, one by the door another by the bed.

Not a good idea to have a WIFI enabled device close to your bed. (#1805)

> with led state but not have any relay action in the switches

(#3575)

ascillato, 2018-08-27:
A lot of information is on the wiki also with examples.

jpadie, 2018-08-27:
Thanks.  I'm not concerned by non-ionising radiation.  It's a red herring. 

The second cross reference seems to be predicated on using mqtt to intermediate.  Between the devices.  That is not great as the connection is not stable.  I could use a broker in the building but that seems unnecessary when the devices support knx. 

I guess the question could equally be asked by phrasing it as how can led states be kept in sync between two devices via knx independently of the relay state.  I suspect I will need to fork the code to achieve this.

ascillato2, 2018-08-27:
> The second cross reference seems to be predicated on using mqtt to intermediate.

It is an example on how using a custom control for leds by rules feature. The MQTT and WIFI were there as the author of that issue was persuing, but in your case you don't need that. I point it out as an example on how using rules and leds.

By KNX you can send values, and using rules you can receive them, so, you can make the sync by that.

Please, check all that features in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki). There are several examples there.

ascillato2, 2018-08-27:
Another example at #3402

ascillato, 2018-08-28:
< OFF TOPIC >

> I'm not concerned by non-ionising radiation. It's a red herring.

If you want do the following experiment. Have 2 similar small indoor plants. Put one less than 1 meter from your WIFI Router and the other one more than 5 meters. Check that both plants have about the same quantity of sunlight and put water to them equally. Wait a couple of month and then check which one has grown up more. You will be surprised.

ascillato, 2018-08-28:
Going back to this topic.

If you have further questions or dubts, please do no hesitate to ask. When you feel that your questions has been answered, please close this issue. Thanks.

ascillato, 2018-08-28:
Remember that for support information there are also:
* [Tasmota Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki)
* [Tasmota Forum](https://groups.google.com/d/forum/sonoffusers)
* [Tasmota Chat](https://discord.gg/Ks2Kzd4)

pmknowles, 2018-08-29:
There is also the WebSend command in the WiFi section which sends durect to the IP address of the receiving device

ascillato, 2018-08-30:
Hi,

If you feel that your question has been answered, please, close this issue. Thanks

closed 2018-09-03 by ascillato2