arendst/Tasmota #3611: INA226/CJMCU-226 with NodeMCU/Wemos? Reports wrong voltage

bearli97, 2018-08-27:
Hi,

I don't own a INA219, but a INA226 that I want to use with a NodeMCU. It's connected like I would connect any I2C device: 
```
NodeMCU / INA226
3.3V / VCC
GND / GND
D1 / SCL
D2 / SDA

for voltage:
5V / VBS
```
```
20:03:47 CMD: gpio
20:03:47 RSL: stat/nodemcu_4/RESULT = {"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"6 (I2C SDA)","GPIO5":"5 (I2C SCL)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"0 (None)","GPIO15":"2 (AM2301)","GPIO16":"0 (None)"}
20:04:43 CMD: i2cscan
20:04:43 MQT: stat/nodemcu_4/RESULT = {"I2CScan":"Device(s) found at 0x40"}
```
In Tasmota it is recognized as INA219, but reports the wrong voltage: 5V ends up in 2V, 3.3V ends in 1V. 
Any ideas? Faulty module or not supported by Tasmota?

Thank you.
Marvin

Jason2866, 2018-08-27:
module not supported

Frogmore42, 2018-08-27:
Yeah, they have different values for the voltage register, 4mv/bit vs 1.25mv/bit 
They also have other differences.  There are some libraries that handle both, but not in Tasmota, at least yet.

ascillato2, 2018-11-06:
Hi, there is someone working on a Tasmota's driver for this sensor?

closed 2018-11-06 by ascillato2