arendst/Tasmota #3612: Add a retain flag for the /Tele/MARGINS topic

TheEggi, 2018-08-27:
**Have you look for this feature in other issues and in the wiki?**
Yes
**Is your feature request related to a problem? Please describe.**

There seems to be no option to set the retain flag for the various power level infos. (PowerLow, CurrentLow,...)
**Describe the solution you'd like**
Easy solution would be a flag in the user config like the MQTT_TELE_RETAIN.

**Describe alternatives you've considered**
Another solution would be to add another command like the SensorRetain command.

ascillato, 2018-08-27:
In the wiki at [commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands):

`PowerRetain`

Jason2866, 2018-08-27:
Just for my understanding. For what is this needed?

TheEggi, 2018-08-27:
I am using this flag for some automations and would like to have the info stored.

Jason2866, 2018-08-27:
I guess you use HomeAssistant?

TheEggi, 2018-08-27:
Yes, exactly.

Jason2866, 2018-08-27:
The designer of HA think °retain" is a good idea.
I saw so many problems with retain stuff here already....
Just my 2 cents:
It makes more problems than it solves. Try to trim your setup to work without it

TheEggi, 2018-08-27:
As there can be changes while HA is offline I somehow need a way to make
sure that those changes are getting received. How else would you solve this
without adding another component to my setup?

ascillato, 2018-08-27:
Hi,

What you are requesting is very easy to add to Tasmota. Just change the line 1472 in the file _sonoff.ino_ from

`MqttPublishPrefixTopic_P(option, PSTR(D_CMND_STATUS "9"));`

to

`MqttPublishPrefixTopic_P(option, PSTR(D_CMND_STATUS "9"), MQTT_TELE_RETAIN);`

My personal opinion is that it is better to know the real value all the time than making decisions upon a retained value.

You can make an automation in HA to ask for status 9 when you need it. There is an example in the wiki to gather all Relay status at HA startup.

Also, if you want, you can set retain: true for those values in the configuration.yaml

TheEggi, 2018-08-27:
Thank you for your answer but I guess we are talking about different things. I just want to have the ON/OFF flag that is sent by tasmota when voltage/current/.. changes below/above a configuered value to be retained. Already added the line which needs to be changed in the first post.

ascillato, 2018-08-27:
Ah, ok.

Great to know that you found a solution for your setup.

Jason2866, 2018-08-27:
Glad too.
@ascillato let @andrethomas telephon ring 😂

ascillato, 2018-08-27:
:rofl:

andrethomas, 2018-08-27:
Fortunately, it goes onto auto-silent after 10pm  😂

ascillato, 2018-08-27:
@TheEggi 

Going back to the issue:

> Having the ON/OFF flag that is sent by tasmota when voltage/current/.. changes below/above a configured value, to be retained.

The solution you provided for that is just to modify the _line 755_ of file _xdrv_03_energy.ino_ from

`MqttPublishPrefixTopic_P(TELE, PSTR(D_RSLT_MARGINS));`

to 

`MqttPublishPrefixTopic_P(TELE, PSTR(D_RSLT_MARGINS), MQTT_TELE_RETAIN);`

So, the PR #3614 was made so as to add that.

Thanks

closed 2018-08-27 by ascillato2