arendst/Tasmota #3613: Sonoff Pow erases the Measures when it initializes. Help Me

vicentecandido, 2018-08-27:
The sonoff pow with tasmota is erasing the data several times. I already reinstalled the system and it did not work. It worked perfectly, but suddenly it presented this problem. I noticed that whenever it loses the signal or that it initializes it erases the recorded data.

Log
```
00:00:05 MQT: tele/sonoff_pow1/INFO1 = {"Module":"Sonoff Pow","Version":"6.1.1.11","FallbackTopic":"DVES_3B41C8","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/sonoff_pow1/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff_pow1-0456","IPAddress":"192.168.0.108"}
00:00:05 MQT: tele/sonoff_pow1/INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 MQT: stat/sonoff_pow1/RESULT = {"EnergyReset":{"Total":0.025,"Yesterday":0.000,"Today":0.000}}
00:00:05 MQT: stat/sonoff_pow1/RESULT = {"EnergyReset":{"Total":0.025,"Yesterday":0.000,"Today":0.000}}
00:00:05 MQT: stat/sonoff_pow1/RESULT = {"EnergyReset":{"Total":0.000,"Yesterday":0.000,"Today":0.000}}
```

ascillato, 2018-08-27:
use `wificonfig 5`

ascillato, 2018-08-27:
Please, also share your `Status 0` as the Troubleshooting template states.

vicentecandido, 2018-08-27:
Ascillato, thanks for answer.
However, i'm noob, Sorry.
I don't understand "wificonfig 5" and "status 0"

How do I configure this?

vicentecandido, 2018-08-27:
Ascillato, Thanks for answer.
However, I'm noob.
I don't understand: "wificonfig 5" and "Status 0"

How do I configure this?

ascillato2, 2018-08-27:
@vicentecandido 

Those are commands for the console.

In the webmenu of Tasmota, go to the console and type there `wificonfig 5` in order to make Tasmota not to reboot when losing wifi or mqtt.

The other command (`status 0`) will show all your configuration. Just copy that and paste here.

When we ask that in the Troubleshooting Template is because gave us a lot of information to understand your problem.

ascillato2, 2018-08-27:
There is a lot of information in Tasmota wiki: https://github.com/arendst/Sonoff-Tasmota/wiki

vicentecandido, 2018-08-27:
Thanks again.
I understood now.

```
18:52:10 RSL: STATUS = {"Status":{"Module":6,"FriendlyName":["Sonoff"],"Topic":"sonoff_pow1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:52:10 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:49:24","StartupUTC":"2018-08-27T20:02:46","Sleep":0,"BootCount":21,"SaveCount":51,"SaveAddress":"F9000"}}
18:52:10 RSL: STATUS2 = {"StatusFWR":{"Version":"6.1.1.11","BuildDateTime":"2018-08-26T22:37:48","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:52:10 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["CiscoE900",""],"TelePeriod":300,"SetOption":["00008001","55818000","00000000"]}}
18:52:10 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":469,"Free":532,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","0C0003A0","23B6179E","00000000"]}}
18:52:10 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff_pow1-0456","IPAddress":"192.168.0.108","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"2C:3A:E8:3B:41:C8","Webserver":2,"WifiConfig":5}}
18:52:10 RSL: STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 27 21:52:10 2018","Local":"Mon Aug 27 18:52:10 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"02:00","Sunset":"15:43"}}
18:52:10 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:52:10 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-08-27T18:52:10","ENERGY":{"Total":0.102,"Yesterday":0.000,"Today":0.102,"Power":59,"Factor":0.74,"Voltage":201,"Current":0.397}}}
18:52:10 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-08-27T18:52:10","Uptime":"0T01:49:24","Vcc":3.213,"POWER":"ON","Wifi":{"AP":1,"SSId":"CiscoE900","RSSI":96,"APMac":"78:11:DC:42:B4:79"}}}
```

ascillato2, 2018-08-27:
Now, with wificonfig 5 your sonoff POW will not restart with wifi or mqtt loss.

If your Sonoff POW loses energy will forget the Today energy. All that information is intended to be saved in your Home Automation Software

vicentecandido, 2018-08-27:
Thank you very much, Ascillato2.

ascillato, 2018-08-27:
If you think that your questions has been answered, please close this issue.

Remember that if you close it, then you can reopen it, if you need more help.

afaucogney, 2018-08-28:
A kind of same issue with me. My sonoff pow is powered only a few hours a day, because the device is mapped in serial with a Peak hour contact. 
The Total Energy is lost every days. You said :

>  All that information is intended to be saved in your Home Automation Software"

Why is the reason of this business rule ?

arendst, 2018-08-28:
Reason: Flash wear. Google flash usage statistics and you learn that writing state changes to flash every second will ruin your device.

Solution in your case is doing a restart first before the power down. The restart will write the current values to flash before the reboot and so you won't lose values.

vicentecandido, 2018-08-28:
The problem has not been resolved.
Below, the graphs display the moments resetting.

[image]

ascillato, 2018-08-28:
Hi,

What is the variable that you are showing in your graph?

If it is Energy today it is reseted as expected.

The value saved in Tasmota is Energy Total

vicentecandido, 2018-08-28:
The graph is the HomeAssistant 0.76
And graph is Energy Today.

Energy Today should be reset after 24h.
The problem is that it resets after random intervals.

Below is the log

"12:18:00 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:00","ENERGY":{"Total":0.538,"Yesterday":0.031,"Today":0.507,"Power":2,"Factor":0.16,"Voltage":202,"Current":0.060}}
12:18:02 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:02","ENERGY":{"Total":0.538,"Yesterday":0.031,"Today":0.507,"Power":6,"Factor":0.12,"Voltage":202,"Current":0.248}}
12:18:15 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:15","ENERGY":{"Total":0.538,"Yesterday":0.031,"Today":0.507,"Power":9,"Factor":0.72,"Voltage":202,"Current":0.063}}
12:18:21 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:21","ENERGY":{"Total":0.538,"Yesterday":0.031,"Today":0.507,"Power":2,"Factor":0.14,"Voltage":202,"Current":0.072}}
12:18:24 RSL: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:24","ENERGY":{"Total":0.538,"Yesterday":0.031,"Today":0.507,"Power":4,"Factor":0.18,"Voltage":201,"Current":0.113}}
12:18:25 MQT: Attempting connection...
12:18:25 MQT: Connect failed to 192.168.107:1883, rc -2. Retry in 10 sec
12:18:31 RSL: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:31","ENERGY":{"Total":0.538,"Yesterday":0.031,"Today":0.507,"Power":2,"Factor":0.11,"Voltage":202,"Current":0.079}}
12:18:36 MQT: Attempting connection...
12:18:36 MQT: Connected
12:18:36 MQT: tele/sonoff_pow1/LWT = Online (retained)
12:18:36 MQT: cmnd/sonoff_pow1/POWER = 
12:18:36 MQT: stat/sonoff_pow1/RESULT = {"POWER":"ON"}
12:18:36 MQT: stat/sonoff_pow1/POWER = ON
12:18:36 MQT: stat/sonoff_pow1/RESULT = {"EnergyReset":{"Total":0.538,"Yesterday":0.031,"Today":0.000}}
12:18:36 MQT: stat/sonoff_pow1/RESULT = {"EnergyReset":{"Total":0.538,"Yesterday":0.000,"Today":0.000}}
12:18:36 MQT: stat/sonoff_pow1/RESULT = {"EnergyReset":{"Total":0.000,"Yesterday":0.000,"Today":0.000}}
12:18:38 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:38","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":7,"Factor":0.32,"Voltage":202,"Current":0.105}}
12:18:42 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:42","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":4,"Factor":0.19,"Voltage":202,"Current":0.093}}
12:18:56 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:56","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":6,"Factor":0.08,"Voltage":202,"Current":0.379}}"

vicentecandido, 2018-08-28:
The graph is my HomeAssistant 0.76
The graph is variable Energy Today

It should be reset after 24h.
The problem is that it resets after random intervals.

Below is the log

```
12:18:00 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:00","ENERGY":{"Total":0.538,"Yesterday":0.031,"Today":0.507,"Power":2,"Factor":0.16,"Voltage":202,"Current":0.060}}
12:18:02 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:02","ENERGY":{"Total":0.538,"Yesterday":0.031,"Today":0.507,"Power":6,"Factor":0.12,"Voltage":202,"Current":0.248}}
12:18:15 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:15","ENERGY":{"Total":0.538,"Yesterday":0.031,"Today":0.507,"Power":9,"Factor":0.72,"Voltage":202,"Current":0.063}}
12:18:21 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:21","ENERGY":{"Total":0.538,"Yesterday":0.031,"Today":0.507,"Power":2,"Factor":0.14,"Voltage":202,"Current":0.072}}
12:18:24 RSL: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:24","ENERGY":{"Total":0.538,"Yesterday":0.031,"Today":0.507,"Power":4,"Factor":0.18,"Voltage":201,"Current":0.113}}
12:18:25 MQT: Attempting connection...
12:18:25 MQT: Connect failed to 192.168.107:1883, rc -2. Retry in 10 sec
12:18:31 RSL: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:31","ENERGY":{"Total":0.538,"Yesterday":0.031,"Today":0.507,"Power":2,"Factor":0.11,"Voltage":202,"Current":0.079}}
12:18:36 MQT: Attempting connection...
12:18:36 MQT: Connected
12:18:36 MQT: tele/sonoff_pow1/LWT = Online (retained)
12:18:36 MQT: cmnd/sonoff_pow1/POWER = 
12:18:36 MQT: stat/sonoff_pow1/RESULT = {"POWER":"ON"}
12:18:36 MQT: stat/sonoff_pow1/POWER = ON
12:18:36 MQT: stat/sonoff_pow1/RESULT = {"EnergyReset":{"Total":0.538,"Yesterday":0.031,"Today":0.000}}
12:18:36 MQT: stat/sonoff_pow1/RESULT = {"EnergyReset":{"Total":0.538,"Yesterday":0.000,"Today":0.000}}
12:18:36 MQT: stat/sonoff_pow1/RESULT = {"EnergyReset":{"Total":0.000,"Yesterday":0.000,"Today":0.000}}
12:18:38 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:38","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":7,"Factor":0.32,"Voltage":202,"Current":0.105}}
12:18:42 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:42","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":4,"Factor":0.19,"Voltage":202,"Current":0.093}}
12:18:56 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2018-08-28T12:18:56","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":6,"Factor":0.08,"Voltage":202,"Current":0.379}}
```

ascillato, 2018-08-29:
Hi,

In the log it shows `stat/sonoff_pow1/RESULT = {"EnergyReset": ........`

That is telling us the result of executing the command ENERGYRESET. So, your POW is receiving the ENERGYRESET command.

So, seems that your HomeAutomation Software has an automation to reset your ENERGY when the POW connects to it. Or your MQTT broker has a retained message and It is sending that command each time your POW reconnects.

In my POW with last Tasmota version I don't have those 3 RESULT lines when reconnecting MQTT.

vicentecandido, 2018-08-29:
Do you have an idea how I can fix this?

ascillato2, 2018-08-29:
Check your configuration files of your Home Assistant:

```
configuration.yaml
automation.yaml
```

If that automation is there, is because you added it.

Which broker are you using? the embedded or mosquitto?
If moquitto try to delete the database

vicentecandido, 2018-08-29:
Broker: mosquitto

My code:

```
switch:
  - platform: mqtt
    name: "sonoff_pow1_reset_counter_today"
    state_topic: "stat/sonoff_pow1/EnergyReset1"
    command_topic: "cmnd/sonoff_pow1/energyreset1"
    qos: 1
    payload_on: "0"
    retain: true
  - platform: mqtt
    name: "sonoff_pow1_reset_counter_yesterday"
    state_topic: "stat/sonoff_pow1/EnergyReset2"
    command_topic: "cmnd/sonoff_pow1/energyreset2"
    qos: 1
    payload_on: "0"
    retain: true
  - platform: mqtt
    name: "sonoff_pow1_reset_counter_total"
    state_topic: "stat/sonoff_pow1/EnergyReset3"
    command_topic: "cmnd/sonoff_pow1/energyreset3"
    qos: 1
    payload_on: "0"
    retain: true
```

ascillato2, 2018-08-29:
Great!

Your issue is all the lines 'retain: true'
Your broker is sending ENERGYRESET everytime your POW connects to it!

Change ALL 'retain: true' for 'retain: false'

and turn off your mosquitto, delete its database, and restart it.

vicentecandido, 2018-08-29:
Ascillato2, thank you so much again.

God bless you.

I'll close case for moment.

closed 2018-08-29 by vicentecandido