arendst/Tasmota #3617: openHAB -  capture device IPAddress

meingraham, 2018-08-28:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- Hardware used: Not relevant 
- Provide the output of command ``status 0``: Not relevant

I am trying to add additional information to my TASMOTA Maintenance frame in openHAB. Following the [TASMOTA wiki page for openHAB](https://github.com/arendst/Sonoff-Tasmota/wiki/openHAB), I have retrieved the Reachable status, RSSI level, and APMac for each of my TASMOTA devices. I also want to capture the device IPAddress and have tried to configure an openHAB Item to parse an INFO2 message

`String den_lamp_IPAddress "Den Lamp: IPAddress [%s]" <none> (gIPAddress) { mqtt="<[broker:tele/sonoff-den_lamp/INFO2:state:JSONPATH($.IPAddress)]" }`

and, having failed to get that to work, I have tried the STATUS5 message as well.

`String den_lamp_IPAddress "Den Lamp: IPAddress [%s]" <none> (gIPAddress) { mqtt="<[broker:tele/sonoff-den_lamp/STATUS5:state:JSONPATH($.StatusNET.IPAddress)]" }`

Neither one seems to be retrieving the IPAddress and populating the Item. I realize that the `INFO2` message occurs only when the device is booted and the `STATUS5` message when either the `STATUS 0` or the `STATUS 5` command is invoked. I triggered a device restart to test my `INFO2` syntax to no effect. For the alternate `STATUS5` based Item declaration, I sent the `STATUS 5` command from the console. In neither case was the IPAddress populated in my Item.

What have I missed?

Thanks.

Mike

curzon01, 2018-08-28:
May be the reason is the format of the returned IP address:

- If you have set `IPAddress1 to 0.0.0.0` (device IP is assigned from DHCP server), `STATUS5` and `INFO2` result format for IPAddress is `"0.0.0.0 (x.x.x.x)`" (where` x.x.x.x` is the real IP e.g. `192.168.7.9`)
- If you have given your device a static IP (e  g, `IPAddress1 192.168.7.9`), `STATUS5` and `INFO2` contains the IP itself without an extra () e. g. `"192.168.7.9"`

The command `IPAddress1` itself always returns the format as "**y.y.y.y** (**x.x.x.x**)" where 
- **x.x.x.x** is the current device address (independent if static or from DHCP) 
- **y.y.y.y**=**x.x.x.x** if address x is assigned **static**
- **y.y.y.y**=**0.0.0.0** if address x is assigned from **DHCP**

#### Summarized

##### Set Static IPAddress1 192.168.7.9
`STATUS5`, `INFO2`: ...`"IPAddress","192.168.7.9"`...
Command `IPAddress1`: "192.168.7.9 (192.168.7.9)"

##### Set DHCP IPAddress1 0.0.0.0
`STATUS5`, `INFO2`: ...`"IPAddress","0.0.0.0 (192.168.7.9)"`...
Command `IPAddress1`: "0.0.0.0 (192.168.7.9)"

meingraham, 2018-08-28:
@curzon01 Norbert,

I have everything set up as dynamic IP addresses. Everything is configured to address the devices symbolically (either the Hostname (rarely, if at all) or the MQTT topic ). I've been able to swap out devices and my openHAB setup keeps on ticking w/o any configuration changes.

In my case, both `STATUS5` and `INFO2` display the single IP address string enclosed with quotation marks.

> MQT: tele/sonoff-den_lamp/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-den_lamp-xxxx","IPAddress":"192.168.1.xxx"}

> MQT: stat/sonoff-den_lamp/STATUS5 = {"StatusNET":{"Hostname":"sonoff-den_lamp-xxxx","IPAddress":"192.168.1.xxx","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"deviceMAC","Webserver":2,"WifiConfig":2}}

I could use `STATUS5` _or_ `IPADDRESS`... I could have a rule that fires off (say at openHAB restart) that sends that command via MQTT. That would trigger TASMOTA to publish a corresponding MQTT message. `STATUS5` would be cleaner as I would not have the additional parsing to grab the IP within the parentheses (`STATUS5` just sends the actual IP address).

I tinkered with my `STATUS5` Item and did find a syntax error. So, that approach works now. However, I still would prefer to use `INFO2`. In the case of `STATUS5`, the JSON payload for `IPAddress` is "embedded" inside the `StatusNET` payload. Since the `INFO2` JSON payload does not have this extra "layer", is my `JSONPATH` syntax correct? Does the correct MQTT syntax indeed include `:state` `/INFO2:state`? I'm just following the other examples' pattern.

`String den_lamp_IPAddress "Den Lamp: IPAddress [%s]" <none> (gIPAddress) { mqtt="<[broker:tele/sonoff-den_lamp/INFO2:state:JSONPATH($.IPAddress)]" }`

Thanks.

Mike

curzon01, 2018-08-28:
I have to revise my previous answer - and your assuming Status 0 is Not relevant ist not relevant :grin:

The description above is valid for FW version v6.1.1 and older. For newer FW >=6.1.1 you will receive the IPAdress only for STATUS5 and INFO2 - only cmnd IPAdress1 will return the format "y.y.y.y (x.x.x.x)"

meingraham, 2018-08-28:
True... I guess there is relevant STATUS 0 info. My bad.

6.1.1c

curzon01, 2018-08-28:
Yes, using `INFO2` result the JSONPATH `$.IPAddress` is the correct one.

meingraham, 2018-08-28:
OK, couldn't get the INFO2 message to be "seen" in openHAB and didn't want to expend any more time when an alternate approach I know would work.
```
String  den_lamp_IPAddress
            "Den Lamp: IPAddress [%s]" <none> (gIPAddress)
            { mqtt="
                    >[broker:cmnd/sonoff-den_lamp/STATUS:command:*:5],
                    <[broker:stat/sonoff-den_lamp/STATUS5:state:JSONPATH($.StatusNET.IPAddress)]
                   " }
```

Then with a Rule triggered on System started, I poke the status to get the IP address:
`den_lamp_IPAddress.sendCommand("")`

closed 2018-08-28 by meingraham