arendst/Tasmota #3618: Sonoff-Tasmota 6.1.1.12 unable to set variable to "0"

jansene, 2018-08-28:
**Describe the bug**
_A clear and concise description of what the bug is._
Assigning "0" to either a temporally or persistent variable results in an empty variable.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : sonoff TH10
- [x ] _Provide the output of command_``status 0`` :
```
09:37:06 CMD: status 0
09:37:06 RSL: STATUS = {"Status":{"Module":4,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
09:37:06 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:25","StartupUTC":"2018-08-28T08:36:41","Sleep":0,"BootCount":12,"SaveCount":26,"SaveAddress":"F8000"}}
09:37:06 RSL: STATUS2 = {"StatusFWR":{"Version":"6.1.1.12","BuildDateTime":"2018-08-28T06:00:45","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
09:37:06 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["oregano",""],"TelePeriod":25,"SetOption":["04008000","55818000","00000000"]}}
09:37:06 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":494,"Free":508,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C0003A0","23B6179E","00000000"]}}
09:37:06 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-8155","IPAddress":"10.10.10.110","Gateway":"10.10.10.10","Subnetmask":"255.255.255.0","DNSServer":"208.67.222.222","Mac":"EC:FA:BC:87:1F:DB","Webserver":2,"WifiConfig":5}}
09:37:06 RSL: STATUS7 = {"StatusTIM":{"UTC":"Tue Aug 28 08:37:06 2018","Local":"Tue Aug 28 09:37:06 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:01","Sunset":"19:41"}}
09:37:06 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-08-28T09:37:06","DS18B20":{"Temperature":36.1},"TempUnit":"C"}}
09:37:06 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-08-28T09:37:06","Uptime":"0T00:00:25","Vcc":3.495,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"oregano","RSSI":64,"APMac":"44:D9:E7:F3:69:C3"}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
CMD: mem1 0
MQT: stat/sonoff-TH/RESULT = {"Mem1":""}
CMD: var1 0
MQT: stat/sonoff-TH/RESULT = {"Var1":""}
-or-
with MQTT disabled:
CMD: mem1 0
RSL: RESULT = {"Mem1":""}
CMD: var1 0
RSL: RESULT = {"Var1":""}

**Expected behavior**
_A clear and concise description of what you expected to happen._
CMD: mem1 0
MQT: stat/sonoff/RESULT = {"Mem1":"0"}
CMD: var1 0
MQT: stat/sonoff-TH/RESULT = {"Var1":"0"}
-or-
with MQTT disabled:
CMD: mem1 0
RSL: RESULT = {"Mem1":"0"}
CMD: var1 0
RSL: RESULT = {"Var1":"0"}

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._
Compiled with core 2.3.0, 2.4.0, 2.4.1 results in same behaviour.
Status 0 output taken from OTA flashed binary; https://thehackbox.org/Sonoff-Tasmota/build/sonoff.bin

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-08-28:
Thnx. Regression from 6.1.1.12

jansene, 2018-08-28:
Unable to test latest fixes due to PlatformIO compile errors:
C:/sonota/Sonoff-Tasmota-development/sonoff/sonoff.ino: In function 'void setup()':
C:/sonota/Sonoff-Tasmota-development/sonoff/sonoff.ino:2522:28: error: 'struct SYSCFG' has no member named 'rule_stop'

C:/sonota/Sonoff-Tasmota-development/sonoff/xdrv_10_rules.ino: In function 'boolean RulesCommand()':
C:/sonota/Sonoff-Tasmota-development/sonoff/xdrv_10_rules.ino:507:29: error: 'struct SYSCFG' has no member named 'rule_stop'

arendst, 2018-08-28:
Sorry, forgot a file

closed 2018-08-28 by jansene