arendst/Tasmota #3622: Sonoff Basic with external button and single press/double press/hold

twieland, 2018-08-28:
Hi,

I've been trying for some time now, but I'm not sure how to achieve what I want (or if it's actually possible). I have a Sonoff Basic with an external push button and here's what I want to do:

for single push: change the local relay state 
for double push: send some command via MQTT (-> want to switch another device on/off)
for HOLD: send another command (or the same with HOLD message) via MQTT (-> want to switch a THIRD device on/off)

I've been trying all kinds of different options but didn't succeed so far. Is it even possible?

Thanks!

benzino77, 2018-08-28:
Last two rows from table: [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices)

twieland, 2018-08-28:
That works for the internal button now as expected! What do I have to set for switchmode/switchtopic to make it work for the external button? It's connected to GPIO14 and set to "09 Switch 1"

ascillato, 2018-08-28:
The external button is a pushbutton or a key (like the wall key to turn on/off a light)?

ascillato, 2018-08-28:
The HOLD feature is only available if you set your pushbutton as button in the module configuration.

twieland, 2018-08-28:
ah, ok. I can only set it to Button2 then, right? how can I configure it then to do the same thing as the internal button?

ascillato, 2018-08-28:
Have you managed to solve your issue?

benzino77, 2018-08-29:
> The HOLD feature is only available if you set your pushbutton as button in the module configuration.

I think `switchmode 5` and `switchmode 6` can do the trick with the HOLD functionality for a switch, but I believe that still you need rules (as @ascillato suggested) to do what you want with the second input.

benzino77, 2018-08-29:
@ascillato the example you provided covers two situations for button:
when the button is HOLD for two secs
`rule on button1#state=3 do power1 2 endon`

and when button is pressed once:
`on button1#state=2 do delay endon`

What is a state number for "double press"?

ascillato, 2018-08-29:
> What is a state number for "double press"? [ FOR BUTTONS ]

There is not a state number for "double press". It is designed that double press will toggle the relay. More information at: https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices

In the case you want to avoid the double press feature you can configure your button as switch and also set the switchmode that fits you (like switchmode 5 to make it behave as a pushbutton) [SWITCH does not have double press ]

ascillato, 2018-08-29:
You can have all 3 actions but defining your GPIO as button. In that case the double press will toggle the relay.

There is also an [option](https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices) to swap the actions of the **single press** and **double press**.

So,

**_BUTTON WITH 3 DIFFERENT ACTIONS_**

- As an example:

**_[ assuming gpio0 is configured as button1 ]_**

And that you want:

**single press**: Turn relay 1
**double press**: send a mqtt message
**hold 2 secs**: send another mqtt message

```
buttontopic 0
setoption1 1
setoption11 1
setoption32 20
rule on button1#state=3 do pubilsh cmnd/topicHOLD/power 2 endon on button1#state=2 do pubilsh cmnd/topicDOUBLEPRESS/power 2 endon
rule 1
```

- Another example:

**_[ assuming gpio0 is configured as button1 ]_**

And that you want:

**single press**: send a mqtt message
**double press**: Turn relay 1
**hold 2 secs**: send another mqtt message

```
buttontopic 0
setoption1 1
setoption11 0
setoption32 20
rule on button1#state=3 do pubilsh cmnd/topicHOLD/power 2 endon on button1#state=2 do pubilsh cmnd/topicSINGLEPRESS/power 2 endon
rule 1
```

please note that `setoption11 0`.

**_SWITCHS WITH 2 DIFFERENT ACTIONS_**

**Switchs do not have double press feature**

- Example:

**_[ assuming gpio0 is configured as switch1 and that you have a pushbutton connected ]_**

And that you want:

**single press**: do nothing
**hold 2 secs**: toggle relay 1

```
switchtopic1 0
switchmode1 5
setoption32 20
rule on switch1#state=3 do power1 2 endon on switch1#state=2 do delay endon
rule 1
```

Will add this set of examples to the wiki at rules. Thanks for pointing this out. :+1:

twieland, 2018-08-30:
Thank you, that's perfect!
FYI, the piece I was missing is setting the configuration to "generic" and the external button on GPIO14 as button1. Maybe that'll help others in the furture

closed 2018-08-30 by twieland

michau83, 2018-09-15:
hi, i'm new with github. i found this one quite similar with my issue. I flashed the 2.4.2 fw to nodemcuv1, then i selected the sonoff_basic in configuration. the relay2/switch2 on gpio4/14 work perfect, however, the relay1/swith1 on gpio 12/0 work as push button (momentum switch). I tried with swithmode1 1 but it still work as button push. are there any way to change its behavior, even with modification the code inside?

Henkonoly, 2024-10-19:
> You can have all 3 actions but defining your GPIO as button. In that case the double press will toggle the relay.
> 
> There is also an [option](https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices) to swap the actions of the **single press** and **double press**.
> 
> So,
> 
> **_BUTTON WITH 3 DIFFERENT ACTIONS_**
> 
> * As an example:
> 
> **_[ assuming gpio0 is configured as button1 ]_**
> 
> And that you want:
> 
> **single press**: Turn relay 1 **double press**: send a mqtt message **hold 2 secs**: send another mqtt message
> 
> ```
> buttontopic 0
> setoption1 1
> setoption11 1
> setoption32 20
> rule on button1#state=3 do pubilsh cmnd/topicHOLD/power 2 endon on button1#state=2 do pubilsh cmnd/topicDOUBLEPRESS/power 2 endon
> rule 1
> ```
> 
> * Another example:
> 
> **_[ assuming gpio0 is configured as button1 ]_**
> 
> And that you want:
> 
> **single press**: send a mqtt message **double press**: Turn relay 1 **hold 2 secs**: send another mqtt message
> 
> ```
> buttontopic 0
> setoption1 1
> setoption11 0
> setoption32 20
> rule on button1#state=3 do pubilsh cmnd/topicHOLD/power 2 endon on button1#state=2 do pubilsh cmnd/topicSINGLEPRESS/power 2 endon
> rule 1
> ```
> 
> please note that `setoption11 0`.
> 
> **_SWITCHS WITH 2 DIFFERENT ACTIONS_**
> 
> **Switchs do not have double press feature**
> 
> * Example:
> 
> **_[ assuming gpio0 is configured as switch1 and that you have a pushbutton connected ]_**
> 
> And that you want:
> 
> **single press**: do nothing **hold 2 secs**: toggle relay 1
> 
> ```
> switchtopic1 0
> switchmode1 5
> setoption32 20
> rule on switch1#state=3 do power1 2 endon on switch1#state=2 do delay endon
> rule 1
> ```
> 
> Added this set of examples to the wiki at [rules](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold) and button usage. Thanks for pointing this out. 👍

Hi any change you could help me out?

I'm trying to get a second button to work on a Sonoff basic R2 and use this to trigger automations in HA i have connected a pushbutton on pins GPO01 and ground]

[image]

I would like to know wich rule to use in console to publish mqtt to see the button 2 in Home Assistant

Is this possible?

Thx in advance