arendst/Tasmota #3626: Any Sonoff device compatible with a clamp on current meter such as STC-013

c3reeman, 2018-08-28:
Has anyone used or been able to use a clamp on current meter with any of the sonoff range ? 

such as...

https://www.aliexpress.com/item/THGS-YHDC-30A-SCT013-0-100A-Non-invasive-AC-New-Sensor-Split-Core-Current-Transformer-New/32735255457.html?spm=a2g0s.13010208.99999999.272.69863c00L9G8aK

Its intended use is to monitor the current pulled from a cellar sump pump, accuracy isnt important really its just to get an indication of pump on/off. I know the obvious answer to this maybe use a sonoff POW R2, i did this however it failed and the onboard switch went open circuit after only a month of service... so the very thing that was supposed to monitor sump pump failure is the very thing that caused the failure ! A non invasive current monitor would eliminate this happening.

Frogmore42, 2018-08-29:
That only puts out 1V at 30A, so it would might need an amplifier depending on the load.  Or, you could put it into the analog input.  The NodeMCU boards have a built-in resistor divider to make full scale 3V3.  You would probably want to full-wave rectify the signal and put a capacitor on it to smooth it out.  This should give you an indication of the pump being on or not.

pmknowles, 2018-08-29:
The PZEM-004T works well with Tasmota. If you buy the cheap ESP8266 relay board from the internet (the one with the sub-board parallel to the long edge). Flash it with Tasmota. Select Generic and choose PZEM Rx for Tx and PZEM Tx for Rx. You can then connect the PZEM to the pin header between the relay and power terminal blocks. Use a USB charger PSU or a 5V buck to connect to the power and it will be fine. You'll need to install into an adaptable box as there is live AC on the PZEM. The relay on the board won't work by the way.
The PZEM is fairly accurate and the CT is rated up to 100A. What current is the sump pump rated at?
There are two versions of CT (split and one piece). It goes aroubd the live wire only.
If you want any further information let me know.

c3reeman, 2018-08-29:
Thanks for all the advice guys, I’ve got quite a few parts coming from aliexpress now including a nodemcu , wemo d1 mini and I’ve just added a PZEM-004T following your advice.

Could you be more specific about the esp board your refering to @pmknowles, Have I already baught it ??

Once again thanks for the help

arendst, 2018-08-29:
I use a sonoff basic back to back with the PZEM-004T encased in the box available from itead.

c3reeman, 2018-08-29:
Ah that’s just perfect, sale on over at aliexpress so Ive already got another 20 sonoff basic’s coming.

Now I know I can use a sonoff and simply select the pins I’m much more in my limited knowledge comfort zone 

Great work thankyou

closed 2018-08-29 by c3reeman

arendst, 2018-08-29:
Picture:

[image]

c3reeman, 2018-08-29:
Looks good and I’ve got some of them boxes spare as well, I’m almost good to go I’m just doing a bit more research now and one thing I’m struggling with is the gpios and where to connect to the sonoff basic, please could you give details of which pins you connected it to, And if you have one plugged in and handy could you be so kind as to post a picture of the configuration layout on tasmota web gui ?

I can’t thank you enough for all the help and the great work you have done with tasmota, my entire home automation system is now 90% tasmota I’ve just got a few unreliable z wave door sensors to swap with tasmota and that’s it

arendst, 2018-08-29:
Tasmota config:

[image]

I also had to add a 1k resistor to the PZEM-004T to change the serial interface power need from 5V to 3V3. 

[image]

The four colored wires are then connected to the Sonoff Basic header:
- 3V3 Red
- Rx Yellow
- Tx Green
- Gnd Grey

[image]

c3reeman, 2018-08-29:
Absolutely brilliant thankyou, 
Looking forward to reviving the parts from China !

arendst, 2018-08-29:
As I wanted to document this anyway I just made an entry in the wiki https://github.com/arendst/Sonoff-Tasmota/wiki/PZEM004T-Energy-Monitor

Feel free to update with your experience once build.

pmknowles, 2018-08-29:
Here’s mine using an HW-655 – it works off 5V and does the 3.3V to 5V on board (the relay is inoperative). It’s mounted in a UK dual gang surface mounted box under the fusebox.
[cid:image003.jpg@01D43F96.2A3E0960]

meingraham, 2018-08-29:
@arendst what is the difference between what you've built to monitor energy and say the POW or the S31? Is your device limited to a 10A load due to the use of the Sonoff Basic?

@pmknowles ditto. Also, other than the 5V input, what is the primary difference between the HW-655 and a Sonoff Basic?

Thanks.

P.S. Still looking for someone who's used a PZEM to monitor a clothes dryer's energy use that handles the higher amperage.

pmknowles, 2018-08-29:
The PZEM-004T can handle up to 100A (measurement only it can’t switch current)

The HW-655 works at 5V but converts the serial to 3.3V. The PZEM-004T needs 5V on the serial connection (unless you modify with a resistor).

arendst, 2018-08-29:
The advantage of the PZEM-004T is also that you can connect a load without using a relay. This can be done with a Pow or S31 but then you have to short the relay.

In my case it's a complete copy of a Pow as I also use the Basic relay. BTW the energy monitoring of the PZEM-004T looks to me better than the old Pow.

meingraham, 2018-08-29:
@pmknowles @arendst 

Apologies for high density factor on my part ;-)

I THINK I get it now...

The Sonoff Basic or HW-655 is powered by appropriate means in order to run the "smarts". The relay on the unit is not (necessarily) used to control the load. In the case of my dryer use case, it would not be used.

The PZEM is powered by 5V and hooked up to the load to be monitored, It sends its "readings" via serial interface to the smart switch (Sonoff or HW-655). The the higher power draw is completely handled by the PZEM.

Did I get it?

Mike

arendst, 2018-08-29:
Yep.

meingraham, 2018-08-29:
@pmknowles 

In your previous post, since you replied via e-mail, the image you'd sent did not make it into this track. Can you post that photo?

Thanks.

Mike

pmknowles, 2018-08-29:
[image]
There you go

meingraham, 2018-08-29:
@pmknowles  What's the part number on your 5V buck?

pmknowles, 2018-08-29:
Here’s the eBay listing
https://www.ebay.co.uk/itm/5Pcs-700mA-AC-DC-5V-3-5W-Buck-Convert-Step-Down-Power-Supply-Module-F-Arduino-M/152339220336?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

pmknowles, 2018-08-30:
Generally comes with the ESP-01S - 3rd option in the drop down here.
https://www.ebay.co.uk/itm/ESP8266-ESP-01-5V-WiFi-Relay-Module-Smart-Home-Remote-Control-Switch-Phone-APP/273151370013?hash=item3f9917231d:m:mp-XSN3bIY0sCytX0KlBEbg

meingraham, 2019-01-02:
My HW-655 based PZEM-004T power monitor for my 240V US standard electric clothes dryer.
[image]
[image]
[image]

amitubale, 2019-01-10:
> My HW-655 based PZEM-004T power monitor for my 240V US standard electric clothes dryer.
> [image]
> [image]
> [image]

Hi - Thanks for the pictures. I am planning to use it for the 220V dryer too.

1. To confirm is this your wiring
---110v-----------------------------------------------------------------Line Dryer pin
---110v------------PZEM Line--------------between CT Coil----------Line Dryer Pin
---Neutral---------PZEM Neutral---------------------------------------Neutral Dryer Pin

2. Since you are using a 5v buck converter and the ESP8266 relay board, you skipped the 1K resistor on the PZEM board on the VCC OUT going to the esp8266 relay board.

3. The Blue enclosure - is that from Lowes or Home Depot ? OR you got it online ?

Thanks

meingraham, 2019-01-10:
@amitubale 

1. Correct except I wouldn't state it as "between CT Coil" but "through CT Coil"
2. I have the 5V and GND out from the buck converter connected to the power input terminals of the HW-655. The  HW-655 has four separate pins for the serial interface. I use the 5V and GND pins on the serial pins as input to the PZEM serial interface along with the Rx/Tx pins. This is the same setup @pmknowles has in his device ([link](https://github.com/arendst/Sonoff-Tasmota/issues/3626#issuecomment-416976400)). I do not believe that a resistor is required.
3. I actually purchased mine at Home Depot. The interesting thing is that when I looked for the part that was on the label of the circuit box, I could only find it on the Lowes website. The Home Depot website didn't have it and I looked for it both with the part number as well as looking at all of the circuit boxes in its online catalog. It's not like they are discontinuing it as there were two or three boxes full of them at the my Home Depot store. My sense is that this is a pretty common part. It worked great to fit everything and the opening flap made it easy to get everything placed inside. BTW, everything is just hot glued in place.

Look at the [Shelly EM](https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/1904226953009993/) coming mid-March. Rated for 120 amps - more than enough. It would be a much easier way to do this. It would consolidate the three separate modules and reduce a lot of the wiring. Had I not already bought the parts and built mine the week before this was announced (!), I would probably go the [Shelly EM](https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/1904226953009993/) route.

Mike

meingraham, 2019-01-10:
@amitubale 

1. Correct except I wouldn't state it as "between CT Coil" but "through CT Coil"
2. I have the 5V and GND out from the buck converter connected to the power input terminals of the HW-655. The  HW-655 has four separate pins for the serial interface. I use the 5V and GND pins on the serial pins as input to the PZEM serial interface along with the Rx/Tx pins. This is the same setup @pmknowles has in his device ([link](https://github.com/arendst/Sonoff-Tasmota/issues/3626#issuecomment-416976400)). I do not believe that a resistor is required.
3. I actually purchased mine at Home Depot. The interesting thing is that when I looked for the part that was on the label of the circuit box, I could only find it on the Lowes website. The Home Depot website didn't have it and I looked for it both with the part number as well as looking at all of the circuit boxes in its online catalog. It's not like they are discontinuing it as there were two or three boxes full of them at the my Home Depot store. My sense is that this is a pretty common part. It worked great to fit everything and the opening flap made it easy to get everything placed inside. BTW, everything is just hot glued in place.
4. You're right, I don't need the relay. In fact, to use the PZEM with the HW-655 means that the relay cannot be used (because the relay is intended to be controlled via serial commands). The HW-655 just made the serial interface simple and didn't require me having to deal with both 5V for the PZEM and 3.3V for the ESP device. The HW-655 took care of the 3.3V regulation for me.
5. I do not believe the PZEM supplies the 5V. I think the PZEM 5V is an input to which you have to provide power. In this post ([link](https://github.com/olehs/PZEM004T/wiki/Basic-Setup)) it has this statement "... the PZEM will return "-1" to everything unless it is connected to power on both sides. That means GND and 5V DC on the data side and 80V-260V AC on the measuring side." I take this to mean you have to provide power to the data side and the AC side is only measuring the AC power metrics. I have not looked at the PZEM circuitry, but this tells me that the AC supply is not used to supply the DC circuitry. I think the reasoning is that doing so would mean that doing that would draw power from the AC thus affecting the actual AC power measurements. In fact, you should take care to where you tap the Dryer's AC Line for the buck converter so that current draw is not registering through the CT Coil. Anyway, long way to say, the PZEM DC it's powered from the HW-655 or directly from the buck converter. But it's just simpler to take it from the HW-655 via the serial interface pins.

As an aside, take a look at [this Shelly device](https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/1904226953009993/) coming mid-March. Rated for 120 amps - more than enough. It would be a much easier way to do this. It would consolidate the three separate modules and reduce a lot of the wiring. Had I not already bought the parts and built mine the week before this was announced (!), I would probably go the [Shelly EM](https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/1904226953009993/) route.

Mike

fragolinux, 2019-01-20:
> The advantage of the PZEM-004T is also that you can connect a load without using a relay. This can be done with a Pow or S31 but then you have to short the relay.
>
> In my case it's a complete copy of [...]

2 questions:

1. so you don't use the relay part at all? Or in case you're limited to a 10A load max of the relay, right?
2. don't you need to turn off serial debugging to use sonoff's rx and tx, or this is due only in case of using them as general gpios?

thanks

ascillato, 2019-01-20:
1- a PZEM is another sensor. See the pictures. You don't need to connect it to any relay if you want.

2- when you select the gpios for the sensor as the tx and rx pins of the esp8266, Tasmota put seriallog to 0 automatically.

pmknowles, 2019-01-20:
The HW-655 uses serial commands to control the relay. I don’t use the relay (but it may still work). Basically the HW-655 simplifies supplying the 3.3V to the ESP8266 and the 5V that the PZEM-004T needs.

Adrian has answered the serial question as I was typing this.

Regards

Phil K

fragolinux, 2019-01-20:
@ascillato @pmknowles  thanks both :)

fragolinux, 2019-01-20:
@ascillato  @pmknowles   thanks both :)
about the relay part, the question was more about @arendst 's setup in that transparent box, it's not clear if the sonoff is just used as a wifi microprocessor to control the pzem output (with no output from the sonoff at all), that just fit in the CT and go away from the other part, or the load passes out from the sonoff AND through the ct... to be clear: the out part is alwasy powered and just metered by the sonoff, or it's only ON in base of the sonoff being ON? In 1st case, you can put loads more than 10A, in the latter you can't, for security

DillonJ, 2019-02-06:
Hey guys - what is the reason for the 5V buck? When you connect a SONOFF basic and use it as designed, it provides its own 3V3 from the mains supply... what is wrong with using it this way? I believe it is not used here for example: https://github.com/arendst/Sonoff-Tasmota/wiki/PZEM004T-Energy-Monitor

meingraham, 2019-02-06:
@ascillato - regarding the use of the HW-655 relay while also using the serial interface to receive energy data from the PZEM... just wanting to get clarification for my notes for the future...

When the Generic module is set up to use the Tx/Rx pins in "PZEM" mode, one can't also issue a POWER or a TOGGLE command; right? One has to control the relay by sending the appropriate string of bytes. And likewise, one has to keep track of the state of the relay oneself, etc. In other words, one cannot use the TASMOTA features for the relay simultaneously with using TASMOTA to capture energy data from the PZEM.

Please confirm.

Thanks.

meingraham, 2019-02-06:
@DillonJ,

The @arendst & @fragolinux devices use the ESP8266 (Sonoff Basic & Wemos D1 respectively) output voltage (3.3V) and use a resistor on the PZEM to account for the voltage the PZEM expects on it's input (5V). Thus they do **not** require a the 5V buck converter.

In the device @pmknowles designed, which I copied, the HW-655 supplies 5V directly to the PZEM - no modification to the PZEM is required. The HW-655 also requires 5V. It handles the voltage regulation onboard to supply the 3.3V the ESP-01 requires. One doesn't need to make any modifications. Thus the buck converter supplies 5V which is what the HW-655 requires and, in turn, the HW-655 supplies 5V to the PZEM.

Mike

ChienTing, 2019-06-10:
Will the new version of PZEM-004T (V3.0)
SonOff can't read the value completely
Does anyone know the reason?
Thanks in advance

homeseer666, 2020-05-10:
A few questions:

My use case is quite simple: I want to know if my 30A Dryer (North american 240V) is running or not; No need to actualy measure the current variation just a plain "is it drawing current or not"

I can assemble Arendst solution above; however I am not sure if I am overcomplicating things unnecessarily and trying to see if I can do this in a non invasive way.

Isn't there a way to simply usa a split-core like below (30A-1V for instance)
And use the GPIO's on a Sonoff basic to check if there is voltage or not on the split core transformer? I know the GPIO's can detect open/closed circuits but can it detect small voltage (0-1V) and/or what could be added in order to do so (small relay or something else?).

https://www.aliexpress.com/item/33021205090.html?spm=a2g0o.productlist.0.0.724b59f7UmtA0w&algo_pvid=fbfb6fdd-3968-400d-9852-b863140b7d9e&algo_expid=fbfb6fdd-3968-400d-9852-b863140b7d9e-3&btsid=0ab6f82415891278831177918e1ebe&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

Also, basic question: can these split cores be put around the full cable (two hot 120V AC wires + neutral) or will they only work if wrapped around a single hot wire? I think it's the latest but want to confirm. Worst case I'll open the back of the dryer and pass it around one hot wire.

Last but not least: if all the above doesn't make sense I'll deploy Arendst solution but I am not 100% sure on the wiring; can you confirm this is accurate (assuming the 1K resistor is added):

[image]
or
[https://www.dropbox.com/t/0Jx4Ot7ZpfzUZnWH](url)
or
[https://www.dropbox.com/s/1be4wcncfn4fgay/2020-05-10_135247f.jpg?dl=0](url)

Thanks, hopefully this will help other people as well.

pmknowles, 2020-05-10:
To be hones the best way is to use a PZEM-004T with the LC Technologies relay board.
ESP8266 works at 3.3V and the PZEM uses 5V as Mike and I did. The PZEM does all the calculations and doesn’t need calibration. The Relay board also cheaply does the level switching in a small form factor and can be powered by a standard 5VDC supply. The CT goes around the single live wire.

Regards

Phil K

https://www.aliexpress.com/item/33021205090.html?spm=a2g0o.productlist.0.0.724b59f7UmtA0w&algo_pvid=fbfb6fdd-3968-400d-9852-b863140b7d9e&algo_expid=fbfb6fdd-3968-400d-9852-b863140b7d9e-3&btsid=0ab6f82415891278831177918e1ebe&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603<https://nam10.safelinks.protection.outlook.com/?url=https%3A%2F%2Fwww.aliexpress.com%2Fitem%2F33021205090.html%3Fspm%3Da2g0o.productlist.0.0.724b59f7UmtA0w%26algo_pvid%3Dfbfb6fdd-3968-400d-9852-b863140b7d9e%26algo_expid%3Dfbfb6fdd-3968-400d-9852-b863140b7d9e-3%26btsid%3D0ab6f82415891278831177918e1ebe%26ws_ab_test%3Dsearchweb0_0%2Csearchweb201602_%2Csearchweb201603&data=02%7C01%7C%7Cac43923a0b734c9d601208d7f50be7fe%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637247303786179845&sdata=RVXMpPqA0GcbpKnZFAjya9LsAQP4MnWIyiDYkxcI0L4%3D&reserved=0>_

[2020-05-10_135247f]<https://nam10.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F56667580%2F81506689-3646a200-92c6-11ea-826e-224f29752914.jpg&data=02%7C01%7C%7Cac43923a0b734c9d601208d7f50be7fe%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637247303786179845&sdata=mQkMCEMYuag2JVNUNegO5UoKYivttkSuRo8Hv%2FTm5dg%3D&reserved=0>

homeseer666, 2020-05-10:
Thanks for the prompt reply.
Ok got it, I am willing to go with that design, 
However the provided links to components either don't work anymore or don't deliver to Canada.
here's what I found can you confirm if accurate?:

https://www.aliexpress.com/item/32811467108.html?spm=a2g0o.productlist.0.0.3c862ec4S6A24q&algo_pvid=95bd8c67-6632-4bbe-bcc0-fbbe217fccd2&algo_expid=95bd8c67-6632-4bbe-bcc0-fbbe217fccd2-42&btsid=0be3746c15891347012658009e4dd8&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

https://www.ebay.ca/itm/1pc-AC-DC-5V-700mA-3-5W-Power-Supply-Buck-Converter-Step-Down-Module-for-Arduino/183413473551?hash=item2ab44b810f:g:570AAOSwZf9bZAi5

https://www.aliexpress.com/item/4000852483683.html?spm=a2g0o.productlist.0.0.3aa67712sJdD5a&algo_pvid=f4e896d0-ed82-471a-91fc-094a7cda2fd0&algo_expid=f4e896d0-ed82-471a-91fc-094a7cda2fd0-6&btsid=0ab6f81615891376606762591e2f95&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

Also some additional questions:

1 - If I decide to go the Sonoff route, can you confirm my design above is OK? My assumptions is that Arendst modification enable us to use a direct serial/gnd/TX/RX connection between the Sonoff unit and PZEM, option 2 being to tap the 5V from the Voltage regulator as shown in the image here : https://tasmota.github.io/docs/PZEM-0XX/  ( 6th image )

2 - Reading above I understand that both the CT+120V feed to the PZEM are required to measure power and not just the CT ?
Therefore I cannot avoid the whole huge Male/Female connector bundle and just tap the Dryer wire directly in a non invasive way, even if I have access to individual wires at the back of the dryer behind the connector, right?

BTW the Shelly 3EM would be interesting but at 99Euros it's a bit steep for a simple ON/OFF Dryer monitor...

pmknowles, 2020-05-10:
Oh heck forgot you have 3 phases to get 240V!. Actually you don’t think need to measure all three phases. You can just put the CT on one phase and use live and neutral (if available) or 1 phase and another. In my version I mounted the whole gubbins in a dual (UK) 13A socket outlet back box. In your case you may be able fit everything do something similar but with a socket outlet.
The links are right (eBay can be slightly quicker than AliExpress). The new PZEM-004T uses Modbus but it’s still useable. An alternative to the buck is a USB charger with the microUSB plug removed.
Another alternative is to do the installation at the fuse box if it’s a discrete circuit.

Regards

https://www.aliexpress.com/item/32811467108.html?spm=a2g0o.productlist.0.0.3c862ec4S6A24q&algo_pvid=95bd8c67-6632-4bbe-bcc0-fbbe217fccd2&algo_expid=95bd8c67-6632-4bbe-bcc0-fbbe217fccd2-42&btsid=0be3746c15891347012658009e4dd8&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603<https://eur05.safelinks.protection.outlook.com/?url=https%3A%2F%2Fwww.aliexpress.com%2Fitem%2F32811467108.html%3Fspm%3Da2g0o.productlist.0.0.3c862ec4S6A24q%26algo_pvid%3D95bd8c67-6632-4bbe-bcc0-fbbe217fccd2%26algo_expid%3D95bd8c67-6632-4bbe-bcc0-fbbe217fccd2-42%26btsid%3D0be3746c15891347012658009e4dd8%26ws_ab_test%3Dsearchweb0_0%2Csearchweb201602_%2Csearchweb201603&data=02%7C01%7C%7C6c45351e3ad04e33dabe08d7f517e432%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637247355261958098&sdata=4DagNK3x0C%2BCVxGqdtqhyeV21EgQCW7rD9bpxr0q7yM%3D&reserved=0>_

https://www.ebay.ca/itm/1pc-AC-DC-5V-700mA-3-5W-Power-Supply-Buck-Converter-Step-Down-Module-for-Arduino/183413473551?hash=item2ab44b810f:g:570AAOSwZf9bZAi5<https://eur05.safelinks.protection.outlook.com/?url=https%3A%2F%2Fwww.ebay.ca%2Fitm%2F1pc-AC-DC-5V-700mA-3-5W-Power-Supply-Buck-Converter-Step-Down-Module-for-Arduino%2F183413473551%3Fhash%3Ditem2ab44b810f%3Ag%3A570AAOSwZf9bZAi5&data=02%7C01%7C%7C6c45351e3ad04e33dabe08d7f517e432%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637247355261968092&sdata=QW0dFN3O%2F52foQRFb%2BLUS5qZbxTU3jKDHhDmZ96K3Gs%3D&reserved=0>

https://www.aliexpress.com/item/4000852483683.html?spm=a2g0o.productlist.0.0.3aa67712sJdD5a&algo_pvid=f4e896d0-ed82-471a-91fc-094a7cda2fd0&algo_expid=f4e896d0-ed82-471a-91fc-094a7cda2fd0-6&btsid=0ab6f81615891376606762591e2f95&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603<https://eur05.safelinks.protection.outlook.com/?url=https%3A%2F%2Fwww.aliexpress.com%2Fitem%2F4000852483683.html%3Fspm%3Da2g0o.productlist.0.0.3aa67712sJdD5a%26algo_pvid%3Df4e896d0-ed82-471a-91fc-094a7cda2fd0%26algo_expid%3Df4e896d0-ed82-471a-91fc-094a7cda2fd0-6%26btsid%3D0ab6f81615891376606762591e2f95%26ws_ab_test%3Dsearchweb0_0%2Csearchweb201602_%2Csearchweb201603&data=02%7C01%7C%7C6c45351e3ad04e33dabe08d7f517e432%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637247355261968092&sdata=V%2F%2FBs7NuGUUXM7g6yeQFhwQiCoQ5KcnWyDF3QwXtm5Y%3D&reserved=0>_

homeseer666, 2020-05-10:
Thanks;  Just to be clear, can I use 120v from any source to feed the PZEM or does it absolutely have to come from the same circuit I am monitoring? If it doesnt I will build a self contained Box with a 120V power cord+CT clamp and will simply hook it to one of the 120V hots inside the dryer. Would that work?

PS: The usb charger would imply an additional external component; would rather use power from the Buck converter.

pmknowles, 2020-05-10:
The input side is essentially a volt and current meter – it is passive. The CT input and AC input are electrically isolated as they are measuring totally different things. I spose you may get some strange readings for apparent power if you’re using different phases.

Regards

Phil K

homeseer666, 2020-05-10:
Ok will do some testing when I get the components and report back. 
If anyone else has experience with North american current/devices please chime in.
Again I don't want to overengineer this, only need to know if the dryer is pulling high current (working) or not (Idle).
Cheers,

CarlosGS, 2020-08-08:
> can you confirm this is accurate (assuming the 1K resistor is added):
> 
> [image]
> 
> Thanks, hopefully this will help other people as well.

# ^^^^^^^ Can confirm that this works perfectly! ^^^^^^

It helps a lot, thank you @homeseer666 :pray: 

**note that the 1k resistor looks like this in new SMD boards:** https://github.com/arendst/Tasmota/issues/6050#issuecomment-510130142

Also leaving this here, just in case others face the same problem:
[image]

homeseer666, 2020-08-10:
Thanks fot the feedback @CarlosGS  !

Glad to hear the visual design helped... and worked =)

I actually just received (part of) my Aliexpress order from 3 months ago....
Will also deploy on my end.
The sonoff basic sounds like a good/simple solution but I also ordered the HW-655+buck converter based on the above thread.  
I havent received the HW-655 yet but will test both options.

Will also test using a separate 120v feed (from the same receptacle as the washer for instance) and just clamp the sensor to one of the phases on the dryer to see if it affects readings. I would prefer this option since I could make the monitoring device self contained with a smaller footprint (vs a bulky 240V bypass box). Easier to switch to another dryer/device if needed.

Will report back, cheers

homeseer666, 2021-01-11:
For anyone interested, I finally had time to tackle this project;
requirements:

- ability to easily move the device around if needed or for testing purposes;
- nothing bulky (which typically doesn't fit behind a dryer) or that would require rewiring.

Therefore I decided to build the device as a stand alone sensing unit;

I tested both designs described above and they work perfectly (HW-655 and Sonoff Basic) thx to @arendst, @pmknowles , @meingraham ) 
However, I had no other use for a HW655/buck converter  laying around so decided to go that route (+ no mod required to the PZEM). Simply used a small project box.

[image]
[image]

Seems like feeding the PZEM with a separate 120v circuit (vs using one "live" from the "sensed" 240v appliance circuit) wasn't tested above so I can confirm that it is working fine.
240v appliances use both sides of the split-phase so not sure if the 120v circuit location makes any difference (feel free to chime in) .

[image]

The unit is just slighly bigger than a deck of cards so you just mount it to the back of the dryer and run the split core around one of the live wires. As I expected, my dryer has a small service panel so installation was a breeze.

[image]
[image]

Since north american houses are on split-phase, shown voltage is 120v which reflects one of the 2 live wires used for the split core; however since we are working with split-phase (and not "2" phases  https://theengineeringmindset.com/120-240v-split-phase-us-can/) , wattage and current seem to be accurate (I doubt my dryer pulls 5500watts. 2800 watts when the element is on makes sense). @meingraham, I guess your results are similar?

I'm not totally familiar with apparent/reactive power/factor, but I'm sure some data post processing should resolve any inaccuracies if the data is relevant to you. Not here as I  just needed to know if device is off/idle/working.

[image]

Hope this helps other North american DIYers.

Cheers,

homeseer666, 2021-04-22:
Just built the Sonoff basic verison; works equally well with less components;
About 20$ for the Sonoff basic + PZEM + Split CT + enclosure and wires.

[image]

dajrekt, 2021-06-21:
Hi Guys, 
I red this thread as I'm going to measure some high voltage devices like cooker, water heater, dryer etc 

I'm going to go with pmknowles project so I would like to connect the clamp in the fuse box. I pretty much understand all PZEM wiring and what does what however not sure of a few things
1. Does the PZEM have to be power up with the same fuse/source of the measure device or can it be any source? I thought PZEM can handle up to 100A when using a clamp for measurement. Is it safe to power it up with the line where the load might be 30-40A? I suppose if it is a paraller connection the load is whatever PZEM takes even from the same fuse (FUSE box) ? If it is serial connection (extension lead) the load is whatever measure device takes? 

2. On the picture of pmknowles what's the cable size of live wire ? is it 1.5? What would be recommendation here?

3. I also notice some of you use sonoff basic for their project and made like an extension lead to control on/off . I'm I right as by doing it this way you can only use it for load up to 10A as Sonoff basic max load is 10A?

Sorry for probably lame questions but I'm not an electrician. I will call for one of to make wiring in the fuse box for sure . Just wanted to know how this should be connected before I call for anyone. 

Please let me know if you need any clarification of my thoughts as I understand it might not be very clear 

Anyway its a great project!
Thanks

pmknowles, 2021-06-21:
Q1. I powered the ESP from the lighting circuit (5A fuse in the UK). I then used an inline 1A fuse from the incoming (100A) breaker to the PZEM.
Q2. The cables are 1.5mm2 but could be any size to match the fuse rating.
Q3. You could use a contactor if you want to switch on/off a higher current remotely – however I’ve replaced 1 PZEM with a Shelly1PM which will switch 16A. I run it in MQTT mode so don’t flash with Tasmota.

Hope that helps

Regards

Phil K

homeseer666, 2021-06-21:
assuming you are in North america and considering all options:

Q1: here not using the same source and so far it works perfectly ; I have 4 units built/deployed (Dryer, Water heater, Hot tub, 240V heater); See design below + above.

[image]

Q3: Correct,  as per design above, the relay is 10A and can be used as you like. 
The sonoff just handles data processing via GPIO and hands it off to Tasmota.

dajrekt, 2021-06-21:
Q1. I powered the ESP from the lighting circuit (5A fuse in the UK). I then used an inline 1A fuse from the incoming (100A) breaker to the PZEM.  

I thought you use the same source for both as per your picture via buck for ESP

Why did you not use lightning circuit for PZEM as well? 
Does it need 1A fuse for PZEM? 
Would you have any diagram?

Q3. You could use a contactor if you want to switch on/off a higher current remotely – however I’ve replaced 1 PZEM with a Shelly1PM which will switch 16A. I run it in MQTT mode so don’t flash with Tasmota. Hope that helps Regards Phil K 

I use Sonoffs Pow 2 which also are 16A , they work great for media switching and measurement . To be honest I don't need to switch off the water heater or the dryer, it was only my curiosity :)

> assuming you are in North america and considering all options:

I'm in Ireland, 230V single phase for domestic If I'm not wrong

> Q1: here not using the same source and so far it works perfectly ; I have 4 units built/deployed (Dryer, Water heater, Hot tub, 240V heater); See design below + above.

Is the measurement accurate? 
I will try it for sure but for some equipment I think will be easier to connect it in the fuse box.
 
Forgive me if I write some rubbish's or asking for obvious things. I know very little in this area however I was able to install my Sonoff pows myself so I'm learning :)

Thanks for you replies guys

pmknowles, 2021-06-22:
I originally used the same source but the 240V to 5V PSU went after a couple of years so I reconfigured so I didn't need to turn off the whole house electricity to change it in future - of course it hasn't failed since. Bear in mind mine is measuring supply to whole house not just an appliance so my use case may be different.
Accuracy is very good less than 1kWh difference to the meter over a month on average (usage is about 400kWh per month). Typically I change the counters to match the meter once a year and it's around 8- 10 kWh out. I used it to send monthly usage to electricity company using a rule in openHAB but now got a SMETS2 smart meter so it will be retired soon.

Regards

Phil K

pmknowles, 2021-06-22:
I used a POW on my 3kW (13A) immersion heater. There was a strange smell one morning and the housing had melted. The POW was still working. It's now in a different housing on the dishwasher. The immersion now has a Shelly1PM with the temperature sensor adaptor.
Regards
Phil K

dajrekt, 2021-06-22:
Thanks guys for your input

I have another question.  Seems I have everything connected and configured but I'm not getting any readings from PZEM. Mine is V3 , someone above mentioned the same. Is there any issue with V3 or is it something wrong with my setup? I checked multiply times how is PZEM connected with hw655 and all looks good to me. 5V-5V , TX-RX, RX-TX , GND- GND. Any idea? ESP shows 0 everywhere . 

Regards

pmknowles, 2021-06-23:
Have you selected PZEM-016 for Rx? Needs to be that for V3 cos RS485.

Regards

Phil K

homeseer666, 2021-06-23:
[image]

pmknowles, 2021-06-23:
I know I had an issue with V3 but I can't remember how I resolved it.

It may have been baud rates or something  but if it comes to me I'll let you know.

Regards

Phil K

[Screenshot_20210622-235302_Office]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F56667580%2F123033095-29dfb480-d3b5-11eb-9e60-e290b7dca905.jpg&data=04%7C01%7C%7Cf466a1bbc7054afab83108d935fa9e6f%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637600172778620180%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=1szVlPIIJlJgI6DY3BFLeqmPFUFf6sW2Hs5xG9F2kdY%3D&reserved=0>

pmknowles, 2021-06-23:
Might have been setting the module address to 1

dajrekt, 2021-06-23:
> Have you selected PZEM-016 for Rx? Needs to be that for V3 cos RS485. Regards Phil K 

I have . but mine is opposite. GPIO1 is RX and GPIO3 is TX and module type is Generic 18 I think.

dajrekt, 2021-06-23:
> Might have been setting the module address to 1 

Do you mean change it to Sonoff basic? 
I haven't try it , thought it doesn't matter

pmknowles, 2021-06-23:
No there’s a command to set the ModBus Address of the PZEM to 1 – it’s in the Power Measurement section of ommands

dajrekt, 2021-06-23:
> ModBus Address of the PZEM to 1

is it this one
ModuleAddress 1 ?

[6 more replies]