arendst/Tasmota #3627: rule with Time#Minute|5 doesn't work in 6.1.1

aberguno, 2018-08-28:
hi, if i use this the trigger is working fine

`23:40:13 CMD: rule 1 on Time#Minute do backlog power toggle endon
23:40:13 MQT: stat/greenhouse/RESULT = {"Rule1":"ON","Once":"OFF","Free":511,"Rules":""}
23:40:20 CMD: rule1 on Time#Minute do backlog power toggle endon
23:40:20 MQT: stat/greenhouse/RESULT = {"Rule1":"ON","Once":"OFF","Free":467,"Rules":"on Time#Minute do backlog power toggle endon"}
23:40:29 CMD: rule1 1
23:40:29 MQT: stat/greenhouse/RESULT = {"Rule1":"ON","Once":"OFF","Free":467,"Rules":"on Time#Minute do backlog power toggle endon"}
23:41:00 RUL: TIME#MINUTE performs "backlog power toggle"
23:41:00 MQT: stat/greenhouse/RESULT = {"POWER1":"OFF"}
23:41:00 MQT: stat/greenhouse/POWER1 = OFF`

BUT if i try to fire it every 5 minutes, it doesn't work
`23:55:24 CMD: rule1 on Time#Minute|5 do backlog Power1 toggle endo
23:55:24 MQT: stat/greenhouse/RESULT = {"Rule1":"OFF","Once":"OFF","Free":465,"Rules":"on Time#Minute|5 do backlog Power1 toggle endo"}
23:55:31 CMD: rule1 1
23:55:31 MQT: stat/greenhouse/RESULT = {"Rule1":"ON","Once":"OFF","Free":465,"Rules":"on Time#Minute|5 do backlog Power1 toggle endo"}`

ascillato, 2018-08-29:
Hi

You are right. That rule was implemented after v6.1.1

Please, use last development version. 
You can compile it by yourself or download the compiled bin from the link in the readme file. 

Thanks

ascillato, 2018-08-29:
*endo* at the last part of your rule needs to be *endon*

ascillato, 2018-08-29:
Backlog is not needed if you are executing just one command

ascillato, 2018-08-29:
Hi,

Tested in last development version (v6.1.1.13) the rule:

```
rule 1
rule on Time#Minute|5 do Power1 2 endon
```

and works as expected.

_( Please, remember that the `Time#Minute|5` rule will trigger when the actual minutes of the real time can be mathematically divided (integer result) by 5, meaning that will trigger at 11:00, 11:05, 11:10, etc.)_

So, the issue was a rule syntax error.

Hope this helps.

Closing issue.

closed 2018-08-29 by ascillato2

aberguno, 2018-08-29:
yes, the issue was a rule syntax error, thank you very much for the help!!