arendst/Tasmota #3629: explicitly include timezones for timestamps in JSON messages

rgitzel, 2018-08-29:
**Have you look for this feature in other issues and in the wiki?**
Yes. Also in the source.

**Is your feature request related to a problem? Please describe.**
Recently I started feeding `tele` messages into InfluxDb via MqttWarn, and found that my graphs were seven hours behind. "!?!?”

Turns out that InfluxDb expects UTC timestamps; meanwhile Tasmota generates timestamps without timezone, on the assumption they are in local time (Pacific, in my case). So by default InfluxDb considered my data 7 hours old. Fair enough.

I fixed the problem by adding 7 hours worth of seconds to the timestamps with MqttWarn, and my graphs lined up.  It worked, yes, but it’s not portable, and I’d have to do that calculation again in other situations.  

**Describe the solution you'd like**
I would like Tasmota to emit (or be able to emit) messages with unambiguous timestamps.

**Describe alternatives you've considered**
I’ve coded up two initial solutions (and note these are PRs in my fork only):
* [Append the numeric timezone to the existing string timestamp](https://github.com/rgitzel/Sonoff-Tasmota/pull/2/files)
  * e.g. `"Time":"2018-08-24T07:17:52-07:00”`
  * Pros:  
     * both human- and machine-readable
     * meets ISO8601 format
  * Cons:  
    * slightly backwards-incompatible (a client might depend on timestamps to be an exact length?)

* [Allow possibility of a numeric Unix timestamp, using a configuration option](https://github.com/rgitzel/Sonoff-Tasmota/pull/1/files)
  * e.g. `"Time":1535119777`
  * I used a `#define` flag in my example, but it could be done other ways
  * _Pros:_ 
    * 'Seconds-since-the-epoch' is a commonly-used format for machines
    * it’s backwards-compatible, as messages would have a numeric timestamp only if you *choose* it
  * _Cons:_
    * not human-readable at all (not for most humans at least)
    * Not compatible with existing downstream client apps like Home Assistant (for now)
      * You would be restricted only to clients aware of the different format
      * But again only if you *choose* this option

(I tried a third option, adding an extra “TimeInSeconds” numeric field alongside the “Time” string field, but that felt just plain wrong.)

Using the ISO8601 strings feels better to me, if only because I can’t read Unix timestamps.  But backwards-compatibility might trump that?

Then again, appending the timezone _could_ be a configuration choice, too.

**Additional context**
I’d be happy to contribute a PR to the main repo, but I wanted to find out first what other people think.

Thanks!

arendst, 2018-08-29:
Or just change the time on your devices to UTC...

pmknowles, 2018-08-29:
Couple of questions.
Have you set your NTP server up in Tasmota?
Have you also set your TimeSTD and TimeDST?
What time does it say in console?
As Theo says, if you're not using the time for anything else just set your ntpserver to pool.ntp.org and set the TimeZone to 0

ascillato, 2018-08-29:
Hi,

In order to have this option for better InfluxDB integration, please make a PR for the case `"Time":"2018-08-24T07:17:52-07:00”` but make it optional with a #define key in user_config.h file, and turned off by default. For people starting in home automations that complete TimeStamp will be confusing. This Feature will be used only for advanced users/integrators.

rgitzel, 2018-09-05:
FYI, my time dried up suddenly, but I've now got a PR just about ready. I ran some tests this morning, will hope to have it posted today.

ascillato2, 2018-09-05:
Thread continues on PR #3711

closed 2018-09-05 by ascillato2