arendst/Tasmota #3632: putting sonoffs in the ceiling void

jpadie, 2018-08-29:
Does anyone know of a pointer to the regs in europe (or in the uK or France) for putting things in ceiling voids or wall spaces?  and whether sonoffs have gone through UL for such use?  

I'm wondering generally whether they are certified for that and secondly, if they are, whether they need a fire retardant bag for ceiling void use (like a spot light for example).  

not asking for anyone to advise definitively - but a pointer to official documentation is very welcome.

pmknowles, 2018-08-29:
They now have a proper CE mark (for what it's worth). In theory they should be no worse than a lighting transformer in a roof space which doesn't need fireproofing. I would fix it to a joist and not leave it 'lying around' but there is no official documentation in the UK for housing use.

curzon01, 2018-08-30:
@jpadie 

The Tasmota [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) (**Devices - Additional Info** - right side) contains sections main Sonoff devices to ITEAD Wiki

e. g. Tasmota section for [Sonoff Basic](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic) contains a link to [Sonoff ITEAD Wiki](https://www.itead.cc/wiki/Sonoff)

Go down to Download section there are links to certifitcation documents.

Keep searching in ITEAD Wiki for other documents or use the links in Tasmota Wiki

jpadie, 2018-08-30:
Thank you.  I think the relevant standard is EN62368 but I can't say for certain whether that means it can be installed in a ceiling void!

jpadie, 2018-08-30:
@pmknowles  thanks for that information.  I've opened up a few sonoffs of various types (about 20) and found the high voltage traces quite poorly soldered; so am a little concerned about them.  In general I only use them for switching very low wattage loads though.

mike2nl, 2018-08-30:
General:
In any case is the humidity important. 
On 100% you can get a short cicuit and fire van be the following issue.

Normally the EN62368-1 or IEC62368-1 has some ranges about the humidity.
Test case based on that norm is: 
- RH = (93±3)%
- Temparture = (40±2)°C
- Duration = min of 60h (depending on the definition) but 120h are EN norm, 
then you can be sure it's working and the 'water' has reached all very small spaces. 

Standard for the most electronic devices is:  
- non-condensing (NOT 100%)
- means  20% - 90% RH

Last but not least:
- You can't get any information about these CE norms on google.
- Everybody has to pay for it and they are very pricy. So you will never get an open copy.
- Companies have they as standard so far they build things are defined in that Norm.

Where i know that from:
- I work in the European Norm Community Group now for 18 years

ascillato2, 2018-09-03:
@jpadie 

Hi,

If you feel your question has been answered, please close this issue. Thanks

closed 2018-09-04 by ascillato2