arendst/Tasmota #3634: JSON Field Bug making Tasmota fail with Yeti integration

luisfpinto, 2018-08-29:
Hello,

Right now I'm not able to create a PR to fix this issue so I'm adding this here to see if anyone can fix it.
A Yeti user reported us that Yeti was finding all devices but one. We asked him to provide the JSON response from both devices and we discovered that: 

In the Status PRM there is a double quote " in the OtaUrl field that shouldn't be there
`"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin"","RestartReason":"Software/System restart","Uptime":"0T00:04:53",`

I think it must be an issue related to that specific type of devices as the other information the user provided was: (As you can see this doesn't have the double quote " as the other one has.
`"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:00:36",`

Both have the same Firmware version

I hope this helps!

arendst, 2018-08-29:
I guess the user once entered the otaurl with a quote at the end.

ascillato, 2018-08-29:
Hi,

Confirmed. If you put `OtaUrl http://sonoff.maddox.co.uk/tasmota/sonoff.bin"` you will have that error in Yeti.

Ask that user to set again the otaurl value by commands in the console or to reset it to default using `OtaUrl 1`

closed 2018-08-29 by ascillato2