arendst/Tasmota #3636: changed topic but old one still present in some console output

Kedryn, 2018-08-29:
**Describe the bug**
On a Tasmotized nodemcu (5.2.x then upgraded to 6.1.1) i had a full topic as "homeassistant/%topic%/"
now i reverted back to "%prefix%/%topic%/".
After a reboot i get this in console:

15:26:12 MQT: tele/home_sensors/INFO1 = {"Module":"Generic","Version":"6.1.1","FallbackTopic":"Node-Casa","GroupTopic":"sonoffs"}
15:26:12 MQT: tele/home_sensors/INFO2 = {"WebServerMode":"Admin","Hostname":"Node-Salotto","IPAddress":"192.168.0.210"}
15:26:12 MQT: tele/home_sensors/INFO3 = {"RestartReason":"Software/System restart"}
15:26:12 MQT: homeassistant/light/home_sensors_1/config =  (salvato)
15:26:12 MQT: homeassistant/switch/home_sensors_1/config =  (salvato)
15:26:12 MQT: homeassistant/light/home_sensors_2/config =  (salvato)
15:26:12 MQT: homeassistant/switch/home_sensors_2/config =  (salvato)
15:26:12 MQT: homeassistant/light/home_sensors_3/config =  (salvato)
15:26:12 MQT: homeassistant/switch/home_sensors_3/config =  (salvato)
15:26:12 MQT: homeassistant/light/home_sensors_4/config =  (salvato)
15:26:12 MQT: homeassistant/switch/home_sensors_4/config =  (salvato)
15:26:12 MQT: homeassistant/light/home_sensors_5/config =  (salvato)
15:26:12 MQT: homeassistant/switch/home_sensors_5/config =  (salvato)
15:26:12 MQT: homeassistant/light/home_sensors_6/config =  (salvato)
15:26:12 MQT: homeassistant/switch/home_sensors_6/config =  (salvato)
15:26:12 MQT: homeassistant/light/home_sensors_7/config =  (salvato)
15:26:12 MQT: homeassistant/switch/home_sensors_7/config =  (salvato)
15:26:12 MQT: homeassistant/light/home_sensors_8/config =  (salvato)
15:26:12 MQT: homeassistant/switch/home_sensors_8/config =  (salvato)

"(salvato)" means "Saved" in italian.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : NodeMcu board with tasmota 6.1.1
- [x] _Provide the output of command_``status 0`` : 
```
15:27:45 MQT: stat/home_sensors/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"home_sensors","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:27:45 MQT: stat/home_sensors/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.0.48:9541/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:39","StartupUTC":"2018-08-29T14:26:06","Sleep":0,"BootCount":85,"SaveCount":317439,"SaveAddress":"3FA000"}}
15:27:45 MQT: stat/home_sensors/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-27T12:09:56","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
15:27:45 MQT: stat/home_sensors/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.0.48","LogPort":514,"SSId":["DarkAngel_2.Net","CostaParchi.priv"],"TelePeriod":300,"SetOption":["000A0009","55818000"]}}
15:27:45 MQT: stat/home_sensors/STATUS4 = {"StatusMEM":{"ProgramSize":545,"Free":2524,"Heap":13,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00000410","0FC0E794","00000000","23B6179E","00000000"]}}
15:27:45 MQT: stat/home_sensors/STATUS5 = {"StatusNET":{"Hostname":"Node-Salotto","IPAddress":"192.168.0.210","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.254","Mac":"5C:CF:7F:6E:35:A7","Webserver":2,"WifiConfig":3}}
15:27:45 MQT: stat/home_sensors/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.48","MqttPort":1883,"MqttClientMask":"Node-Casa","MqttClient":"Node-Casa","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:27:45 MQT: stat/home_sensors/STATUS7 = {"StatusTIM":{"UTC":"Wed Aug 29 14:27:45 2018","Local":"Wed Aug 29 15:27:45 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:02","Sunset":"19:39"}}
15:27:45 MQT: stat/home_sensors/STATUS10 = {"StatusSNS":{"Time":"2018-08-29T15:27:45","Switch2":"ON","BME280":{"Temperature":24.4,"Humidity":54.6,"Pressure":1014.9},"BH1750":{"Illuminance":47},"TempUnit":"C"}}
15:27:46 MQT: stat/home_sensors/STATUS11 = {"StatusSTS":{"Time":"2018-08-29T15:27:45","Uptime":"0T00:01:39","Vcc":3.025,"Wifi":{"AP":1,"SSId":"DarkAngel_2.Net","RSSI":82,"APMac":"C0:25:06:C3:60:4C"}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
I think it was the changing of the topic. There is no "homeassistant" string anywhere in config, but it keeps outputting that.

**Expected behavior**
Not have that log output

arendst, 2018-08-29:
If you only knew what I have to do to get that crappy homeassistant satisfied.

The messages you see are there because you enabled homeassistant (HASS) support. The topic starting with homeassistant are used to let HASS know what the status is of your device. The name is configured in user_config.h and needs to be the same as in the crappy HASS config. As HASS is non-persistent and needs to reboot more often than Windows NT losing it's config on the way it relies on the retained information of the MQTT server. The goal of the message is to let the MQTT server know that non configured switches and lights information needs to be discarded.

A decent Home Automation System would store this information in a nice database and keep it's database in sync with the messages it receives from the Tasmota devices.

So bottomline, it works as designed.

Kedryn, 2018-08-29:
Sorry arendst, but i didnt mention HASS, i mentioned a "string" with "homeassistant" in the fulltopic of tasmota. HASS right now is not calculated in the problem, i'm using the nodemcu via web interface directly.
All my changes in the config of the nodemcu were made via web.

Kedryn, 2018-08-29:
But it keeps appearing at every reboot of the nodemcu, and rebooted HASS keeps complaining about "unable to parse JSON home_sensors_1" with numbers from 1 to 8. Both the console output i mentioned and the HA errors where not present before i changed the full topic. Everything was working. And i changed topic in HA sensor and switches too, so there is no "homeassistant" topic anywhere anymore but in the nodemcu console.

Kedryn, 2018-08-29:
Nor i have those messages in another nodemcu i'm using with nearly same sensors (another room) that didnt has "homeassistant" in topic from beginning.

ascillato, 2018-08-29:
what you are seeing is the homeassistant discovery. just turn it off using `SetOption19 0`

ascillato2, 2018-08-29:
Also clear your MQTT database for retained messages.

Kedryn, 2018-08-30:
Thanks to you replies, i've cleared the MQTT database, i've found that i had logging of MQTT set to debug (thats why i had errors about "_1"), and now it's everyting working fine again (i did not SetOption19 0, because i want autodiscover for now).

I'm closing this, thanks.

closed 2018-08-30 by Kedryn