arendst/Tasmota #3647: Support Wiegand 26 reader

wirelesssolution, 2018-08-31:
It is possible to support RFID reader using Wiegand 26 bits protocal ?

https://www.letscontrolit.com/wiki/index.php/Wiegand-26

Jason2866, 2018-08-31:
Possible maybe ;-) At the moment not supported

ascillato, 2018-08-31:
RFID readers requested at #2915

closed 2018-08-31 by ascillato2

ozett, 2019-01-02:
i think that wiegand-protocol for NFC-reader is not the same as PN532 reader.
i have fingerprints with Wiegand-interfaces, and also NFC-Reader combis.

i would love to see this in tasmota. maybe together with the pn532?

could be easy coded, i guess, it is almost done on espeasy.
maybe some developer has a ❤️  and copy and paste the important bits over to tasmota 😄 

wiegand lines [here on espeasy-source on github](https://github.com/letscontrolit/ESPEasy/blob/f53a9533a4be9a790abf47d54fd2409e4ebe31fe/src/_P008_RFID.ino)

```
event->String1 = formatGpioName_input(F("D0 (Green, 5V)"));
event->String2 = formatGpioName_input(F("D1 (White, 5V)"));
```
get wiegand data simply from interrupt [here on espeasy-source on github](https://github.com/letscontrolit/ESPEasy/blob/f53a9533a4be9a790abf47d54fd2409e4ebe31fe/src/_P008_RFID.ino)

```
*********************************************************************/
void Plugin_008_interrupt1()
/*********************************************************************/
{
  // We've received a 1 bit. (bit 0 = high, bit 1 = low)
  Plugin_008_keyBuffer = Plugin_008_keyBuffer << 1;     // Left shift the number (effectively multiplying by 2)
  Plugin_008_keyBuffer += 1;         // Add the 1 (not necessary for the zeroes)
  Plugin_008_bitCount++;         // Increment the bit count
}

/*********************************************************************/
void Plugin_008_interrupt2()
/*********************************************************************/
{
  // We've received a 0 bit. (bit 0 = low, bit 1 = high)
  Plugin_008_keyBuffer = Plugin_008_keyBuffer << 1;     // Left shift the number (effectively multiplying by 2)
  Plugin_008_bitCount++;           // Increment the bit count
}
```