arendst/Tasmota #3649: Friedland DC55

gadjet, 2018-08-31:
Would it be possible to support the Friedland DC55 Wireless door/Window contacts?
I have some of these but I get no response from the sonoff bridge when these are triggered.

Thanks

curzon01, 2018-08-31:
Don' think so, looks like it uses a [proprietary protocol](https://forum.pimatic.org/topic/531/device-not-recommended-friedland-dc55-compatibility) - see [here](https://cybergibbons.com/category/alarms-2/friedland-response-reverse-engineering/)

gadjet, 2018-08-31:
OK, I used these with RFlink running on an Arduino Mega2560 and a raspberry Pi and I could see and use the data.
I suppose I will have to order some different wireless contacts then :-(

Jason2866, 2018-08-31:
Maybe not impossible. Take a look at Portisch Github site. 
https://github.com/Portisch/RF-Bridge-EFM8BB1
He is the maker of the custom EFM firmware. This Chip is responsible for RF decoding.
Try to sniff with command rfraw 177. Perhaps you get working results!

Jason2866, 2018-08-31:
This sensors work well: KERUI D026
You get DIFFERENT codes for open / close

gadjet, 2018-08-31:
Do you know if the ITEAD DW1 sends out different codes?

gadjet, 2018-08-31:
I've just ordered a couple of Kerui D026 ..... Then I noticed I'd actually ordered Secrui D026s, let's hope they're the same just a different brand ! they are the same size and same functions

Jason2866, 2018-08-31:
DW1 sends only 1 Code -> Open.
I found this one. Like user feedback the send open close too.
https://www.aliexpress.com/item/2pcs-lot-New-Hot-Selling-Wireless-Door-Window-Sensor-Door-Gap-Contact-for-WIFI-GSM-GPRS/32741514329.html?spm=2114.search0204.3.2.74b859d43qTFIq&ws_ab_test=searchweb0_0,searchweb201602_2_10320_10065_10068_5723115_10547_5723715_10548_5723815_5725120_10696_5723215_5723415_10084_5722215_10083_10618_5725020_10304_10307_10820_10821_5990020_10302_5723315_5723615_5724920_5980020_5724120_10843_5722315_5724020_10059_5723015_5722115_5890020_100031_5723915_10319_5724320_10103_5970020_5724220_5722415_5910020-5990020,searchweb201603_54,ppcSwitch_5_ppcChannel&algo_expid=0a0d56ed-1dd4-49e7-8b16-e4e821a8f2eb-0&algo_pvid=0a0d56ed-1dd4-49e7-8b16-e4e821a8f2eb&priceBeautifyAB=0

Jason2866, 2018-08-31:
Please give feedback if Secrui D026s sends command for close too.
In Feature list i can see open battery and tamper. Lets hope the just forgot "close".
But the look very identical even the manual is the same as from Kerui.
Hopefully the copied it perfect 1:1 in ervery detail :-)  The price is promising!

gadjet, 2018-08-31:
No problem I'll feedback my findings and thanks for the help

Jason2866, 2018-08-31:
Your welcome, please close this issue if it solves your problem. Thank you

gadjet, 2018-08-31:
I'll close this issue with the conclusion that the Friedland DC55s are not supported.

closed 2018-08-31 by gadjet

Jason2866, 2018-11-12:
@gadjet 
Does the Secrui D026s send for open close different codes?

gadjet, 2018-11-12:
Yes there are codes for open, close and low battery