arendst/Tasmota #3650: Survey/Best Practice:  Examples of 3d printed cases for Tasmota based projects

Jason2866, 2018-08-31:
Hi together,

i just ordered a 3d printer (cant wait delivery...) **Main reason was i want/need cute cases for my "Things" builded with Tasmota!**

Cases for WemosMinis (and others) with Power Supply, Sensors, Displays, Switches, Buttons, Touch Inputs and many many more (mp3 Player / Loudspeaker)....

It would be great if you share your designs! 

Thx in advance!

meingraham, 2018-08-31:
@Jason2866 where are your designs posted?

ascillato, 2018-08-31:
A very complete website with lots of cases and stuff for 3D printing is https://www.thingiverse.com

Jason2866, 2018-08-31:
@meingraham 
Still no designs. I am a beginner. Printer arrived today :-)

Jason2866, 2018-08-31:
@ascillato Thx for the tipp/link

ascillato, 2018-09-05:
Link added to the wiki

closed 2018-09-19 by Jason2866