arendst/Tasmota #3651: Gui does not show switches and their state

atsitras, 2018-08-31:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

atsitras, 2018-08-31:
I flashed tasmota in a ESP8266 nodemcu and i got access to it from the gui. I have chosen the 18 generic and the system rebooted. Then i connected 3 reed switches to it. 1st on is attached to D0 and Ground and is named as switch1 in Tasmota. 2nd in D1 and Ground and is named as switch2 in Tasmota, 3rd in D2 and Ground  and is named as switch3 in Tasmota. The problem is that in the console i can see the state of each sensor as ON/OFF, but not in the gui. MQTT has been disabled and i expect to see the switches's state in the gui. Someone noticed that problem, please?

Jason2866, 2018-08-31:
Hello @atsitras 
Please, read the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs for restrictions on connections. Thanks
There is NO WebGui for Status of the switches. You can define "virtual" relays on GPIOs you dont need.
This relays will represent the switch status.
This is NO error. Design decision because every SONOFF device has a relais!

Jason2866, 2018-08-31:
Please close this issue if your question is answered
Thx

atsitras, 2018-08-31:
Thanks a lot for your reply. So, to summarize the available GPIOS that i could use are:
D1 = GPIO5
D2 = GPIO4
D6 = GPIO12
D7 = GPIO13
D3 = GPIO0

Please confirm that this is correct.

ascillato, 2018-08-31:
Yes, as explained in the wiki link Jason has provided you.

ascillato, 2018-08-31:
Yes

Jason2866, 2018-08-31:
Nearly, D3=GPIO0 dont use!

D1 = GPIO5
D2 = GPIO4
D6 = GPIO12
D7 = GPIO13
D5 = GPIO14

atsitras, 2018-08-31:
Thanks for the replies

closed 2018-08-31 by atsitras

madduck, 2023-04-30:
This is a duplicate of #105