arendst/Tasmota #3652: Sonoff 4CH R2 - Random on/off toggling

Lodeli, 2018-08-31:
Hi all,
to integrate my first (and only, by now) Sonoff device into my existing "dumb" elecric system, I hacked its hardware buttons soldering wires on their pins and connecting them to four monostable relays.
I can now control lights also from my existing wall push-buttons, using Sonoff device instead of the original step-relays.
This Sonoff is also controlled via MQTT by a HomeAssistant server.

It's all working well but from time to time the POWER1 line (and just this one) is turning on and off with no reason.

Cables length between relays and Sonoff device (signal low-voltage wires) are about 1.5m (4.9ft) each and I'm thinking about some radio noise.
Is it plausible? (but then, why only on POWER1?)
If yes, is there anything I could try other then shorten the signal wires?

I'd also like to exclude some HomeAssistant misconfiguration of mine, but from Console I can't find any CMD: when HA sends a command, all I can see are the STAT messages:

```
stat/sonoff/RESULT = {"POWER1":"OFF"}
MQT: stat/sonoff/POWER1 = OFF
```

There's no difference from when I send command using HA or the physical buttons.
How can I check if the command came from MQTT?

Really thank you!

pmknowles, 2018-08-31:
If you download MQTT fx you can monitor the MQTT messages between the HASS and the device (in both directions). Depending on the frequency you could disable MQTT and then be sure it's one or the other.
By monostable relay do you mean a 'normal' ie non-latching relay? So when you switch the lights on the relay close and stays closed until you switch off? If so, the problem is that the 4ch toggles on the pushbutton and, unless you have done something in the config, they will be configured as Buttons not Switches. If you have pushbuttons which just pulse the relay then something else is going awry.
The pushbuttons take the 3.3V down to 0V so interference isn't likely to be the issue. It could be a poor solder joint or loose wire. Have you used one 0V return wire or separate ones? It might be worthwhile removing the 3 redundant 0V wires to see whether the problem disappears and gets worse. The 0Vs are the lower pins towards the edge of the board.

Frogmore42, 2018-08-31:
It sounds like you have wires in the wall that are many meters long that connect to your switches and also the GPIO on the Sonoff.  Or, is it the original relay now controls the GPIO and the Sonoff relay controls the lights?

Rereading, it sounds like the second one.  That means the GPIO wires are 1.5m.  that should be okay.  I would put a 1K ohm pull-up to help minimize the chance of noise.  This goes between 3V3 and the GPIO.

The log indicates the GPIO was getting triggered.  The most likely thing is noise.

Lodeli, 2018-09-01:
@Frogmore42 yep, the second you said.
I've kept original configuration, with wall pushbuttons carrying main AC line to the relays. This way I had less job to do to integrate with Sonoff device.

I'm waiting to receive a better soldering iron and a new multimeter, in the meantime: this "noise sensitivity" could be related specifically to GPIO (POWER1 button)?
I'm not using the fourth button: what if I use that? Is it less sensible?

Frogmore42, 2018-09-01:
I would not expect one GPIO to be more or less sensitive to noise than the others, but I suppose it is possible.  When you get you soldering iron, you could try swapping the GPIOs and see if the problem moves.

Since these are button inputs and they are normally held high by the pull-up and the button/relay makes them go low, if the error occurred when no button/relay was active your possibly defective soldering doesn't explain it, but noise could.

The stronger pull-up makes it harder for noise to cause the signal to change.  The default internal pull-up on an MCU is typically 20-50K ohm.  It is fairly easy for noise on a long wire to overcome it.  If there is significant variance in the values of the internal pull-ups of your device, it would explain why one input is more sensitive.

Lodeli, 2018-09-03:
I already soldered wires to all the four Sonoff buttons; the fourth button was simply unused.
So my solution by now is to simply use POWER4 instead of POWER1.
This has been working for two days with (finger crossed) no issues.

Anyway I can't be totally sure that I'll never need to use POWER1 so, in that case, I'll surely try soldering a pull-up resistor to get a "stronger" 3.3v signal.
Before closing this issue, just a last question: could be also useful a capacitor between GPIO and GND? What value would you suggest, in case?

Jason2866, 2018-09-03:
A few pf capacitor (ceramic type) is useful too.

Frogmore42, 2018-09-03:
Few pF is not likely to do much.  I would try a 0.01uF or even as high as 0.1uF.

Here is more detail than you probably want.

https://www.digikey.com/en/articles/techzone/2012/apr/protecting-inputs-in-digital-electronics

ascillato2, 2018-09-04:
Hi,

Have you managed to solve your issue?

Lodeli, 2018-09-05:
@Frogmore42 thank you for your link.
Electronic studies are fare away now but it's always a pleasure to have a recap!

@ascillato yep, not using the POWER1 channel, all it's working perfectly.
I'll implement the given solutions once I need to use the "problematic" channel.

Thank you all for the support!

closed 2018-09-05 by Lodeli

danpadure, 2019-11-09:
I know this is closed, but I have exactly the same problems. With Power1
my wires are long (15m)

Frogmore42, 2019-11-09:
Very long wires (15m is considered VERY long) for digital signals are problematic, read that as very difficult to get working correctly.  There is no easy solution to this issue.  Read the article I linked for ideas on some ways to do it.

meingraham, 2019-11-09:
SwitchDebounce or ButtonDebounce?

danpadure, 2019-11-09:
I installed  a 10k pull-up. Did not help
I installed a capacitor. This seem to have done the trick.  Will test to see the stability over days. 
if not, I will install a relay. this should fix the problems. maybe add a diode too.

Lodeli, 2019-11-11:
Since you guys are still here, update: after having installed a capacitor to get rid of the noise on the Sonoff button, it went well for some months. Waaay more than my previous setups.

Anyway, after this goldilocks timeframe, it started again with ghost on/off behavior.
Just a remind: I had 1 meter or less of cable connected to a direct relay that was simply closing the contacts of the Sonoff button, acting as a "decoupling proxy" for my wall 220V push buttons.

I've given up an switched to Shelly 1 + step-relay.
To date, more reliable in less space.

danpadure, 2019-11-11:
update: 
- after a few hours CH4 started to get random on/offs. Installed a capacitor there too. ok for now
- this morning CH1 does not switch at all using the wall buttons.  does switch using OH.  Will investigate more

I will install relays, maybe this could help with the issues....

rbrown67, 2020-04-27:
> update:
> 
> * after a few hours CH4 started to get random on/offs. Installed a capacitor there too. ok for now
> * this morning CH1 does not switch at all using the wall buttons.  does switch using OH.  Will investigate more
> 
> I will install relays, maybe this could help with the issues....

I have the same problem. What size Cap and where did you place it?  Thanks in advance.

pmknowles, 2020-04-27:
There's a description here
https://tasmota.github.io/docs/Expanding-Tasmota/#electrical-considerations
small pF capacitor between the GPIO and 0V

danpadure, 2020-04-27:
I ended up using relays.
caps helped for a while, then started toggling again

pmknowles, 2020-04-27:
With long cable runs it’s the only solution really (plus allows existing switches to be used without worrying about mains and low voltage getting connected).

rbrown67, 2020-04-27:
My input leads are only 12” long. Hopefully the cap will work.

pmknowles, 2020-04-27:
With 12” you should only need a pullup resistor.

rbrown67, 2020-04-27:
I’ll start with the pull up. Thanks for the advice!