arendst/Tasmota #3653: Switch a light with HC-SR501 but only at night

Kaimane, 2018-08-31:
I connected the HC-SR501 to GPIO14 and the light is switched on (and off) if the sensor detected a movement. Is there any solution to do that only at night? I thought about adding a TSL2561 but I found no possible way to bring the two sensors into dependence.
Next idea was using the variables %sunrise% and %sunset% ... If sunrise is true disable switching the light. If sunset is true enable switching the light.
But how to set this rules? Would it be possible?

Jason2866, 2018-08-31:
Take a look in wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#examples

ascillato, 2018-08-31:
Hi,

If you feel that your question has been answered, please remember to close this issue. Thanks

closed 2018-09-01 by ascillato2