arendst/Tasmota #3654: special character "$" and/or "*" dont't work in Wifi password entered via Webgui

Karl-opec, 2018-08-31:
**Describe the bug**
See Title. when these characters are used for Wifi password via user_config.h they work

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ x] _Searched the problem in issues and in the wiki_
- [ x] _Hardware used_ :  sonoff basic

**To Reproduce**
Enter a wifi password with these Character the connecting to the AP fails with "invalid Password"
on the serial console

**Expected behavior**
it should be possible to use these characters in the Webgui

Frogmore42, 2018-08-31:
Probably best to avoid those characters, unless you are a tester for Wi-Fi compliance 😀.

ascillato, 2018-08-31:
Yes, sorry. Those characters are not supported.

ascillato2, 2018-09-03:
Hi, sorry. Those are not supported by now.

arendst, 2018-09-23:
Investigating.

ascillato, 2018-09-23:
Hi, @arendst, I was thinking to make Tasmota not to show the * in the password fields on the UI and leave it blank. So, if the user add something there, the password will change. If nothing is added, the password will be left unchanged.

What do you think?

ascillato, 2018-09-23:
The $ symbol works fine. The issue is only with *

arendst, 2018-09-23:
How do you want to let the user add a blank password?

Currently an empty field returns a blank password. If there are no asterixes it will show the default text of "APx Password".

ascillato, 2018-09-23:
Yes, you are right.

We can leave as it is right now and specify that the symbol * is not supported in a password

arendst, 2018-09-23:
I'm afraid it has to be that way. I made some changes to the debug code to let me see the passwords on request.

If you enter a password in the gui between quotes like "I$Know*It" the password is recognized as I$Know*It so that way it works as expected.

It's probably an anomaly with HTML input fields with type password as the normal console input field does recognize the command ``password I$Know*It`` correctly...

closed 2018-10-11 by ascillato2

localhost61, 2018-12-05:
> If you enter a password in the gui between quotes like "I$Know*It" the password is recognized as I$Know*It so that way it works as expected.
Good to learn there is a workaround but I unfortunately it didn't help me today...

Actually on 6.3.0.15, using the URL command line, I wanted to normalize the Hostname of my 6 Dual R2 in production as Roller shutters. Something like : `http://192.168.144.125/wi?h=Dual_05&save=`
I took less than 1 minute to do that this way for all of them, but what a carnage !!!
Even if my passphrase was not mentioned in the URL, it was retrieved and as it contains a star character, because of this issue the Sonoffs couldn't reconnect, they were all blinking :-/ (Christmas before time !)
Not to mention I had to reset all the configs, using a stepladder to reach them and pushing the main button during 40s :-( ...
Not to mention that the config backups I had were outdated for half of them (v6.0.0) and throw an exception when restored... 
I chance I was not obliged to unmount them and reflash on USB.

My punishment to have a star in the passphrase, but I guess that entering an erroneous one would result in the same. Isn't it? That make me think that it would be a good practice to fill the 2nd AP settings with those of my smartphone, that I would activate to workaround those situations.