arendst/Tasmota #3655: Latest release completely dead in the water

Ikkeme, 2018-08-31:
I just tried to compile the newest 6.2.0 release of Tasmota using platformio cli.
It seems to compile fine. After uploading the image to the device (Sonoff Basic r2) it is completely dead in the water. No connection to my wifi and it does not respond to any button presses no matter how many.
Bringing up the AP by pressing 4 times is not working either. In the previous release it all worked fine.

As a side note: has anyone ever tried to use "user_config_override.h"? So far I've never been able to make the settings stick not in platformio cli and also not in Arduino IDE 1.8.0. The only way to make it work is changing the settings directly in "user_config.h" but that doesn't work either right now.
(Yeah, I know... adding two seperate issues in one bug report is not done. I apologize for that)

According to the template I'm supposed to produce the result of "status 0"., Unfortunately I don't know how to do this in platformio cli.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X ] _Searched the problem in issues and in the wiki_
- [X ] _Hardware used_ : 
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
Compile and upload

**Expected behavior**
A device that is actually working just as in earlier releases.

arendst, 2018-08-31:
Try the released binaries first pls.

Frogmore42, 2018-08-31:
Since you have used earlier releases, you must have been pretty lucky to have never needed to learn how to use the serial port.  I use TeraTerm to communicate with the serial port that the USB adapter provides.

The wiki has some troubleshooting tips for no wifi.  First thing to try is completely erase the flash and reflash the device.

Yes, MANY people have used user_config.h
In recent releases you have to enable that feature.  This explained in the wiki, numerous other issues, and the code itself.

If you just want to use the software, you should stick to the official releases.  While many people find it easy to compile, flash, configure, and then use the code, if you are not experienced in the art of software development it can be quite challenging.

Ikkeme, 2018-08-31:
Thanks you're pretty fast. Right after posting the issue it occurred to me that it would be probably a good idea to try a binary to rule out any problems with the device itself.
I can confirm that sonoff.bin works fine but with that one I need to change a lot of settings manually in order to customize it (hence that secondary remark about  "user_config_override.h"

@Frogmore42: Yes, I'm aware about erasing and reflashing. You advised me to do so in another thread and it worked pretty well. In this case however it makes no difference.
And I've previously flashed a device with a self-compiled binary of the previous release. It worked well but the settings defined in user_config_override.h didn't stick. Changing the same settings in user_config.h settings did the job. So either I'm doing something wrong with that override or there is really something wrong.
Anyway, for now let's stick to the main problem of the not working self-compiled binary.

arendst, 2018-08-31:
See #3623 and consider windows.

Ikkeme, 2018-08-31:
If that's triue why is it working with the previous release then?

arendst, 2018-08-31:
Because the alignment errors appear randomly. Remember the amount of issues people had with 6.1.1 release files compiled on linux. After I provided the windows compiled ones the issues were closed.

I think the alignment problem may also appear on some flash hardware as most problems are raised on R2 hardware.

Just check your alignment as documented in above issue to rule out your specific issue.

andrethomas, 2018-09-01:
@Ikkeme Can you please assist and test the binary releases at http://thehackbox.org/tasmota/ and let me know if the error persists - They are still compiled on Linux but there are some differences between that and the one you'd get from platformio.

Would appreciate the feedback

Thanks

mkh595, 2018-09-01:
> As a side note: has anyone ever tried to use "user_config_override.h"? So far I've never been able to make the settings stick not in platformio cli and also not in Arduino IDE 1.8.0. The only way to [...]

I am using user_config_override.h from version 5.0 and it works also in 6.2.0. I am using self compiled binaries via Visual Studio Code.

Frogmore42, 2018-09-01:
for user_config_override.h, look here:
https://github.com/arendst/Sonoff-Tasmota/blob/development/platformio.ini

uncomment line 64
`;  -DUSE_CONFIG_OVERRIDE  `

so it looks like this:
`  -DUSE_CONFIG_OVERRIDE  `

Ikkeme, 2018-09-01:
@andrethomas Yes sure, I'm pleased to be able to do something useful for the project.
I've flashed sonoff-1m0.bin (commit 6d640cb). After flashing and a restart the device brings up its AP. It is not reachable. After pressing the button 4 times., The AP is still up and now it is reachable on the standard address of 192.168.4.1. I've not been able to try a self-compile because the development is fast today and the github is already at a newer commit. Do you want me to check the other bins as well?

andrethomas, 2018-09-01:
@Ikkeme Thanks for testing - sounds like it worked so the other bins should be fine. Thanks a lot, appreciated!

andrethomas, 2018-09-01:
@Ikkeme Just read further up in the thread - If you previously had settings saved (i.e. a blank flash with default settings) those settings will persist even if you change user_config.h and recompile and upload.

If you want to force the new settings you need to do a complete erase and then flash, or after flashing perform a Reset Settings from the web-ui - that will revert to whatever was specified in the last uploaded binary's user_config.h

Ikkeme, 2018-09-01:
@andrethomas Yes, I've erased the device first before doing this test in order to have a reproducible situation. In the meantime the binaries are renewed so I've tried a self-complie using the current files in the github (commit 52af61b). Result: failure. Which is more or less as expected given the observations in this thread. After that I've erased the device again and flashed it again with sonoff-1m0.bin (commit  52af61b). Result just as before: Success.
It looks like you're after something.

Thanks for the additional information that the settings are persistent I assumed they were erased after a reflash. Now I know they're not reverted to default settings I can safely upgrade my devices with your precompiled binaries without the need to change the settings over and over again. My devices are not involved in mission critical tasks so I'm not that worried about eventual problems with the precompiled binaries.

If I can help with some more testing  please let me know.

B.T.W.: I'm using Ubuntu 18.04.1 64-bit.

andrethomas, 2018-09-01:
@Ikkeme It is likely due to how platformio is compiling the code as mentioned earlier by @arendst but the fact that the development codebase binaries are working for you is a good sign to confirm that it is not necessarily Linux related but more related to how platformio is linking the binaries. The binaries on the link I provided is built on the live codebase at the top of each hour if there were new commits/merges in the hour prior so you're more than welcome to update from there if you want to stay in line with the development codebase until you figure out how to get your end working.

What was your reason for wanting to compile from scratch?

Ikkeme, 2018-09-02:
@andrethomas Is is also not entirely a platformio related issue. Compiling things in Arduino IDE also results in failure. I know that platformio uses Arduino binaries so apparently there goes something wrong, probably the linking process as you've mentioned.
Just for my curiosity: what did you do to correct the linking problems?

The reason for wanting to compile from scratch is that I'm a tweaking/tinkering type of guy. If something is presented as possible I'm usually going to try it if I see a reason for it. Since the binaries in the github were about a month behind when I started using Tasmota and the development branch contained some features I liked I decided to try a compile. Interestingly that compile succeeded. I believe it was 6.1.14 if I recall correctly.
Furthermore I assumed that a firmware upload would erase the settings and since I own multiple Sonoff devices I was planning to invent an automated method  to compile with the settings in place for each individual device. Now you've told me that the settings are not erased and you provide precompiled binaries of the development branch there's in fact no particular reason for me to compile things myself anymore. The backlog feature makes it relatively easy to import the correct settings and I only have to do it once for each new device.
I'm only going to update the firmware if development contains new features (or bugfixes) I particularly like.

And I see there's still an "awaiting for feedback" tag on this thread. Am I supposed to provide any particular information?

andrethomas, 2018-09-02:
So far all I can confirm is that linking is done in a different order when comparing windows vs linux binaries.

The awaiting feedback is just a tag the maintainers of the firmware adds to your issue insofar that they are awaiting feedback on anything further required.

If you are happy with what you have then you can close the issue - if you want to continue trying to get your development environment working I guess you can leave it open and continue asking questions or make proposals on solutions.

The alignment issue appears to be pretty random and I've looked at the linker scripts in depth and I cannot see fault with them - best guess is some of the compiler warnings need to be resolved to ultimately resolve this issue.

The main observation so far is that compiler and object linking binaries provided by expressif do not behave the same on Windows as they do on Linux. This may or may not have to do with the build environment used when they created the compiler (and other build utilities) for a particular build platform.

Ikkeme, 2018-09-03:
@andrethomas Thanks, unfortunately I've some disappointing news. Last night I've tried to to upgrade a device that was previously flashed with the official 6.1.1 precompiled binary from Github with the precompiled binary that was current yesterday evening on your site. The result is now that the upgrade apparently worked but the device is dead just as described in the first post. I believe the only way to revive it is completely erasing it and starting from scratch just as i did before. So now there's sort of a problem I quess. Apparently it is possible to generate from the same source code different series of binaries that are fully functional yet not compatible and therefore not interchangeable. My guess is that the alignment is still wrong and that the settings are stored in a different location.

andrethomas, 2018-09-03:
@Ikkeme Can you share what device you're configuring it as and which sensors you are using?

andrethomas, 2018-09-03:
Also, if you can reproduce by erase and flash working .bin file give us the output of status 0 before you start the configuration and then again another status 0  output after you have configured the device

ascillato2, 2018-09-04:
Hi,

@Ikkeme 

Have you managed to solve your issue?

Ikkeme, 2018-09-04:
@andrethomas I don't have changed anything in the screen you've shown. I don't have to because the selected module type is 01 Sonoff Basic which is correct. I don't have attached any additional sensors yet.
As I pointed out in the first post I don't know how to issue a status 0 command. Right now I flash the module using esptool and configure it with the backlog feature.
In the meantime I've tried what happens if I upgrade a module that is previously flashed with one of your binaries with a new version of your binaries. This works.

@ascillato2 No, the issue is not solved.

andrethomas, 2018-09-04:
You can give status 0 command from the console on the web page, or using a serial comm utility to give commands directly over serial @ 11500bps

andrethomas, 2018-09-04:
@Ikkeme I think its fair that if you do not have time to explore further at this point that we close this one and you can re-open it when you have time to explore further. There's also the discord channel where it can be discussed in real time when you eventually get the time for it. Thanks

ascillato, 2018-09-04:
Agree with @andrethomas 

**_Questions_** and **_Troubleshooting_** has proven to be faster in the [Tasmota Chat](https://discord.gg/Ks2Kzd4). Questions and answers, and tests are very dynamic there.
However, using **GitHub Issues** for this, is very slow as the follow up is very difficult due to the slow questions and answers. Each time something is added, people have to read all the thread again to see what was all that about.

**GitHub Issues** are better for the follow up of **_Tasmota Bug Reports_** and **_Feature Requests_** because those are mainly just a report. If the templates are being followed, the report is likely to have all the information needed to work on it.

ascillato2, 2018-09-05:
Hi,

When you have time, please, report your findings back. Thanks

closed 2018-09-05 by ascillato2

Ikkeme, 2018-09-06:
@andrethomas Back again :-)
I still don't get what's up with that status 0 request you want me to provide or either you don't understand what I said.
The point is in the situation described the device is unresponsive to about anything. It does not respond to button presses and does not bring up the AP. So how am I suppose to access the webconsole (which is probably not even there given the alignment problems). The only way to access the device in this situation is on the cli using the USB TTL converter. So if you can tell me if it is possible (and how) to issue that command in the cli then I can probably produce the result.

And when I say I'm going to respond probably slower I mean I'm not monitoring this 24/7. But I'm certainly going to help if possible so don't be afraid that I'm going to abandon this issue.

andrethomas, 2018-09-06:
> The only way to access the device in this situation is on the cli using the USB TTL converter

I mean that - connect with a serial monitor @ 115200 8N1 no hardware flow control and you will be able to send text commands to the device and see any data that it sends back.

Ikkeme, 2018-09-07:
Finally, progress. I completely misunderstood what a serial monitor is. This time I used the serial monitor in Arduino IDE and after figuring out that it needs a proper line ending in order to have the command actually send to the device it produced the following output in response to the command status 0:

00:01:41 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:01:41 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:41","StartupUTC":"","Sleep":0,"BootCount":12,"SaveCount":14,"SaveAddress":"F6000"}}
00:01:41 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1.2","BuildDateTime":"2018-09-07T19:13:01","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:01:41 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
00:01:41 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":473,"Free":528,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","240003A0","23B617CE","000002C0"]}}
00:01:41 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3012","IPAddress":"0.0.0.0","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"84:F3:EB:49:6B:C4","Webserver":2,"WifiConfig":2}}
00:01:41 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_496BC4","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:01:41 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:01:41 1970","Local":"Thu Jan 01 00:01:41 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:01:41 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:01:41"}}
00:01:41 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:01:41","Uptime":"0T00:01:41","Vcc":3.161,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"44:D1:FA:2A:1C:35","Channel":1,"RSSI":100}}}

Info to reproduce:
I took the current files in the Github made no change whatsoever and uploaded the compiled result to the device.

Result:
No response to button presses and despite the positive messages in the output no detectable AP.

andrethomas, 2018-09-07:
Ok, that may be due to old configuration.

Since you have console access now issue command:
ssid1 yourwifiaccesspointname
wait for it to reboot then write this command
password1 yourwifipassword

Can it connect then to your access point?

andrethomas, 2018-09-07:
Ok, let me try to reproduce your problem by compiling from source @ https://github.com/arendst/Sonoff-Tasmota/archive/v6.2.1.zip using platformio...

andrethomas, 2018-09-07:
Ok, here goes

I downloaded it and extracted it to

[image]

Then modify platformio.ini to compile for sonoff

[image]

Erased the flash of the device with esp's flash tool

[image]

Clicked verify button inside platformio extension and output produced during compile time

[image]

File produced from compile

[image]

Flashed the .bin file using esp's flash tool

[image]

After reboot, status 0 as follows

```
00:00:15 CMD: status 0
00:00:15 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:15 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:15","StartupUTC":"","Sleep":0,"BootCount":2,"SaveCount":3,"SaveAddress":"F9000"}}
00:00:15 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-07T21:46:50","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:00:15 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
00:00:15 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
00:00:15 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0361","IPAddress":"0.0.0.0","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"60:01:94:03:21:69","Webserver":2,"WifiConfig":5}}
00:00:15 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_032169","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:15 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:15 1970","Local":"Thu Jan 01 00:00:15 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:00:15 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:15"}}
00:00:15 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:15","Uptime":"0T00:00:15","Vcc":2.990,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","RSSI":100,"APMac":"FF:FF:FF:FF:FF:FF"}}}
```

4 button pressses, and serial console output as follows:

```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff35d8
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_032169, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-0361 with IP address 192.168.4.1
```

Then search for wifi on phone and see it here:
[image]

After connecting I get presented with this page
[image]

I enter my wifi details, and output on serial monitor of device as follows:
```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_032169, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff-0361...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on sonoff-0361.local with IP address 192.168.47.239
```

So from this it worked - the only difference is that you need to configure static ip address in your user_config.h but remember that settings are perpetual so you need to perform a complete erase of your flash, set settings in user_config.h to set static ip address, then compile then upload binary.

Aside from the last paragraph above there is no problem that I can see

Ikkeme, 2018-09-08:
I appreciate your hard work on this matter.
Based on your screenshots I assume you've did this using a version of Windows. But hyo

andrethomas, 2018-09-08:
Yes, but I do automated compilation on Linux also it works.

Ikkeme, 2018-09-08:
Since you are apparently a quick reader please let me add the comment as it was supposed to be:

I appreciate your hard work on this matter. Based on your screenshots I assume you did this using a version of Windows. But earlier you've established that there apparently are certain differences between binaries compiled in Windows and those in Linux (https://github.com/arendst/Sonoff-Tasmota/issues/3655#issuecomment-417817795 and https://github.com/arendst/Sonoff-Tasmota/issues/3655#issuecomment-417946070 in this thread). So what happens if you repeat this with Linux?

And in the meantime I've done another test. Now I know how to use the serial monitor I've decided to see if my settings in user_config_override.h are really incorporated in the binary and I can confirm they are. Unfortunately still no connection to my network. The serial monitor shows:

WIF: Connecting to AP1 accesspoint in mode 11N as sonoff3-3012...
WIF: Connect failed as AP cannot be reached

and repeats it every 7 seconds. Then suddenly after one minute it shows:

RSL: tele/sonoff3/STATE = {"Time":"1970-01-01T00:00:59","Uptime":"0T00:01:00","Vcc":3.176,"POWER":"OFF","Wifi":{"AP":1,"SSId":"accesspoint","BSSId":"44:D1:FA:2A:1C:35","Channel":1,"RSSI":100}}

and 4 seconds later it starts to repeat the two errormessages. So apparently for a brief moment it seems to connect then immediately disconnects. The official binaries and the ones provided by you are able to connect when the module is in the exact same position (and after being configured of course).
B.T.W.: The SSID of my accesspoint is indeed accesspoint.

Frogmore42, 2018-09-08:
@Ikkme based on the results, it appears you have a choice.
1. Use the released binaries, since they work.
2. Use Windows to compile your own binaries, since that also works.
3. Work with the PlatformIO team to figure out why sometimes their tool (using tools from Espressif and others) does not produce the same output.

Option #3 is likely to be time consuming and will require the ability to compare binary files and understand what might be happening.  I am not sure how interested people at PlatformIO or Espressif will be in figuring this out.  Most people would require a minimal example that shows the issue, but it is likely the alignment issue only shows up in certain circumstances that are much less likely with simple programs.

This is unlikely to be a Tasmota issue and since there are at least two ways around it, the simple answer is don't do that.  I am not sure what else anyone here can do.

Frogmore42, 2018-09-08:
Also, it is probably Espressif you need to work with, since you said it was an issue using Arduino too.

Ikkeme, 2018-09-08:
Well, I'd rather discuss this with @andrethomas as he's apparently capable of generating binaries on Linux that are consistently working.

@andrethomas: can we probably engage in a more private way of communicating outside this issue tracker?

andrethomas, 2018-09-08:
@Ikkeme 

I agree with @Frogmore42 

Users who entertained your questions above have been more than generous to assist as far as possible and I don't think its viable to continue trying to solve this issue here because it is not a Tasmota issue.