arendst/Tasmota #3656: statetext invert - 6.20 not working

timtimsson, 2018-08-31:
i use my switchbutton inverse - now it work with 6.2.0, but status 0 dosnt work...

MQT: /ESP_02/stat/RESULT = {"StateText1":"1"}
MQT: /ESP_02/stat/RESULT = {"StateText2":"0"}

CMD: status 0
MQT: /ESP_02/stat/**STATUS1** = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoff4096","OtaUrl":"http://10.1.1.......
**only status1**

timtimsson, 2018-08-31:
... i chanched my switchmode - thats ok for me...

arendst, 2018-09-01:
Ah nice one...

To get status 0 output you need to execute command status 1

To get status 1 output you need to execute command status 0

arendst, 2018-09-02:
The issue as noted has been solved by removing the possibility of allowing StateText as input to a command parameter.

- StateText1 is the text used for a device that is turned off
- StateText2 is the text used for a device that is turned on

If a user configures StateText1 as "1" and StateText2 as "0" be prepared to see messages like
```
11:08:46 CMD: power 0
11:08:46 MQT: stat/ring2/POWER = {"POWER":"1"}
11:08:46 MQT: stat/ring2/POWER = 1
11:08:50 CMD: power 1
11:08:50 MQT: stat/ring2/POWER = {"POWER":"0"}
11:08:50 MQT: stat/ring2/POWER = 0
```
for the simple reason that the command ``power 0`` wants to turn off power and user configured the resulting text as "1".

closed 2018-09-03 by ascillato2