arendst/Tasmota #3657: Belkin emulation , Alexa and relay control .

Chipz0r, 2018-08-31:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Hardware used : Wemos D1 mini
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
00:36:35 MQT: stat/xmasc1/STATUS = {"Status":{"Module":18,"FriendlyName":["XmasC1","Sonoff2","Sonoff3"],"Topic":"xmasc1","ButtonTopic":"0","Power":3,"PowerOnState":1,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:36:35 MQT: stat/xmasc1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:28","StartupUTC":"2018-08-31T23:35:07","Sleep":0,"BootCount":190,"SaveCount":1755,"SaveAddress":"FA000"}}
00:36:35 MQT: stat/xmasc1/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-08-31T15:15:32","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:36:35 MQT: stat/xmasc1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["hax0r5","hax0r5"],"TelePeriod":300,"SetOption":["00008009","55A18000","00000001"]}}
00:36:35 MQT: stat/xmasc1/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
00:36:35 MQT: stat/xmasc1/STATUS5 = {"StatusNET":{"Hostname":"xmasc1-2646","IPAddress":"192.168.2.180","Gateway":"192.168.2.1","Subnetmask":"255.255.254.0","DNSServer":"8.8.8.8","Mac":"68:C6:3A:91:8A:56","Webserver":2,"WifiConfig":4}}
00:36:35 MQT: stat/xmasc1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.117","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_918A56","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:36:35 MQT: stat/xmasc1/STATUS7 = {"StatusTIM":{"UTC":"Fri Aug 31 23:36:35 2018","Local":"Sat Sep 01 00:36:35 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:07","Sunset":"19:33"}}
00:36:35 MQT: stat/xmasc1/STATUS10 = {"StatusSNS":{"Time":"2018-09-01T00:36:35"}}
00:36:35 MQT: stat/xmasc1/STATUS11 = {"StatusSTS":{"Time":"2018-09-01T00:36:35","Uptime":"0T00:01:28","Vcc":0.028,"POWER1":"ON","POWER2":"ON","POWER3":"OFF","Wifi":{"AP":1,"SSId":"hax0r5","RSSI":60,"APMac":"00:50:7F:E2:AD:5C"}}}
**(Please, remember to close the issue when the problem has been addressed)**

Hi All

I have a D1 mini with 3 toggle's active, i also have the belkin emulation on so i can connect and control via Alexa.
When i have 1 toggle activated and i command to alexa power off , it says in console " Power1 off " which is correct , then i enable another toggle ( these are relay outputs ) and command alexa to power off , in console it says " Power 2 Off " , and enabling a third toggle , alexa then commands Power 3 and not the previous power .

I'm not sure if this in intentional behavior or is there a way to just bind the belkin wemo emulation to 1 Power command and 1 relay.

I hope i have made sense .

Many thanks

altelch, 2018-09-01:
Hi Chipz0r,

the Belkin Wemo in original has only one relay and the 
protocol/emulation doesn't support more than one relay as there is no 
support for more in the original device. Thats why the emulation can 
handle more, too. If you need more than one relay, use HUE emulation.

Greetings,

     Altelch.

Chipz0r, 2018-09-01:
HI, thanks for your reply.
This is not the problem i am having , i'm not trying to add more relays to the emulation.
It's really hard for me to explain .
I have a board with a wemos d1 and a relay connected to D6/GPIO12 and bind it to relay 1 in the module config and belkin emulation on .
When i command alexa to turn it on , it works perfectly and in the console alexa has turned " power 1 on "
Then i enable another GPIO output on D1/GPIO5 and bind it to  relay 2 to control something different.

When i command alexa to turn the unit off as before she now turns off " Power 2 " which is not the relay as previously described .

And again when i add another GPIO output and bind it to relay 3 , alexa then controls this one and ignores the previous 2.

I hope this makes better understanding .

Many thanks .

Jason2866, 2018-09-01:
Belkin is in no way designed for 2 relais. So unexcpected behaviour....
Use HueEmulation if more as 1 relais connected. It works with ALEXA too.

Chipz0r, 2018-09-01:
Hi.
Let me try and clarify this , i do not want to control more than 1 relay with the belkin emulation , what i am asking is that then you add another toggle to control something else , the relay 1 Power1 on changes to Power2 and means you have to rejig the relay outputs.

What i am asking is it normal for it to change from Power1 to power 2 just on the belkin emulation when adding another toggle .

Thanks

Jason2866, 2018-09-01:
Belkin isnt designed to act with more than one switch / relais.
So i takes one switch (which one?!). If this behaviour is always the same i dont know!
If you want defined results you have to use HueEmulation

altelch, 2018-09-02:
Hi,

the "problem" is here:

ExecuteCommandPower(/devices_present/, POWER_ON, SRC_WEMO);

With every relay you define devices_present is incremented so you always 
use the latest relay. The emulation was designed for a one-device-only 
sonoff-basic. We would need an extra setting to declare which device 
should be used by belkin emulation on multi-relay devices.

Greetings,

Altelch.

arendst, 2018-09-02:
In that case I always want to use the first relay.

If it's up to a user some want to change everything, others just want it to work.

I'll change it right now.

Chipz0r, 2018-09-02:
Thanks for finally getting to the bottom of this.

Keep up the good work .

closed 2018-09-02 by Chipz0r