arendst/Tasmota #3658: Wifi not working, v6.20 on Sonoff Basic

onkelbeh, 2018-09-01:
Tried 6.20 on a Sonoff Basic, Wifi did not connect, but after 7 presses to the button (OTA download) it got connected and an IP assigned.

After undef USE_ENERGY_SENSOR in sonoff_post.h it connected without problems.

ascillato, 2018-09-01:
Hi,

Sorry, I could not reproduce your issue. Please, complete the troubleshooting template in order to help you properly.

jk2020, 2018-09-02:
just updated sonoff basic from 6.1.1 to v6.20 (OTA update).
works fine on my side

onkelbeh, 2018-09-02:
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)

Found a similar one (#3150), resolution: turn off "USE_ENERGY_SENSOR" works.

- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)

Found a similar one: https://groups.google.com/forum/#!topic/sonoffusers/02AZ9hcjxvQ

- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Hardware used :

Sonoff Basic "TH_V1.1 2017-5-5"

As mentioned, this Problem ONLY occurs on Sonoff Basic, i tried 2 of them.
On S20, Touch and ESP8266 Wifi works.
 
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE

00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: Serial active for 3 minutes
00:00:16 CMD: status 0
00:00:16 RSL: stat/sonoff3/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff3"],"Topic":"sonoff3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"Pow
erRetain":0}}
00:00:16 RSL: stat/sonoff3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://tasmota.in.edevau.net/tasmota/sonoff_current.bin","RestartReason":"Power on","Uptime":"0T00:00:16","
StartupUTC":"","Sleep":0,"BootCount":27,"SaveCount":54,"SaveAddress":"F6000"}}
00:00:16 RSL: stat/sonoff3/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-09-02T15:28:00","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:00:16 RSL: stat/sonoff3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"syslog.in.edevau.net","LogPort":514,"SSId":["AREA-N",""],"TelePeriod":300,"SetOption":["00008009","55818000","0
0000000"]}}

00:00:16 RSL: stat/sonoff3/STATUS4 = {"StatusMEM":{"ProgramSize":470,"Free":532,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","240003A1","23B617CE","0000000
0"]}}
00:00:16 RSL: stat/sonoff3/STATUS5 = {"StatusNET":{"Hostname":"sonoff3-3459","IPAddress":"0.0.0.0","Gateway":"172.16.2.129","Subnetmask":"255.255.255.192","DNSServer":"172.16.2.6","Mac":"DC:4F:22:37:AD:83","We
bserver":2,"WifiConfig":1}}
00:00:16 RSL: stat/sonoff3/STATUS6 = {"StatusMQT":{"MqttHost":"mosquitto.in.edevau.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_37AD83","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_S
IZE":1000,"KEEPALIVE":15}}
00:00:16 RSL: stat/sonoff3/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:16 1970","Local":"Thu Jan 01 00:00:16 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Su
nrise":"07:01","Sunset":"15:21"}}
00:00:16 RSL: stat/sonoff3/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:16"}}
00:00:16 RSL: stat/sonoff3/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:16","Uptime":"0T00:00:16","Vcc":3.185,"POWER":"OFF","Wifi":{"AP":1,"SSId":"AREA-N","RSSI":100,"APMac":"FF:FF:FF:FF:FF:FF"}}}
00:03:15 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

**(Please, remember to close the issue when the problem has been addressed)**

onkelbeh, 2018-09-02:
for the sake of completeness: same Basic, same settings, only difference ``USE_ENERGY_SENSOR`` commented out: 
```
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff3-3459.local with IP address 172.16.2.183
00:00:05 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: tele/sonoff3/LWT = Online (retained)
00:00:07 MQT: cmnd/sonoff3/POWER =
00:00:07 MQT: stat/sonoff3/RESULT = {"POWER":"OFF"}
00:00:07 MQT: stat/sonoff3/POWER = OFF
00:00:07 MQT: tele/sonoff3/INFO1 = {"Module":"Sonoff Basic","Version":"6.2.0","FallbackTopic":"DVES_37AD83","GroupTopic":"sonoffs"}

00:00:07 MQT: tele/sonoff3/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff3-3459","IPAddress":"172.16.2.183"}
00:00:07 MQT: tele/sonoff3/INFO3 = {"RestartReason":"Power on"}
00:00:07 MQT: stat/sonoff3/RESULT = {"POWER":"OFF"}
00:00:07 MQT: stat/sonoff3/POWER = OFF
15:02:04 MQT: tele/sonoff3/UPTIME = {"Time":"2018-09-02T15:02:04","Uptime":"0T00:00:09"}
15:02:09 MQT: tele/sonoff3/STATE = {"Time":"2018-09-02T15:02:09","Uptime":"0T00:00:14","Vcc":3.183,"POWER":"OFF","Wifi":{"AP":1,"SSId":"AREA-N","RSSI":100,"APMac":"54:A0:50:5B:20:18"}}
15:02:19 CMD: status 0
15:02:19 MQT: stat/sonoff3/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff3"],"Topic":"sonoff3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"Pow
erRetain":0}}
15:02:19 MQT: stat/sonoff3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://tasmota.in.edevau.net/tasmota/sonoff_current.bin","RestartReason":"Power on","Uptime":"0T00:00:24","
StartupUTC":"2018-09-02T14:01:55","Sleep":0,"BootCount":32,"SaveCount":61,"SaveAddress":"F6000"}}
15:02:19 MQT: stat/sonoff3/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-09-02T16:00:54","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:02:19 MQT: stat/sonoff3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"syslog.in.edevau.net","LogPort":514,"SSId":["AREA-N",""],"TelePeriod":300,"SetOption":["00008009","55818000","0
0000000"]}}
15:02:19 MQT: stat/sonoff3/STATUS4 = {"StatusMEM":{"ProgramSize":462,"Free":540,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","240003A1","23B617CA","0000000
0"]}}
15:02:19 MQT: stat/sonoff3/STATUS5 = {"StatusNET":{"Hostname":"sonoff3-3459","IPAddress":"172.16.2.183","Gateway":"172.16.2.129","Subnetmask":"255.255.255.192","DNSServer":"172.16.2.6","Mac":"DC:4F:22:37:AD:83
","Webserver":2,"WifiConfig":3}}
15:02:19 MQT: stat/sonoff3/STATUS6 = {"StatusMQT":{"MqttHost":"mosquitto.in.edevau.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_37AD83","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_S
IZE":1000,"KEEPALIVE":15}}
15:02:19 MQT: stat/sonoff3/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 02 14:02:19 2018","Local":"Sun Sep 02 15:02:19 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Su
nrise":"05:26","Sunset":"18:48"}}
15:02:19 MQT: stat/sonoff3/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:02:19 MQT: stat/sonoff3/STATUS10 = {"StatusSNS":{"Time":"2018-09-02T15:02:19"}}
15:02:19 MQT: stat/sonoff3/STATUS11 = {"StatusSTS":{"Time":"2018-09-02T15:02:19","Uptime":"0T00:00:24","Vcc":3.185,"POWER":"OFF","Wifi":{"AP":1,"SSId":"AREA-N","RSSI":100,"APMac":"54:A0:50:5B:20:18"}}}

onkelbeh, 2018-09-03:
What's the difference between sonoff-classic.bin and sonoff.bin?

ascillato, 2018-09-03:
Hi, the difference between those are just the features included.

More information at https://github.com/arendst/Sonoff-Tasmota/blob/development/RELEASENOTES.md#available-features-and-sensors

ascillato, 2018-09-03:
Have you tried the precompiled bin from release? May be you are having a bad compilation issue that leads to having wifi issues, please, try to erase all flash, and then flash last release. If you are building by yourself, please use esplib 2.3.0 (if you will use sleep feature) or 2.4.2 lwIP1.4 if you are not going to use sleep.

ascillato2, 2018-09-04:
@onkelbeh 

Hi,

Have you managed to solve your issue?

onkelbeh, 2018-09-04:
Hi, @ascillato or @ascillato2 

No, not yet. Thanks a lot for your patience.

Today i tried some things. First i saw that a lot of code changed in 6.2.0.2, so i did this first.
My self-compiled build works like a charm on 
- S20,
- Dual R2
- MagicHome LED Controllers and my
- NodeMCU's

but **NOT** on the Basics.
I'm wondering how often i can flash this thing :-) 
For completeness, here's his status 0:

```00:00:31 CMD: status 0
00:00:31 RSL: stat/sonoff3/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff3"],"Topic":"sonoff3","ButtonTopic":"0","Power
":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:31 RSL: stat/sonoff3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://tasmota.in.edevau.
net/tasmota/sonoff_current.bin","RestartReason":"Power on","Uptime":"0T00:00:31","StartupUTC":"","Sleep":0,"BootCount":44,"Save
Count":106,"SaveAddress":"F9000"}}
00:00:31 RSL: stat/sonoff3/STATUS2 = {"StatusFWR":{"Version":"6.2.0.2","BuildDateTime":"2018.09.04 19:26:45","Boot":31,"Core":"
2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:00:31 RSL: stat/sonoff3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":1,"LogHost":"syslog.in.edevau.net","LogPor
t":514,"SSId":["AREA-N",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}

00:00:31 RSL: stat/sonoff3/STATUS4 = {"StatusMEM":{"ProgramSize":473,"Free":528,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1
024,"FlashMode":3,"Features":["00000407","0FDAE794","240003A1","23B617CE","00000000"]}}
00:00:31 RSL: stat/sonoff3/STATUS5 = {"StatusNET":{"Hostname":"sonoff3-3459","IPAddress":"0.0.0.0","Gateway":"172.16.2.129","Su
bnetmask":"255.255.255.192","DNSServer":"172.16.2.6","Mac":"DC:4F:22:37:AD:83","Webserver":2,"WifiConfig":3}}
00:00:31 RSL: stat/sonoff3/STATUS6 = {"StatusMQT":{"MqttHost":"mosquitto.in.edevau.net","MqttPort":1883,"MqttClientMask":"DVES_
%06X","MqttClient":"DVES_37AD83","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:31 RSL: stat/sonoff3/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:31 1970","Local":"Thu Jan 01 00:00:31 1970","StartDS
T":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:01","Sunset":"15:21"}}
00:00:31 RSL: stat/sonoff3/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:31"}}
00:00:31 RSL: stat/sonoff3/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:31","Uptime":"0T00:00:31","Vcc":3.179,"POWER":"OFF
","Wifi":{"AP":1,"SSId":"AREA-N","BSSId":"FF:FF:FF:FF:FF:FF","Channel":1,"RSSI":100}}}
00:03:16 APP: starte neu

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3fff3378
~ld

00:00:00 Projekt sonoff Sonoff3 (Topic sonoff3, Fallback DVES_37AD83, GroupTopic sonoffs) Version 6.2.0.2-2_3_0
00:00:00 WIF: verbinden mit AP1 AREA-N in Modus 11N wie sonoff3-3459...
00:00:07 WIF: Verbindung fehlgeschlagen, da AP nicht erreicht werden konnte
00:00:07 WIF: verbinden mit AP1 AREA-N in Modus 11N wie sonoff3-3459...
00:00:14 WIF: Verbindung fehlgeschlagen, da AP nicht erreicht werden konnte
00:00:14 WIF: WifiManager aktiv für 3 Minuten
00:00:14 HTP: Web-Server aktiv bei sonoff3-3459 mit IP-Adresse 192.168.4.1
```

Sorry for testing with the german headers, still wondering why the english MQTT strings have to be different, today i saw they have changed in `language/de-DE.h` again :-(

Next I tried without USE_ENERGY, this helped in 6.2.0 ,now in 6.2.0.2 nevermore, so wifi did NOT connect:

```00:00:14 WIF: Verbindung fehlgeschlagen, da AP nicht erreicht werden konnte
00:00:14 WIF: WifiManager aktiv für 3 Minuten
00:00:14 HTP: Web-Server aktiv bei sonoff3-3459 mit IP-Adresse 192.168.4.1
00:00:17 CMD: status 0
00:00:17 RSL: stat/sonoff3/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff3"],"Topic":"sonoff3","ButtonTopic":"0","Power
":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:17 RSL: stat/sonoff3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://tasmota.in.edevau.
net/tasmota/sonoff_current.bin","RestartReason":"Power on","Uptime":"0T00:00:17","StartupUTC":"","Sleep":0,"BootCount":48,"Save

[... 10 lines cut ...]
%06X","MqttClient":"DVES_37AD83","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:17 RSL: stat/sonoff3/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:17 1970","Local":"Thu Jan 01 00:00:17 1970","StartDS
T":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:01","Sunset":"15:21"}}
00:00:17 RSL: stat/sonoff3/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:17"}}
00:00:17 RSL: stat/sonoff3/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:17","Uptime":"0T00:00:17","Vcc":3.178,"POWER":"OFF
","Wifi":{"AP":1,"SSId":"AREA-N","BSSId":"FF:FF:FF:FF:FF:FF","Channel":1,"RSSI":100}}}
```

The i tried the sonoff-DE binary from the release folder, surprised, it works flawless.
I am compiling on OS X High Sierra 10.13.6 (17G65), PlatformIO 3.6.1a1,
here are the libs:

```
Platform Espressif 8266
--------
Updating espressif8266                   @ 1.5.0          [Fixed]
Updating framework-esp8266-rtos-sdk      @ 1.5.0-beta.5   [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating tool-mkspiffs                   @ 1.102.0        [Up-to-date]
Updating tool-espotapy                   @ 1.0.0          [Up-to-date]
Updating framework-arduinoespressif8266  @ 1.20300.1      [Up-to-date]
```

Think this should be esplib 2.3.0 _(i am new on this, excuse me i don't know exactly where to look for the library versions)_, an yes, i am using sleep 100, exept on the LED controllers. The only thing i have changed in platformio.ini was ```upload_port```. Can't explain why everything works without any problems on the other hardware....

ascillato2, 2018-09-05:
This issue is the same as https://github.com/arendst/Sonoff-Tasmota/issues/3678

Let's continue there.

closed 2018-09-05 by ascillato2