arendst/Tasmota #3660: v6.2.0 Sonoff Pow Energy report is broken

myfingerhurt, 2018-09-01:
**Describe the bug**

TelePeriod is set to 10, but the SENSOR reports are generated randomly, and the Period field sometimes disappeared.

I have just updated from 6.1.1 to 6.2.0 through two steps OTA, and my Grafana reports errors immediately.
- Step 1: Upgrade by web server to minimal 
- Step 2: Upgrade by file upload to full version of sonoff pow, the file is download from the release. 

- [V] _Searched the problem in issues and in the wiki_
- [V] _Hardware used_ :  Sonoff Pow gen1
- [V] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE
14:46:01 CMD: status 0
14:46:01 MQT: stat/sonoff_pow5/STATUS = {"Status":{"Module":6,"FriendlyName":["Sonoff_pow5"],"Topic":"sonoff_pow5","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
14:46:01 MQT: stat/sonoff_pow5/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:45:46","StartupUTC":"2018-09-01T05:00:15","Sleep":0,"BootCount":25,"SaveCount":61,"SaveAddress":"FA000"}}
14:46:01 MQT: stat/sonoff_pow5/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-08-31T15:15:32","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
14:46:01 MQT: stat/sonoff_pow5/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["CensoredWIFI",""],"TelePeriod":10,"SetOption":["00208009","55818000","00000001"]}}
14:46:01 MQT: stat/sonoff_pow5/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
14:46:01 MQT: stat/sonoff_pow5/STATUS5 = {"StatusNET":{"Hostname":"sonoff_pow5-7706","IPAddress":"192.168.9.232","Gateway":"192.168.9.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.9.39","Mac":"2C:3A:E8:3B:3E:1A","Webserver":2,"WifiConfig":4}}
14:46:01 MQT: stat/sonoff_pow5/STATUS6 = {"StatusMQT":{"MqttHost":"xxx.130.127.xxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3B3E1A","MqttUser":"Censored","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:46:02 MQT: stat/sonoff_pow5/STATUS7 = {"StatusTIM":{"UTC":"Sat Sep 01 05:46:01 2018","Local":"Sat Sep 01 14:46:01 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":9,"Sunrise":"14:07","Sunset":"03:33"}}
14:46:02 MQT: stat/sonoff_pow5/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:46:02 MQT: stat/sonoff_pow5/STATUS10 = {"StatusSNS":{"Time":"2018-09-01T14:46:02","ENERGY":{"Total":0.004,"Yesterday":0.150,"Today":0.004,"Period":0,"Power":6,"Factor":0.45,"Voltage":98,"Current":0.132}}}
14:46:02 MQT: stat/sonoff_pow5/STATUS11 = {"StatusSTS":{"Time":"2018-09-01T14:46:02","Uptime":"0T00:45:47","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"HYPERSPEED","RSSI":100,"APMac":"E0:3F:49:93:DB:00"}}}
```

**To Reproduce**
Just sit and wait to check out the Console.
```
14:34:50 CMD: TelePeriod 10
14:34:50 MQT: stat/sonoff_pow5/RESULT = {"TelePeriod":"10"}
14:34:51 MQT: tele/sonoff_pow5/STATE = {"Time":"2018-09-01T14:34:51","Uptime":"0T00:34:36","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"HYPERSPEED","RSSI":100,"APMac":"E0:3F:49:93:DB:00"}}
14:34:51 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:34:51","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Period":0,"Power":7,"Factor":0.47,"Voltage":99,"Current":0.154}}
14:34:52 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:34:52","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Power":4,"Factor":0.26,"Voltage":99,"Current":0.154}}
14:34:56 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:34:56","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Power":7,"Factor":0.45,"Voltage":99,"Current":0.158}}
14:35:01 MQT: tele/sonoff_pow5/STATE = {"Time":"2018-09-01T14:35:01","Uptime":"0T00:34:46","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"HYPERSPEED","RSSI":100,"APMac":"E0:3F:49:93:DB:00"}}
14:35:01 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:35:01","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Period":0,"Power":6,"Factor":0.37,"Voltage":99,"Current":0.166}}
14:35:11 MQT: tele/sonoff_pow5/STATE = {"Time":"2018-09-01T14:35:11","Uptime":"0T00:34:56","Vcc":3.178,"POWER":"ON","Wifi":{"AP":1,"SSId":"HYPERSPEED","RSSI":100,"APMac":"E0:3F:49:93:DB:00"}}
14:35:11 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:35:11","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Period":0,"Power":7,"Factor":0.51,"Voltage":99,"Current":0.141}}
14:35:20 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:35:20","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Power":6,"Factor":0.36,"Voltage":99,"Current":0.161}}
14:35:21 MQT: tele/sonoff_pow5/STATE = {"Time":"2018-09-01T14:35:21","Uptime":"0T00:35:06","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"HYPERSPEED","RSSI":100,"APMac":"E0:3F:49:93:DB:00"}}
14:35:21 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:35:21","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Period":0,"Power":7,"Factor":0.36,"Voltage":99,"Current":0.188}}
14:35:31 MQT: tele/sonoff_pow5/STATE = {"Time":"2018-09-01T14:35:31","Uptime":"0T00:35:16","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"HYPERSPEED","RSSI":100,"APMac":"E0:3F:49:93:DB:00"}}
14:35:31 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:35:31","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Period":0,"Power":6,"Factor":0.56,"Voltage":99,"Current":0.106}}
14:35:41 MQT: tele/sonoff_pow5/STATE = {"Time":"2018-09-01T14:35:41","Uptime":"0T00:35:26","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"HYPERSPEED","RSSI":100,"APMac":"E0:3F:49:93:DB:00"}}
14:35:41 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:35:41","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Period":0,"Power":5,"Factor":0.41,"Voltage":99,"Current":0.129}}
14:35:48 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:35:48","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Power":8,"Factor":0.52,"Voltage":99,"Current":0.149}}
14:35:51 MQT: tele/sonoff_pow5/STATE = {"Time":"2018-09-01T14:35:51","Uptime":"0T00:35:36","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"HYPERSPEED","RSSI":100,"APMac":"E0:3F:49:93:DB:00"}}
14:35:51 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:35:51","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Period":0,"Power":6,"Factor":0.34,"Voltage":99,"Current":0.164}}
14:36:01 MQT: tele/sonoff_pow5/STATE = {"Time":"2018-09-01T14:36:01","Uptime":"0T00:35:46","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"HYPERSPEED","RSSI":100,"APMac":"E0:3F:49:93:DB:00"}}
14:36:01 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:36:01","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Period":0,"Power":6,"Factor":0.34,"Voltage":99,"Current":0.175}}
14:36:03 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:36:03","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Power":11,"Factor":0.72,"Voltage":99,"Current":0.150}}
14:36:07 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:36:07","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Power":5,"Factor":0.29,"Voltage":99,"Current":0.163}}
14:36:11 MQT: tele/sonoff_pow5/STATE = {"Time":"2018-09-01T14:36:11","Uptime":"0T00:35:56","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"HYPERSPEED","RSSI":100,"APMac":"E0:3F:49:93:DB:00"}}
14:36:11 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:36:11","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Period":0,"Power":7,"Factor":0.43,"Voltage":99,"Current":0.154}}
14:36:16 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:36:16","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Power":13,"Factor":0.93,"Voltage":99,"Current":0.144}}
14:36:18 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:36:18","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Power":6,"Factor":0.41,"Voltage":99,"Current":0.144}}
14:36:21 MQT: tele/sonoff_pow5/STATE = {"Time":"2018-09-01T14:36:21","Uptime":"0T00:36:06","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"HYPERSPEED","RSSI":100,"APMac":"E0:3F:49:93:DB:00"}}
14:36:21 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:36:21","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Period":0,"Power":7,"Factor":0.48,"Voltage":99,"Current":0.146}}
```

[... cut ...]
Do you see the second line is shorter and Period field is gone?
Do you notice the reported intelvals is also violated, should be 10 seconds.
```
14:34:51 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:34:51","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Period":0,"Power":7,"Factor":0.47,"Voltage":99,"Current":0.154}}
14:34:52 MQT: tele/sonoff_pow5/SENSOR = {"Time":"2018-09-01T14:34:52","ENERGY":{"Total":0.003,"Yesterday":0.150,"Today":0.003,"Power":4,"Factor":0.26,"Voltage":99,"Current":0.154}}
```

arendst, 2018-09-01:
Set powerdelta 0 to get rid of unwanted data.

myfingerhurt, 2018-09-01:
@arendst  Thanks, it works. 
But I have checked my 6.1.1 with `powerdelta 80`, it doesn't show the same behaviour as 6.2.0 does.
Is it a bug?

closed 2018-09-01 by myfingerhurt

curzon01, 2018-09-01:
duplicate issue - see #3648